elFarto/nvidia-vaapi-driver #164: Freeze in egl_allocateBackingImage () at /usr/lib64/dri/nvidia_drv_video.so

stransky, 2023-01-25:
Hello,

I installed your library on NVIDIA GeForce GTX 1070/PCIe/SSE2 (driver	4.6.0 NVIDIA 525.78.01), Fedora 37. Tested both Wayland and X11 with the same result.

Looks like firts VA-API frame is decoded correctly:

```
[vp9 @ 0x7fd570a68400] Format vaapi chosen by get_format().
[vp9 @ 0x7fd570a68400] Format vaapi requires hwaccel initialisation.
[vp9 @ 0x7fd570a68400] Considering format 0x3231564e -> nv12.
[vp9 @ 0x7fd570a68400] Considering format 0x30313050 -> p010le.
[vp9 @ 0x7fd570a68400] Considering format 0x32313050 -> p012le.
[vp9 @ 0x7fd570a68400] Picked nv12 (0x3231564e) as best match for yuv420p.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x7.
[AVHWFramesContext @ 0x7fd5500cb400] Direct mapping disabled: deriving image does not work: 1 (operation failed).
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x8.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x9.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0xa.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0xb.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0xc.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0xd.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0xe.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0xf.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x10.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x11.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x12.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x13.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x14.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x15.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x16.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x17.
[AVHWFramesContext @ 0x7fd5500cb400] Created surface 0x18.
[vp9 @ 0x7fd570a68400] Decode context initialised: 0x19/0x1a.
[vp9 @ 0x7fd570a68400] Param buffer (type 0, 92 bytes) is 0x1b.
[vp9 @ 0x7fd570a68400] Slice 0 param buffer (316 bytes) is 0x1c.
[vp9 @ 0x7fd570a68400] Slice 0 data buffer (975837 bytes) is 0x1d.
[vp9 @ 0x7fd570a68400] Decode to surface 0x18.
```

but vaExportSurfaceHandle() is blocked:

```
#0  futex_wait (private=0, expected=2, futex_word=0x7fd570a20228) at ../sysdeps/nptl/futex-internal.h:146
#1  __GI___lll_lock_wait (futex=futex@entry=0x7fd570a20228, private=0) at lowlevellock.c:49
#2  0x00007fd570c91412 in lll_mutex_lock_optimized (mutex=0x7fd570a20228) at pthread_mutex_lock.c:48
#3  ___pthread_mutex_lock (mutex=0x7fd570a20228) at pthread_mutex_lock.c:93
#4  0x00007fd56db28311 in egl_allocateBackingImage () at /usr/lib64/dri/nvidia_drv_video.so
#5  0x00007fd56db2ce8a in egl_realiseSurface () at /usr/lib64/dri/nvidia_drv_video.so
#6  0x00007fd56db2a29d in nvExportSurfaceHandle () at /usr/lib64/dri/nvidia_drv_video.so
#7  0x00007fd562212535 in vaExportSurfaceHandle (dpy=0x7fd570a3e760, surface_id=24, mem_type=1073741824, flags=5, descriptor=0x7fd55e500170) at ../va/va.c:1561
#8  0x00007fd567aa41f0 in mozilla::FFmpegVideoDecoder<46465650>::GetVAAPISurfaceDescriptor(_VADRMPRIMESurfaceDescriptor*)
```

So Firefox switches to SW decode.

elFarto, 2023-01-25:
There should be a fix for that in master. However, there's currently a NVIDIA driver issue preventing the EGL backend in this driver from working with the NVIDIA 525 driver series. You'll have to use the direct backend for the moment (set `NVD_BACKEND=direct` to select that).

stransky, 2023-01-25:
Mozilla bug: https://bugzilla.mozilla.org/show_bug.cgi?id=1812277

stransky, 2023-01-25:
Confirmed, NVD_BACKEND=direct env variable works and VA-API is used. Good work!

closed 2023-02-12 by elFarto