elFarto/nvidia-vaapi-driver #168: Latest release not available as ubuntu package

GiscardBiamby, 2023-01-31:
The link on the README.md page of this repo links to an Ubuntu [package repository](https://packages.ubuntu.com/kinetic/nvidia-vaapi-driver) that has version .0.0.5-1, but I think the latest should be .0.0.8. 

Also, would it be possible to build a version for Ubuntu 22.04 (Jammy Jellyfish)? 20.04 is a bit outdated.

I tried building under Ubuntu but got some dependency errors (will post as a separate issue).

elFarto, 2023-02-05:
You'd need to send a message to the maintainers of that package to ask them to update it, as I'm not the one that builds it.

elFarto, 2023-02-08:
I have just updated the build dependencies for Ubuntu, which might help you get it built.

GiscardBiamby, 2023-02-08:
Thanks! I was able to build. I was also missing this dependency: `libdrm-dev`.

Hardware decoding seems to be enabled in firefox but the cpu usage still goes higher than I expected during playback.

I tried this:
I set these in my bash env:

```bash
export LIBVA_DRIVER_NAME=nvidia
export VDPAU_DRIVER=nvidia
export NVD_BACKEND=direct
export MOZ_DISABLE_RDD_SANDBOX=1
export EGL_PLATFORM=wayland
export __EGL_VENDOR_LIBRARY_FILENAMES="/usr/share/glvnd/egl_vendor.d/10_nvidia.json"
```

And ran this:

```bash
MOZ_LOG="PlatformDecoderModule:5" firefox-trunk 'https://www.youtube.com/watch?v=nGmLhLNYitE' &> ~/firefox.log
```
 
and the log seems to show that av1 acceleration is supported. vainfo now works too.  But cpu usage still seems high, around 20%.

elFarto, 2023-02-12:
Can you run firefox with NVD_LOG=1 set, and post the log here?

closed 2023-03-11 by elFarto