elFarto/nvidia-vaapi-driver #170: [Feature Request] VP9 with Alpha support

jjgr8, 2023-02-01:
Would it be possible to add Alpha support for VP9?

elFarto, 2023-02-01:
I'm not sure that either VA-API or NVDEC actually support decoding that (I've not seen any mention of it). Do you have a link to a sample video for me to test with?

jjgr8, 2023-02-01:
You can generate test files by adding the -pix_fmt option to the ffmpeg line from the samples script.  The following 2 lines are examples of what won't hardware decode (yuva420p.webm) and what will hardware decode (yuv420p.webm) in Firefox Developer Edition.

fmpeg -f lavfi -i smptebars=duration=10:size=640x360:rate=30 -c:v vp9 -pix_fmt yuva420p yuva420p.webm
fmpeg -f lavfi -i smptebars=duration=10:size=640x360:rate=30 -c:v vp9 -pix_fmt yuv420p yuv420p.webm
[yuva420p.webm](https://user-images.githubusercontent.com/124106722/216055057-d28245f9-b2e7-4ba9-8ee3-bbecdb721268.webm)
[yuv420p.webm](https://user-images.githubusercontent.com/124106722/216055060-f9229c44-f762-436e-a313-7cbcff57f82c.webm)

In the yuva420p.webm file, I'm not sure if an actual alpha channel is generated but it does generate the "alpha_mode: 1" flag in the metadata according to ffprobe which seems to be enough for hardware decoding to not work.

Also, thanks for all your great work!

elFarto, 2023-02-01:
I tried playing that back with mpv and it does use hardware decoding, so it might be a Firefox issue. However I don't see any mention in the output about it using yuva or an alpha channel, so it could just be ignoring it.

jjgr8, 2023-02-01:
Attached is a file...
[COLOR BARS WITH ALPHA 640x480.webm](https://user-images.githubusercontent.com/124106722/216154525-c1cccd42-71b8-4b22-908a-f518eff66fe3.webm)
that should show color bars but with a black circle.  The alpha is used to create the circle.  You'll see it does not use hardware decoding in Firefox.  Might this help?  To be fair, I noticed in Windows Chrome and Chromium, I can get hardware decoding to work with YUV420p file but not YUVA420p files so perhaps it's not possible with a video file that has an alpha?

elFarto, 2023-02-01:
Ok, as expected mpv is playing it back without the alpha, so it can use hardware decoding. Under firefox it doesn't even attempt to use the driver (it even doesn't initialise it), so it's not even checking if the driver is capable of decoding it.

I've checked over VA-API and NVDEC, and they don't seem to have any support for alpha channels. Without them supporting it, there's not much I can do.

jjgr8, 2023-02-01:
Thanks so much for checking into and confirming this and keep up the great work!

closed 2023-02-01 by jjgr8