elFarto/nvidia-vaapi-driver #179: Unable to decode video on Firefox with direct backend

MrAru, 2023-03-04:
I am using Arch Linux with X11, and I have GTX 1660Ti Mobile GPU card. I confirmed that I have set the kernel parameter `nvidia-drm.modeset`, set the environment variables `NVD_BACKEND`, `MOZ_DISABLE_RDD_SANDBOX`, and `LIBVA_DRIVER_NAME`, also I have set Firefox config parameters `media.ffmpeg.vaapi.enabled`, `media.rdd-ffmpeg.enabled`, and `gfx.x11-egl.force-enabled`. Hardware acceleration works well on mpv, while it doesn't work on Firefox (Hardware acceleration works fine on mpv, but not on Firefox (as confirmed by the fact that `Video Engine Utilization` is always zero when playing video on Firefox). Here is some information that might help.
```
❯ vainfo
Trying display: wayland
Trying display: x11
vainfo: VA-API version: 1.17 (libva 2.17.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
```

I've also tried `NVD_LOG=1 firefox-nightly`, but that doesn't give me any output related to Nvidia or VA-API or anything else. Does anyone have some clues? I'd be happy to provide further relevant information.

TJokiel, 2023-03-04:
I believe latest Firefox update broke something.
If I start Firefox 102.x ESR then 'Video Engine Utilization' shows usage when playing videos.

Also - if I start Firefox 102.x from command line (even without playing any video) I get this output:
`./firefox
libva info: VA-API version 1.16.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/va/drivers/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
`
If I start latest Firefox 110.0.1 from command line I do not get any output.

elFarto, 2023-03-04:
@MrAru I assume that because you have a mobile GPU, that you're using a laptop with an integrated GPU? If so, then you will likely have issues attempting to use this driver as Optimus setups aren't really supported (it's mostly Firefox that causes these issues).

@TJokiel The driver is working fine for me on FF110. Can you make sure that you have all the settings correct on FF110?, I'm not sure if the normal and ESR versions use different profiles. The fact that you see no libva output means it's not even trying to initialise it, which indicates it's not enabled.

MrAru, 2023-03-05:
@elFarto Thanks for pointing that out. It's true, I do use this driver with Optimus. So I suppose I'll have to wait for some fix from Firefox. I'll leave this issue as open since there are other discussions here, and you can close it anytime.

marioroy, 2023-03-05:
> I believe latest Firefox update broke something.

Same here. I needed to clear or remove ~/.mozilla and start fresh to including enabling vaapi in about:config. Adding extension "Adblocker Ultimate" breaks it. Though, not sure if due to the extension. Disabling or removing the extension does not resolve the issue. Again, I needed to clear or remove ~/.mozilla and start fresh.

marioroy, 2023-03-05:
Exporting `MOZ_DRM_DEVICE` resolved the issue for me. Your path may differ. The "Adblocker Ultimate" extension works including VA-API decoder accerleration.
```bash
export MOZ_DRM_DEVICE=/dev/dri/renderD128
```

TJokiel, 2023-03-05:
I confirm that exporting the 'MOZ_DRM_DEVICE=' variable resolves the problem for me.

Thanks!

MrAru, 2023-03-05:
Thanks @marioroy , after exporting the `MOZ_DRM_DEVICE` variable I was able to see the output in the terminal when running `NVD_LOG=1 firefox-nightly`. The full output is too long so I attached it below for anyone who is interested. The hardware acceleration, however, still doesn't work. I suppose this is mainly due to Optimus, just as elFarto said.

[For those interested here is the link but may not help](https://pastebin.mozilla.org/g30qS11s)

jeois, 2023-03-07:
The newest series of NVIDIA drivers (530) doesn't seem to work.  I downgraded to 525.89.02, and it works fine again with FF 110.0.1 and the direct backend, even without setting MOZ_DRM_DEVICE.  Btw, I'm also using GTX 1660 TI, except it's a dedicated card in a desktop.
Perhaps I misconfigured something, but this could be a possible driver regression.  I don't recall what NVD_LOG said exactly, but it didn't seem to be pertinent at first glance.  Sorry, I haven't tested it thoroughly yet.

hustin-paul, 2023-03-09:
Driver 525.89.02, exporting `MOZ_DRM_DEVICE` worked for me!

marioroy, 2023-03-10:
I tried Firefox inside Flatpak. Confirming not working in Flatpak using NVIDIA driver (530). But okay using (525).

https://github.com/elFarto/nvidia-vaapi-driver/issues/23#issuecomment-1464563852

jeois, 2023-03-12:
Well, now Nvidia driver 530.30.02 also works fine for me.  

I'm not sure what changed.  I've updated various system packages in the past few days, but I haven't really done anything to specifically fix it.  I'm now using linux kernel 6.2.5 and newer versions of ffmpeg and mesa, plus a new build of firefox-appmenu (110.0.1-4), but in that case, I think the automated build repo I use just rebuilt FF 110 for no particular reason, and that's likely unrelated to this issue.  Regardless, proprietary driver version 530 series works now.

elFarto, 2023-03-12:
I believe this has all been caused by a change the FireFox team made to block DMA-BUF from operating on certain NVIDIA driver versions. See [this comment](https://github.com/elFarto/nvidia-vaapi-driver/issues/182#issuecomment-1465138721) for more details.

crimist, 2023-03-12:
I can confirm that setting `widget.dmabuf.force-enabled` to `true` does mitigate the issue without setting `MOZ_DRM_DEVICE`. Seems like c66686e resolves this, marked as resolved / upstream issue?

MrAru, 2023-03-24:
I can confirm that with Nvidia driver version 530.41.03 everything just working as expected. Therefore I would like to close this issue as this thread seems no longer active. Thanks for any suggestions above!