elFarto/nvidia-vaapi-driver #180: driver stops working in firefox after some time (ff profile gets altered somehow)

POMATu, 2023-03-04:
Hello, this driver works great when it works. But I am having very strange issue with it. I configure fresh firefox profile do all about:config tweaks (all environment variables are in /etc/environment) and I can see it working both in console log and in nvidia-settings as Video Engine utilization. Then I use firefox for a bit and it stops working. I thought maybe its some addon, but nah I installed all my addons to fresh profile and its working. Disabling addons in borked profile doesnt help either. Happened to me 3 times already, once in firefox profile, another time in waterfox and in librewolf too.

Whats even more mindfucking here that when I tried tweaking about:config in old waterfox profile that I was using for a month even before I inserted nvidia card to my PC - it didnt worked straight away, but in fresh waterfox profile it worked and then died.

I assume firefox writes something to profile (which is NOT the options from howto because they are fine) that makes it not use this vaapi driver anymore. What it can be? I tried comparing two about:config pages of working and non-working profile and havent found anything useful. Ironically vaapi was blacklisted in some old working profile which i tweaked recently (probably from old times when I had no vaapi). So i realized there is no way I can parse it or find the option that makes it broken.

I will test more but right now I think i have to use firefox for youtube only and do other business in chrome because i am utterly tired creating new profiles and loggin in everywhere. Kindly elaborate any ideas what can possibly cause this.
`
Linux desktop 5.4.0-144-generic #161-Ubuntu SMP Fri Feb 3 14:49:04 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
Mozilla Firefox 110.0.1
ubuntu 20.04
nvidia RTX 1080
nvidia-driver-515
`

reliant8307, 2023-03-04:
Here Firefox stopped using nvidia-vaapi-driver because it couldn't find the DRM device, so I solved it by forcing Firefox to use the Nvidia card:

MOZ_DRM_DEVICE=/dev/dri/renderD129

maybe it could help in your case?

POMATu, 2023-03-04:
> MOZ_DRM_DEVICE=/dev/dri/renderD128

I tried that, and now it shows logs from the plugin, but my Video Engline still not utilized
```
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
ATTENTION: default value of option mesa_glthread overridden by environment.
libva info: va_openDriver() returns 0
libEGL warning: DRI2: failed to create dri screen
ATTENTION: default value of option mesa_glthread overridden by environment.
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
```
On top of that another profile of firefox broke ffs

POMATu, 2023-03-04:
So when i switched up to working profile of firefox it also has those Permission denied messages but it works
I believe the only problem is that firefox somehow shits into itself and profile becomes broken

Is there any way i can possibly trace that?

Fxzzi, 2023-03-05:
> so I solved it by forcing Firefox to use the Nvidia card:
> 
> MOZ_DRM_DEVICE=/dev/dri/renderD129

This seemed to fix it for me! I set mine to `/dev/dri/renderD128` and it's working

POMATu, 2023-03-07:
> > so I solved it by forcing Firefox to use the Nvidia card:
> > MOZ_DRM_DEVICE=/dev/dri/renderD129
> 
> This seemed to fix it for me! I set mine to `/dev/dri/renderD128` and it's working

I can confirm when I set it in /etc/environment my profiles dont die anymore. But didnt helped with old ones apparently firefox blacklisted something somewhere.
So i think this is mandatory option from now on, it should be marked as required.

shadywack, 2023-03-10:
Tried everything here, and I still get the permission denied error on renderD128. renderD129 doesn't exist.

Agreed that this seems like a Firefox problem, but it stops this from working. Currently broken.

elFarto, 2023-03-12:
I believe this has all been caused by a change the FireFox team made to block DMA-BUF from operating on certain NVIDIA driver versions. See [this comment](https://github.com/elFarto/nvidia-vaapi-driver/issues/182#issuecomment-1465138721) for more details.

POMATu, 2023-03-15:
> I believe this has all been caused by a change the FireFox team made to block DMA-BUF from operating on certain NVIDIA driver versions. See [this comment](https://github.com/elFarto/nvidia-vaapi-driver/issues/182#issuecomment-1465138721) for more details.

widget.dmabuf.force-enabled=true didnt helped me on dead profile. But it was enabled on working profile. Maybe thats the reason its still working beside everything else.
These soydevs and oversmart software driving me nuts. What do i need to install to see KISS again?

closed 2023-06-16 by elFarto