elFarto/nvidia-vaapi-driver #181: Missing DRM render device on Wayland

alphabitserial, 2023-03-06:
Firefox 110.0.1
KWin 5.27.2
NVIDIA driver 525.89.02
Arch Linux
RTX 3050, no integrated graphics on my system whatsoever

Everything is working okay in X11 but when I switch to Wayland, Firefox can't find the DRM render device and therefore doesn't use nvidia-vaapi-driver. I'm using the most recent commit of nvidia-vaapi-driver and open-gpu-kernel-modules as of the time I'm writing this. I manually built nvidia-vaapi-driver following the instructions from this repo, not using any AUR packages.

**/etc/environment:**
MOZ_ENABLE_WAYLAND=1
EGL_PLATFORM=wayland
MOZ_DISABLE_RDD_SANDBOX=1
NVD_BACKEND=direct
LIBVA_DRIVER_NAME=nvidia
GBM_BACKEND=nvidia-drm

**Firefox settings:**
media.ffmpeg.vaapi.enabled = true
media.rdd-ffmpeg.enabled = true
gfx.x11-egl.force-enabled = true

**Firefox logs:**
❯ MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" NVD_LOG=1 firefox
[Parent 30714: Main Thread]: D/Dmabuf wl_drm is available.
[Parent 30714: Main Thread]: D/Dmabuf zwp_linux_dmabuf_v1 is available.
[Parent 30714: Main Thread]: D/Dmabuf We're missing DRM render device!

**vainfo:**
Trying display: wayland
vainfo: VA-API version: 1.17 (libva 2.17.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD

Please let me know if there's anything I'm missing or anything I can provide to help with debugging.

alphabitserial, 2023-03-06:
Manually specifying the DRM render device with `MOZ_DRM_DEVICE=/dev/dri/renderD128` enables nvidia-vaapi-driver to work on Wayland, so it seems the issue specifically is with enumerating render devices.

thesword53, 2023-03-06:
I also have this issue on Wayland and X11 even with my Intel HD graphics (i915).