elFarto/nvidia-vaapi-driver #182: HWacceleration stopped working with recent updates - with solution

Korothi, 2023-03-09:
Hi,
tested on wayland and latest everything. i solved the issue already by adding changing this: widget.dmabuf.force-enabled to true
heres how about:support looks:
[image]
[image]
env (wayland):
[image]
proof that it works (somehow):
[image]

ill give a try in x11 later.

crimist, 2023-03-09:
I also require `widget.dmabuf.force-enabled=true` when using Wayland. Maybe we should update the README to reflect that it may still be required on NVIDIA drivers >495?

elFarto, 2023-03-11:
Seems I'm suffering from this issue aswell. It seems deleting the `gfx.blacklist.dmabuf` keys from `about:config` and restarting will let it work, until I restart and they reappear.

In the `about:support` page, do you have `Blocklisted; failure code FEATURE_FAILURE_DL_BLOCKLIST_NO_ID` in the DMABUF section when it's not working?

Monsterovich, 2023-03-11:
> Maybe we should update the README to reflect that it may still be required on NVIDIA drivers >495?

@elFarto The driver 525.85.05 now requires this option.

elFarto, 2023-03-12:
> > Maybe we should update the README to reflect that it may still be required on NVIDIA drivers >495?
> 
> @elFarto The driver 525.85.05 now requires this option.

I'm not so sure. Something's changed in Firefox, and I think it's just a bug as clearing out the errors in about:config lets it work normally, once, after restarting. Using MOZ_DRM_DEVICE or force-enable to override this is just working around it.

elFarto, 2023-03-12:
Ok, after running the latest nightly version I think I see what's happened. There's an [unrelated issue](https://bugzilla.mozilla.org/show_bug.cgi?id=1788573) that's caused the FireFox team to disable DMA-BUF for certain NVIDIA driver versions.

So yes, for the moment `widget.dmabuf.force-enabled=true` is needed to override this. I'll update the README.

Monsterovich, 2023-03-12:
> I'm not so sure. Something's changed in Firefox, and I think it's just a bug as clearing out the errors in about:config lets it work normally, once, after restarting. Using MOZ_DRM_DEVICE or [...]

I'm still sitting on FF 107.0, and all that has changed for me is the driver.

philipl, 2023-03-12:
Despite the bug saying it is fixed in 530, I saw it blocked on 530 here. 🙄

jeois, 2023-03-12:
Sorry for hijacking that other thread #179 where integrated GPU users were setting MOZ_DRM_DEVICE.  

I'm still not sure if I how I fixed Nvidia driver 530 as I likely had `widget.dmabuf.force-enabled` set to `true` this whole time, but perhaps I did toggle it at some point.  In any case, this is the best explanation after reading that bugzilla [thread](https://bugzilla.mozilla.org/show_bug.cgi?id=1788573).

I've always had FF crashes upon resume from sleep.  However, I can't believe their solution was to disable DMABUF for all Nvidia drivers 510-530.  So in order to "fix" something which only requires the user to restart their application upon resume, they blacklisted everything and disabled our functionality even when it was running fine, smh.

Based on their conversation, they think this issue cropped up in Nvidia version 525 series and then added 510 and 515, but I have had that suspend/crash issue for years and disabling DMA_BUF never solved it.  They do seem to be getting our crash reports though.  They assumed it would be an easy fix from Nvidia in the next version, which would have been 530 series at the time, but it seems they also blocked 530 as well, wth.

Darkspirit, 2023-03-13:
> So yes, for the moment widget.dmabuf.force-enabled=true is needed to override this. I'll update the README.

a) Please don't because users would run into a confirmed use-after-free Nvidia EGL driver bug: https://bugzilla.mozilla.org/show_bug.cgi?id=1788573#c26
b) Please do it only if you explicitly state that the Nvidia suspend service must be enabled and  NVreg_PreserveVideoMemoryAllocations=1 must be set.

POMATu, 2023-03-21:
> a) Please don't because users would run into a confirmed use-after-free Nvidia EGL driver bug: https://bugzilla.mozilla.org/show_bug.cgi?id=1788573#c26

I do not use suspend/hibernate at all because i am on desktop. My PC never turns off. But without that setting you just get no hardware video decoding.
Hence I think all you mentioned should be explained in readme so users at least could know what they can try to make it work again.

Because in 2023th year its not possible to find anything useful in search engines. And reading each github issue of this project is not very useful either.

Vednier, 2023-03-25:
> So yes, for the moment `widget.dmabuf.force-enabled=true` is needed to override this. I'll update the README.

In that case please add that its highly recommended to enable GPU process in Firefox with layers.gpu-process.enabled=true, which will absorb scary crashes mentioned by Darkspirit. According to Mozilla devs GPU process was disabled by default because its incompatible with Wayland, however its still perfectly working on X11.

Vednier, 2023-03-28:
Bad news - fix for [@ NvGlEglGetFunctions ] didnt made it to newest Nvidia drivers, so Mozilla decided to block DMABUF on Nvidia globally - https://bugzilla.mozilla.org/show_bug.cgi?id=1824778#c10

POMATu, 2023-03-29:
> Bad news - fix for [@ NvGlEglGetFunctions ] didnt made it to newest Nvidia drivers, so Mozilla decided to block DMABUF on Nvidia globally - https://bugzilla.mozilla.org/show_bug.cgi?id=1824778#c10

Will there be any way to force enable dmabuf? I am on old nvidia drivers (515) and dont experience any crashes

Darkspirit, 2023-03-29:
You need to enable the Nvidia suspend service, NVreg_PreserveVideoMemoryAllocations=1 must be set, and then you can enable  widget.dmabuf.force-enabled on about:config in Firefox:
*    https://askubuntu.com/a/1309807
*    https://wiki.archlinux.org/title/NVIDIA/Tips_and_tricks#Preserve_video_memory_after_suspend
*    https://download.nvidia.com/XFree86/Linux-x86_64/510.39.01/README/powermanagement.html

Vednier, 2023-03-29:
> then you can enable widget.dmabuf.force-enabled on about:config in Firefox

And that is risks i you simply force enable DMABUF without suspend service? Just crashes? GPU process makes this problem relatively minor.

ManuLinares, 2023-04-05:
I had a problem, didn't see the change in the readme. Thanks to this thread I enabled "widget.dmabuf.force-enabled" and now it works again.

I don0t suspend/hibernate either, so only enabled this.

```
nvidia-dkms 530.41.03-1
firefox 111.0.1-1.1

```

closed 2023-06-16 by elFarto