elFarto/nvidia-vaapi-driver #187: Firefox tries to use hardware decoding for a few seconds once after boot and then never again

X-m7, 2023-03-12:
While my setup does have NVIDIA Optimus (Intel HD 530 and GTX 960M) I am also using `nvidia-xrun`, so everything runs on the NVIDIA GPU. After boot and trying to play a YouTube video with the enhanced-h264ify extension the video engine usage spikes for a few seconds, and then it never rises again even after restarting Firefox, unless I restart the whole system.

`mpv --hwdec` works as expected, video engine utilisation is up and CPU usage is down.

I have `LIBVA_DRIVER_NAME=nvidia`, `MOZ_DISABLE_RDD_SANDBOX=1`, `MOZ_DRM_DEVICE=/dev/dri/renderD128`, and `NVD_BACKEND=direct` set, and in Firefox I also have `gfx.x11-egl.force-enabled` and `media.ffmpeg.vaapi.enabled` set to true.

<details>
<summary>Output of running `vainfo` with `NVD_LOG=1`</summary>

```
Trying display: wayland
Trying display: x11
       691.168243319 [2238-2238] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x56028caa2380 10
       691.168257165 [2238-2238] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
       691.168263212 [2238-2238] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
       691.173938315 [2238-2238] ../nvidia-vaapi-driver-0.0.8/src/direct/direct-export-buf.c:  94      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
       691.173950219 [2238-2238] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
       691.173958233 [2238-2238] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 1 0 0
       691.174070733 [2238-2238] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.89.02
vainfo: VA-API version: 1.17 (libva 2.17.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      <unknown profile>               : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
       691.234864010 [2238-2238] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x56028caa2380
       691.234928210 [2238-2238] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
```

</details>

<details>
<summary>Output of running Firefox with `NVD_LOG=1`</summary>

```
        92.149724157 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f0666c23c40 31
        92.149746310 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
        92.149756082 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
        92.156682144 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
        92.156701704 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 1 0 0
        92.156817229 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.89.02
        92.195669014 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1220      nvQueryImageFormats In nvQueryImageFormats
        92.224140787 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 6 with 0 attributes
        92.224161900 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
        92.224169237 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7f0646beee80 7
        92.226016665 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 7 with 0 attributes
        92.226034407 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
        92.226040160 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7f0646beee80 7
        92.228265752 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 578           nvCreateConfig got profile: 13 with 0 attributes
        92.228283702 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) (nil) 0
        92.228289527 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1596 nvQuerySurfaceAttributes with 4 (8) 0x7f0646beee80 7
        92.230910737 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x7f0666c23c40
        92.231020176 [1470-1507] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
        93.263245583 [1470-1517] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f0666c227a0 31
        93.263265968 [1470-1517] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
        93.263277300 [1470-1517] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
        93.263316488 [1470-1517] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
        93.263328039 [1470-1517] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 100 1 0 0
        93.263447327 [1470-1517] ../nvidia-vaapi-driver-0.0.8/src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.89.02
        93.301775087 [1470-1517] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1220      nvQueryImageFormats In nvQueryImageFormats
        93.302321266 [1470-1517] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x7f0666c227a0
        93.302459694 [1470-1517] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
        93.324322271 [1470-1522] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x7f0666c22b60 31
        93.324339963 [1470-1522] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
[... 115 lines cut ...]
        93.848296403 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640917e20)
        93.848309282 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640917ce0)
        93.848323113 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640917c40)
        93.848338899 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640917ba0)
        93.848353282 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640917b00)
        93.848368045 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640917a60)
        93.848381766 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f06409179c0)
        93.848394093 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640917920)
        93.848410277 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640917d80)
        93.848424627 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640917880)
[... cut ...]
        93.848555679 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f06409165c0)
        93.848569777 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c: 776        nvDestroySurfaces Destroying surface -1 (0x7f0640916520)
        93.848585997 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1843              nvTerminate Terminating 0x7f0646bd4d40
        93.848746979 [1470-1538] ../nvidia-vaapi-driver-0.0.8/src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
```

</details>

elFarto, 2023-03-12:
This error is from your Intel driver, which means that Firefox isn't fully running on the NVIDIA GPU.
```
eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: requested buffer attributes are not supported
```
This would indicate that nvidia-xrun isn't setting/overriding what EGL drivers Firefox can see. Can you try running `eglinfo` under it, and see what drivers it can see?

X-m7, 2023-03-12:
I forgot to mention that I also have `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json` set already as well, so I don't think the Intel driver should be doing anything (although I guess given how much trouble Optimus has given me over the years it wouldn't be surprising if I was wrong I suppose).

<details>
<summary>Anyway, here's the output of `eglinfo` :</summary>

```
EGL client extensions string:
    EGL_EXT_platform_base EGL_EXT_device_base
    EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions
    EGL_KHR_debug EGL_KHR_platform_x11 EGL_EXT_platform_x11
    EGL_EXT_platform_device EGL_MESA_platform_surfaceless
    EGL_EXT_explicit_device EGL_KHR_platform_wayland
    EGL_EXT_platform_wayland EGL_KHR_platform_gbm EGL_MESA_platform_gbm

GBM platform:
EGL API version: 1.5
EGL vendor string: NVIDIA
EGL version string: 1.5
EGL client APIs: OpenGL_ES OpenGL
EGL extensions string:
    EGL_EXT_buffer_age EGL_EXT_client_sync
    EGL_EXT_create_context_robustness EGL_EXT_image_dma_buf_import
    EGL_EXT_image_dma_buf_import_modifiers EGL_MESA_image_dma_buf_export
    EGL_EXT_output_base EGL_EXT_output_drm EGL_EXT_protected_content
    EGL_EXT_stream_consumer_egloutput EGL_EXT_stream_acquire_mode
    EGL_EXT_sync_reuse EGL_IMG_context_priority EGL_KHR_config_attribs
    EGL_KHR_create_context_no_error EGL_KHR_context_flush_control
    EGL_KHR_create_context EGL_KHR_fence_sync
    EGL_KHR_get_all_proc_addresses EGL_KHR_partial_update
    EGL_KHR_swap_buffers_with_damage EGL_KHR_no_config_context
    EGL_KHR_gl_colorspace EGL_KHR_gl_renderbuffer_image
    EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_3D_image
    EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base
    EGL_KHR_reusable_sync EGL_KHR_stream EGL_KHR_stream_attrib
    EGL_KHR_stream_consumer_gltexture EGL_KHR_stream_cross_process_fd
    EGL_KHR_stream_fifo EGL_KHR_stream_producer_eglsurface
    EGL_KHR_surfaceless_context EGL_KHR_wait_sync EGL_NV_nvrm_fence_sync
    EGL_NV_quadruple_buffer EGL_NV_stream_consumer_eglimage
    EGL_NV_stream_cross_display EGL_NV_stream_cross_object
    EGL_NV_stream_cross_process EGL_NV_stream_cross_system
    EGL_NV_stream_dma EGL_NV_stream_flush EGL_NV_stream_metadata
    EGL_NV_stream_remote EGL_NV_stream_reset EGL_NV_stream_socket
    EGL_NV_stream_socket_inet EGL_NV_stream_socket_unix
    EGL_NV_stream_sync EGL_NV_stream_fifo_next
    EGL_NV_stream_fifo_synchronous EGL_NV_stream_consumer_gltexture_yuv
    EGL_NV_stream_attrib EGL_NV_stream_origin EGL_NV_system_time
    EGL_NV_output_drm_flip_event EGL_NV_triple_buffer
    EGL_NV_robustness_video_memory_purge EGL_EXT_present_opaque
    EGL_WL_bind_wayland_display EGL_WL_wayland_eglstream
Configurations:
     bf lv colorbuffer dp st  ms    vis   cav bi  renderable  supported
  id sz  l  r  g  b  a th cl ns b    id   eat nd gl es es2 vg surfaces 
---------------------------------------------------------------------
0x01 32  0  8  8  8  8 24  8  0 0 0x34325241--         y  y  y     win,pb,str
0x02 32  0  8  8  8  8 24  0  0 0 0x34325241--         y  y  y     win,pb,str
0x03 32  0  8  8  8  8  0  0  0 0 0x34325241--         y  y  y     win,pb,str
0x04 32  0  8  8  8  8 24  8  2 1 0x34325241--         y  y  y     win,pb,str
0x05 32  0  8  8  8  8 24  0  2 1 0x34325241--         y  y  y     win,pb,str
0x06 32  0  8  8  8  8  0  0  2 1 0x34325241--         y  y  y     win,pb,str
0x07 32  0  8  8  8  8 24  8  4 1 0x34325241--         y  y  y     win,pb,str
0x08 32  0  8  8  8  8 24  8  4 1 0x34325241--         y  y  y     win,pb,str
0x09 32  0  8  8  8  8 24  0  4 1 0x34325241--         y  y  y     win,pb,str
0x0a 32  0  8  8  8  8 24  0  4 1 0x34325241--         y  y  y     win,pb,str
0x0b 32  0  8  8  8  8  0  0  4 1 0x34325241--         y  y  y     win,pb,str
0x0c 32  0  8  8  8  8  0  0  4 1 0x34325241--         y  y  y     win,pb,str
0x0d 32  0  8  8  8  8 24  8  8 1 0x34325241--         y  y  y     win,pb,str
0x0e 32  0  8  8  8  8 24  0  8 1 0x34325241--         y  y  y     win,pb,str
0x0f 32  0  8  8  8  8  0  0  8 1 0x34325241--         y  y  y     win,pb,str
0x10 24  0  8  8  8  0 24  8  0 0 0x34325258--         y  y  y     win,pb,str
0x11 24  0  8  8  8  0 24  0  0 0 0x34325258--         y  y  y     win,pb,str
[... 214 lines cut ...]
0x1d 24  0  8  8  8  0 24  0  8 1 0x00--         y  y  y     pb,str
0x1e 24  0  8  8  8  0  0  0  8 1 0x00--         y  y  y     pb,str
0x1f 16  0  5  6  5  0 24  8  0 0 0x00--         y  y  y     pb,str
0x20 16  0  5  6  5  0 24  0  0 0 0x00--         y  y  y     pb,str
0x21 16  0  5  6  5  0 16  0  0 0 0x00--         y  y  y     pb,str
0x22 16  0  5  6  5  0  0  0  0 0 0x00--         y  y  y     pb,str
0x23 16  0  5  6  5  0 24  8  2 1 0x00--         y  y  y     pb,str
0x24 16  0  5  6  5  0 24  0  2 1 0x00--         y  y  y     pb,str
0x25 16  0  5  6  5  0 16  0  2 1 0x00--         y  y  y     pb,str
0x26 16  0  5  6  5  0  0  0  2 1 0x00--         y  y  y     pb,str
0x27 16  0  5  6  5  0 24  8  4 1 0x00--         y  y  y     pb,str
0x28 16  0  5  6  5  0 24  8  4 1 0x00--         y  y  y     pb,str
0x29 16  0  5  6  5  0 24  0  4 1 0x00--         y  y  y     pb,str
0x2a 16  0  5  6  5  0 24  0  4 1 0x00--         y  y  y     pb,str
0x2b 16  0  5  6  5  0 16  0  4 1 0x00--         y  y  y     pb,str
0x2c 16  0  5  6  5  0 16  0  4 1 0x00--         y  y  y     pb,str
0x2d 16  0  5  6  5  0  0  0  4 1 0x00--         y  y  y     pb,str
0x2e 16  0  5  6  5  0  0  0  4 1 0x00--         y  y  y     pb,str
0x2f 16  0  5  6  5  0 24  8  8 1 0x00--         y  y  y     pb,str
0x30 16  0  5  6  5  0 24  0  8 1 0x00--         y  y  y     pb,str
0x31 16  0  5  6  5  0 16  0  8 1 0x00--         y  y  y     pb,str
0x32 16  0  5  6  5  0  0  0  8 1 0x00--         y  y  y     pb,str
```

</details>

elFarto, 2023-03-12:
That eglinfo output seems fine, so I'm not sure what's going on here.

X-m7, 2023-03-12:
I just realised that running `mpv --hwdec` just like that makes it use NVDEC directly instead of going through VAAPI, so I tried using `mpv --hwdec=vaapi` and it also showed the same EGL_BAD_PARAMETER error, and it gave up on hardware decoding as well. However, `mpv --hwdec=vaapi-copy` does work, with CPU usage sitting in between software decoding and using NVDEC directly.

Also, I've updated to 0.0.9 now in case that makes a difference.

<details>
<summary>Output of running `mpv --hwdec=vaapi` with `NVD_LOG=1`</summary>

```
 (+) Video --vid=1 (*) (h264 1920x1080 60.000fps)
 (+) Audio --aid=1 (*) (aac 2ch 48000Hz)
     12531.125389036 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     12531.125402554 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
     12531.125408007 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2200       __vaDriverInit_1_0 Selecting Direct backend
     12531.125436396 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
     12531.125442569 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 217            init_nvdriver Initing nvdriver...
     12531.125449623 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 222            init_nvdriver Got dev info: 100 1 0 0
     12531.125539186 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 283            init_nvdriver NVIDIA kernel driver version: 525.89.02
     12531.154640547 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:1408      nvQueryImageFormats In nvQueryImageFormats
     12531.179538189 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 647           nvCreateConfig got profile: 0 with 0 attributes
     12531.179556871 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:1762 nvQuerySurfaceAttributes with 1 (8) (nil) 0
     12531.179562671 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:1762 nvQuerySurfaceAttributes with 1 (8) 0x7fd3c4b5bcc0 8
     12531.180912592 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
     12531.180933315 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 930        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7fd3c4a62fb0)
     12531.180940579 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:1502            nvDeriveImage In nvDeriveImage
     12531.180947667 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 930        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7fd3c4a638f0)
     12531.180952692 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:1502            nvDeriveImage In nvDeriveImage
     12531.180992927 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2052    nvExportSurfaceHandle Exporting surface: -1 (0x7fd3c4a638f0)
     12531.180999329 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/direct-export-buf.c: 198 direct_allocateBackingImage Allocating BackingImages: 0x7fd3c4a63990 128x128
     12531.181004921 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 430              alloc_image Calculated GOB size: 64x128 (0x4)
     12531.181010045 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 439              alloc_image Aligned image size: 128x128 = 16384
     12531.181169117 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 515              alloc_image created image: 128x128 300000000400014 128 4000
     12531.181176158 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 430              alloc_image Calculated GOB size: 64x64 (0x3)
     12531.181181633 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 439              alloc_image Aligned image size: 128x64 = 8192
     12531.181295440 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 515              alloc_image created image: 64x64 300000000400013 128 2000
     12531.181301955 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/direct-export-buf.c: 204 direct_allocateBackingImage Importing images
     12531.181306641 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 128x128 = 4000
     12531.181486365 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 64x64 = 2000
     12531.181669501 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2065    nvExportSurfaceHandle Exporting with 128 128 0 128 300000000400014 0 128 300000000400013
     12531.181697953 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/direct-export-buf.c:  42                    debug [EGL] eglCreateImageKHR: EGL_BAD_PARAMETER error: In eglCreateImageKHR: requested buffer attributes are not supported

     12531.181718482 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd3c4a638f0)
     12531.181723949 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/direct-export-buf.c: 243      destroyBackingImage Destroying BackingImage: 0x7fd3c4a63990
[... cut ...]
     12531.222603075 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/direct/direct-export-buf.c: 243      destroyBackingImage Destroying BackingImage: 0x7fd3c4a6d2f0
     12531.222781108 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd3c4a64960)
     12531.222790141 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2075              nvTerminate Terminating 0x7fd3c4440600
     12531.222858061 [32507-32518] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2089              nvTerminate Now have 0 (0 max) instances
```

</details>