elFarto/nvidia-vaapi-driver #188: build fails on i686: ../src/vabackend.c:418:13: error: format ‘%llX’

kwizart, 2023-03-12:
`../src/vabackend.c:418:13: error: format ‘%llX’ expects argument of type ‘long long unsigned int’, but argument 6 has type ‘CUdeviceptr’ {aka ‘unsigned int’} [-Werror=format=]`

Building for i686 target still relevant today for compat with 32bit userspace.

Code works on others supported arches:
https://koji.rpmfusion.org/koji/taskinfo?taskID=587308

elFarto, 2023-03-12:
Thanks for that. I've pushed a fix up. Annoyingly there's no printf format specifier for platform sized int, so I've just cast it to a void*.

kwizart, 2023-03-12:
Thanks for the fix.

Unfortunately I've found more error, (fixed by the merge request).
```
../src/direct/nv-driver.c: In function ‘nv_alloc_object’:
../src/direct/nv-driver.c:26:29: error: initialization of ‘long long unsigned int’ from ‘void *’ makes integer from pointer without a cast [-Werror=int-conversion]
   26 |         .pRightsRequested = NULL,
      |                             ^~~~
../src/direct/nv-driver.c:26:29: note: (near initialization for ‘alloc.pRightsRequested’)
../src/direct/nv-driver.c:27:24: error: initialization of ‘long long unsigned int’ from ‘void *’ makes integer from pointer without a cast [-Werror=int-conversion]
   27 |         .pAllocParms = params
      |                        ^~~~~~
../src/direct/nv-driver.c:27:24: note: (near initialization for ‘alloc.pAllocParms’)
../src/direct/nv-driver.c: In function ‘nv_rm_control’:
../src/direct/nv-driver.c:69:19: error: initialization of ‘long long unsigned int’ from ‘void *’ makes integer from pointer without a cast [-Werror=int-conversion]
   69 |         .params = params,
      |                   ^~~~~~
../src/direct/nv-driver.c:69:19: note: (near initialization for ‘control.params’)
../src/direct/nv-driver.c: In function ‘nv_get_versions’:
../src/direct/nv-driver.c:161:33: error: initialization of ‘long long unsigned int’ from ‘char *’ makes integer from pointer without a cast [-Werror=int-conversion]
  161 |         .pDriverVersionBuffer = driverVersionBuffer,
      |                                 ^~~~~~~~~~~~~~~~~~~
../src/direct/nv-driver.c:161:33: note: (near initialization for ‘params.pDriverVersionBuffer’)
../src/direct/nv-driver.c:162:27: error: initialization of ‘long long unsigned int’ from ‘char *’ makes integer from pointer without a cast [-Werror=int-conversion]
  162 |         .pVersionBuffer = versionBuffer,
      |                           ^~~~~~~~~~~~~
../src/direct/nv-driver.c:162:27: note: (near initialization for ‘params.pVersionBuffer’)
../src/direct/nv-driver.c:163:25: error: initialization of ‘long long unsigned int’ from ‘char *’ makes integer from pointer without a cast [-Werror=int-conversion]
  163 |         .pTitleBuffer = titleBuffer
      |                         ^~~~~~~~~~~
../src/direct/nv-driver.c:163:25: note: (near initialization for ‘params.pTitleBuffer’)
../src/direct/nv-driver.c: In function ‘alloc_image’:
../src/direct/nv-driver.c:476:29: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
  476 |         .nvkms_params_ptr = (uint64_t) &nvkmsParams,
      |                             ^
../src/direct/nv-driver.c: In function ‘init_nvdriver’:
../src/direct/nv-driver.c:285:5: warning: ‘ver’ may be used uninitialized [-Wmaybe-uninitialized]
  285 |     free(ver);
      |     ^~~~~~~~~
../src/direct/nv-driver.c:281:11: note: ‘ver’ was declared here
  281 |     char *ver;
```

kwizart, 2023-03-12:
Side note:
Before finding your fix I've fixed the original error with:

`LOG("Mapped surface %d to %llX (%d)", surface->pictureIdx, (unsigned long long) deviceMemory, pitch);`

now I've just found that deviceMemory is defined as:
CUdeviceptr deviceMemory = (CUdeviceptr) NULL;

So maybe the same cast should be used there...

Thanks for the quickfix !

elFarto, 2023-03-12:
CUdeviceptr is a tricky one, it's technically an uint32 or an uint64 depending on platform, so I shouldn't really be using NULL here. But with that said, it is a 'pointer' to GPU memory (even if it's never deferenced), so a pointer seems appropriate.