elFarto/nvidia-vaapi-driver #193: Error Twitch streaming after upgrading to Fedora 38 Beta

ale2695, 2023-03-22:
Hi there,
I've just upgraded to Fedora 38 Beta from my Fedora 37 installation, and the driver stopped working when i played a livestream from Twitch or a video from Facebook Watch.
The NVD_LOG=1 firefox command show this error:
[2023-03-22T20:15:54Z ERROR error_support::handling] Error opening database: Incompatible database version: 2
With YouTube it works just fine, only shows this errors before start to decode the video:
[2023-03-22T20:18:28Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2023-03-22T20:18:28Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2023-03-22T20:18:28Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
[2023-03-22T20:18:28Z ERROR mp4parse] Found 2 nul bytes in "\0\0"
I've tried both the package from RPMFusion and the package installed from scratch, and the error occurs in both cases.
Can you please help me?
Thank you so much in advance

elFarto, 2023-03-22:
Can you check if `widget.dmabuf.force-enabled=true` is set in `about:config`? If not, can you retry with that set.

ale2695, 2023-03-22:
Hi, the key is set correctly

ale2695, 2023-03-23:
Update: just now has landed a new update for Firefox on the Fedora 38 repos, after the update the error is gone but the decoding doesn't working either.
I imagine that there is something wrong with the H.264 decoding, the VP9 (used by YT) decoding works just fine

aidanlw505, 2023-04-05:
@ale2695 I ran into a similar issue after a fresh install of Fedora 38. Installing the `mozilla-openh264` package fixed it for me.

elFarto, 2023-04-16:
Is this still an issue?

ale2695, 2023-06-15:
Nope resolved