elFarto/nvidia-vaapi-driver #196: Failed to open drm render node /dev/dri/renderD128 error Permission denied

schulzh, 2023-03-27:
I cannot get Firefox on X11 (Arch) to use hardware acceleration.

Firefox shows this:
```
MOZ_LOG="PlatformDecoderModule:5,Dmabuf:5" NVD_LOG=1 MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia LIBVA_MESSAGING_LEVEL=1 MOZ_X11_EGL=1 MOZ_DRM_DEVICE=/dev/dri/renderD128 firefox
[Parent 321272: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Parent 321272: Main Thread]: D/Dmabuf nsDMABufDevice::Configure()
[Parent 321272: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 321272: Main Thread]: D/Dmabuf DMABuf is enabled
[GPU 321338: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in GPU process
[Child 321703: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Child 321703: Main Thread]: D/Dmabuf Failed to open drm render node /dev/dri/renderD128 error Permission denied
[Child 321505: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[...]
```
I am assuming the error is `Failed to open drm render node /dev/dri/renderD128 error Permission denied`, but solutions from similar issues have not worked for me. I've added my user to the `video` and `render` group and I've tried chmodding everything in `/dev/dri` to 777, but the error persists. `/dev/dri/renderD128` and `/dev/dri/card0` map to the nvidia gpu (`card1` and `render129`  is my iGPU):

```
ls -la /dev/dri        
total 0
drwxrwxrwx   3 root root        140 27. Mär 10:50 .
drwxr-xr-x  21 root root       5440 27. Mär 15:07 ..
drwxr-xr-x   2 root root        120 27. Mär 10:57 by-path
crwxrwxrwx+  1 root video  226,   0 27. Mär 10:57 card0
crwxrwxrwx+  1 root video  226,   1 27. Mär 10:50 card1
crwxrwxrwx   1 root render 226, 128 27. Mär 10:50 renderD128
crwxrwxrwx   1 root render 226, 129 27. Mär 10:50 renderD129

ls -la /dev/dri/by-path
total 0
drwxr-xr-x 2 root root 120 27. Mär 10:57 .
drwxrwxrwx 3 root root 140 27. Mär 10:50 ..
lrwxrwxrwx 1 root root   8 27. Mär 10:50 pci-0000:00:02.0-card -> ../card1
lrwxrwxrwx 1 root root  13 27. Mär 10:50 pci-0000:00:02.0-render -> ../renderD129
lrwxrwxrwx 1 root root   8 27. Mär 10:57 pci-0000:01:00.0-card -> ../card0
lrwxrwxrwx 1 root root  13 27. Mär 10:50 pci-0000:01:00.0-render -> ../renderD128

lspci
[...]
00:02.0 Display controller: Intel Corporation AlderLake-S GT1 (rev 0c)
[...]
01:00.0 VGA compatible controller: NVIDIA Corporation TU102 [GeForce RTX 2080 Ti Rev. A] (rev a1)
[...]
```

vainfo works:
```
$ LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 vainfo
Trying display: wayland
Trying display: x11
     15907.833598500 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     15907.833606514 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
     15907.833609376 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2197       __vaDriverInit_1_0 Selecting EGL backend
     15907.836445978 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 132       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
     15907.836454816 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 149       findGPUIndexFromFd Looking for GPU index: 0
     15907.837840167 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 161       findGPUIndexFromFd Found 4 EGL devices
     15907.837950748 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     15907.837986973 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
     15907.838919181 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 277         egl_initExporter Driver supports 16-bit surfaces
vainfo: VA-API version: 1.17 (libva 2.17.1)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
     15907.926402399 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2075              nvTerminate Terminating 0x557ae2b93460
     15907.926407579 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c:  61      egl_releaseExporter Releasing exporter, 0 outstanding frames
     15907.926410628 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c:  78      egl_releaseExporter Done releasing frames
     15907.926412464 [333127-333127] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2089              nvTerminate Now have 0 (0 max) instances
```

elFarto, 2023-03-27:
I'm not sure I can help you with that. It's not an issue with the driver, or with Firefox, but with the OS setup itself and I don't use Arch. You can try running `vainfo --display drm --device /dev/dri/renderD128` which will tell vainfo to access the card via DRM, rather than via X which should give you the same error.

Also when adding yourself to the group, it doesn't normally take effect until you log out and back in again.

schulzh, 2023-03-27:
The command works:
```
vainfo --display drm --device /dev/dri/renderD128
Trying display: drm
vainfo: VA-API version: 1.17 (libva 2.17.1)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
```
It also works with the intel card, but both give the same error in Firefox.
I've tried some more things and it looks like the content sandbox is causing the permission denied. If I run firefox with `MOZ_DISABLE_CONTENT_SANDBOX=1`, the error diappears and it looks like its doing something with vaapi (and also this driver), but it still does not work (0% Video Engine Utilization). I've captured some logs: https://paste.ee/p/fiiYr

elFarto, 2023-04-16:
My apologies this issue completely slipped my mind. There might be two issues here. One of them is that you're trying to use the EGL backing, try adding `NVD_BACKEND=direct`.

In addition needing to set MOZ_DISABLE_CONTENT_SANDBOX might indicate that you have some of the `media.rdd*` settings in `about:config` disabled, specifically `media.rdd-process.enabled`, `media.rdd-ffvpx.enabled`, and `media.rdd-ffmpeg.enabled`.

schulzh, 2023-04-16:
Thanks a lot, you were correct on both issues. It is working now with direct backend and without disabling the content sandbox.

closed 2023-04-16 by schulzh