elFarto/nvidia-vaapi-driver #197: Screen OFF on fullscreen on "direct backend"

ManuLinares, 2023-04-06:
I managed to test a video that when playing fullscreen, always blacks the monitor for a few seconds at a specific time at any given resolution.

This video https://youtu.be/PsZszv4qHu4?t=619 at 10min 20sec.

Captured logs with `NVD_LOG=1 firefox`
https://pastebin.com/c3ZE0JTe

System Arch linux:

kernel 6.2.9-1
nvidia-dkms 530.41.03-1
libva-nvidia-driver 0.0.9-1
firefox 111.0.1-1.1

GPU: NVIDIA GA104 [GeForce RTX 3070]
the video codec on "stats for nerds" is `av01.0.08M.08 (399) / opus (251)`

elFarto, 2023-04-06:
That's bizarre. The driver has no knowledge of when the playback is fullscreen or not so I'm not sure I can do anything about it. Could you try using one of the Firefox plugins that lets you force the video codec and see if it happens with VP9 or H264?

ManuLinares, 2023-04-06:
I couldn't reproduce today, at least with that video. I'll log everything when I can reproduce again.

ManuLinares, 2023-04-06:
Ok, weird. on firefox I have an extension that reduces the brightness of pages.

https://addons.mozilla.org/en-US/firefox/addon/adjust-screen-brightness/

When I set it on 80% (yes 80%) I can reproduce this bug, any video codec.

elFarto, 2023-04-07:
I can't seem to reproduce this on my machine, but I've had to revert to my old Geforce 760, on the older 470 drivers for the moment.

elFarto, 2023-04-16:
I've retested this with my 1060, with the screen brightness plugin set to 80% and I can't reproduce it.

As far as I'm aware, the video decode engine can't do anything to influence the monitor, so I doubt there's anything I can do to fix the issue. This is such an obscure bug I'm not sure I can guess where the issue lies. I can't see any references to DPMS in Firefox so I doubt the problem is there.

Is there anything appearing the system log (dmesg/journalctl) when this occurs?

closed 2023-06-16 by elFarto