elFarto/nvidia-vaapi-driver #198: nvidia-drm.modeset=1 OR nvidia_drm.modeset=1

newk0001, 2023-04-08:
The README.md file says this parameter should be nvidia-drm.modeset=1. However, I've seen at least one other place that says it should be nvidia_drm.modeset=1 (https://wiki.archlinux.org/title/NVIDIA) under section 1.3 DRM kernel modesetting states, "To enable DRM ([Direct Rendering Manager](https://en.wikipedia.org/wiki/Direct_Rendering_Manager)) [kernel mode setting](https://wiki.archlinux.org/title/Kernel_mode_setting), add the nvidia_drm.modeset=1 [kernel parameter](https://wiki.archlinux.org/title/Kernel_parameter)." So, my issue is, which is it supposed to be; nvidia-drm.modeset=1 OR nvidia_drm.modeset=1?

elFarto, 2023-04-09:
I have `nvidia-drm.modeset=1` set, and it's working for me. Kernel parameters for loaded modules should have the module name first, and the nvidia modules contain `-` not `_`. It's possible the kernel is doing some fuzzy matching on the module name. Interestingly the NVIDIA documentation also mentions `nvidia_drm` as the module name, so that's likely where the confusion comes from.

birdie-github, 2023-04-10:
You should use a dash but an underscore could probably work as well.

newk0001, 2023-04-14:
@elFarto and @birdie-github  Okay, thank you both for the clarification. I have tried with both variants of the kernel parameter and it does seem to work both ways (Whether you use a `-` or an `_` in the module name for the kernel parameter). I have proposed an edit to the corresponding ArchWiki page that the documentation be changed to omit the portion about setting the kernel parameter to `nvidia_drm.modeset=1` and updating it to the correct kernel parameter of `nvidia-drm.modeset=1`. Thank you both for your time and your responses, I appreciate it. Have a great day.

closed 2023-04-15 by elFarto

Golfkornphisit, 2023-08-13:
sudo cat /sys/module/nvidia_drm/parameters/modeset

Y OR N