elFarto/nvidia-vaapi-driver #199: nvidia-vaapi-driver-git missing from the AUR

tatsumara, 2023-04-13:
just noticed nvidia-vaapi-driver-git isn't in the AUR anymore, and since i've had some problems with libva-nvidia-driver in the past i was wondering if there was a specific reason for this?

elFarto, 2023-04-14:
Those packages are not maintained by me, so I can't say why they have disappeared.

clsferguson, 2023-04-15:
I have been trying to install it on ubuntu 22.04 using apt. This is the output. 
`$ sudo apt install nvidia-vaapi-driver
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package nvidia-vaapi-driver
`

VitalyAnkh, 2023-04-15:
@tatsumara The package is not missing. It's named libva-nvidia-driver-git following the naming conventions and lies [here](https://aur.archlinux.org/packages/libva-nvidia-driver-git).

tatsumara, 2023-04-15:
so it's been simply renamed? good to know

VitalyAnkh, 2023-04-15:
Yep, it's simply renamed.

elFarto, 2023-04-15:
> I have been trying to install it on ubuntu 22.04 using apt. This is the output.

This issue is about the Arch package, not an Ubuntu one. However, it seems the driver hasn't been packaged for that version of Ubuntu, [here's](https://packages.ubuntu.com/search?keywords=nvidia-vaapi-driver+&searchon=names) all the packages that have been created.

closed 2023-04-15 by tatsumara