elFarto/nvidia-vaapi-driver #201: All codecs are hardware accelerated except AV1

trofosila, 2023-04-22:
On Fedora 38, Gnome, Wayland session.

Configuration:
Firefox 112
nvidia-vaapi-driver 0.0.9
akmod-nvidia 530.41
libva 2.18

Environment setup:
```
export MOZ_DISABLE_RDD_SANDBOX=1
export NVD_BACKEND=direct
export EGL_PLATFORM=wayland
```

Firefox configuration:
```
media.ffmpeg.vaapi.enabled: true
media.rdd-ffmpeg.enabled: true
media.av1.enabled: true
gfx.x11-egl.force-enabled: false
widget.dmabuf.force-enabled: true
```

GPU: RTX A2000 https://www.nvidia.com/content/dam/en-zz/Solutions/design-visualization/rtx-a2000/nvidia-rtx-a2000-datasheet.pdf

vainfo
```
Trying display: wayland
libva info: VA-API version 1.18.0
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.18 (libva 2.18.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
```

On VP9 (default for me on youtube) I can confirm it uses hardware acceleration (high GPU activity in `invidia-smi`, `C+G` flag for `firefox`).

If (using `enhanced-h264ify` extension) I disable h264, VP8 and VP9  (only AV1 should be enabled) video is not rendered: "Your browser can't play this video."

trofosila, 2023-04-22:
I get this log if I do `NVD_LOG=1 firefox`:
```
[ERROR glean_core] Error setting metrics feature config: Json(Error("EOF while parsing a value", line: 1, column: 0))
libva info: VA-API version 1.18.0
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      2897.123014154 [20398-20411] ../src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      2897.123017764 [20398-20411] ../src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2897.123021634 [20398-20411] ../src/vabackend.c:2200       __vaDriverInit_1_0 Selecting Direct backend
      2897.127324467 [20398-20411] ../src/direct/nv-driver.c: 217            init_nvdriver Initing nvdriver...
      2897.127331407 [20398-20411] ../src/direct/nv-driver.c: 222            init_nvdriver Got dev info: 2b00 1 2 6
      2897.127389347 [20398-20411] ../src/direct/nv-driver.c: 283            init_nvdriver NVIDIA kernel driver version: 530.41.03
libva info: va_openDriver() returns 0
      2897.168264929 [20398-20411] ../src/vabackend.c:1408      nvQueryImageFormats In nvQueryImageFormats
      2897.186328699 [20398-20411] ../src/vabackend.c: 647           nvCreateConfig got profile: 6 with 0 attributes
      2897.186339649 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      2897.186343629 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7f16535a3040 8
      2897.187551324 [20398-20411] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2897.187562614 [20398-20411] ../src/vabackend.c: 647           nvCreateConfig got profile: 7 with 0 attributes
      2897.187566464 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      2897.187568324 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7f16535a3280 8
      2897.188681010 [20398-20411] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2897.188691330 [20398-20411] ../src/vabackend.c: 647           nvCreateConfig got profile: 13 with 0 attributes
      2897.188695590 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      2897.188698360 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7f16535a31c0 8
      2897.189677016 [20398-20411] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2897.189685716 [20398-20411] ../src/vabackend.c: 647           nvCreateConfig got profile: 14 with 0 attributes
      2897.189689486 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 9 (8) (nil) 0
      2897.189691656 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 9 (8) 0x7f16535a3100 8
      2897.190657642 [20398-20411] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2897.190666002 [20398-20411] ../src/vabackend.c: 647           nvCreateConfig got profile: 19 with 0 attributes
      2897.190669372 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (8) (nil) 0
      2897.190672302 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (8) 0x7f16535a31c0 8
      2897.191610059 [20398-20411] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      2897.191618489 [20398-20411] ../src/vabackend.c: 647           nvCreateConfig got profile: 32 with 0 attributes
      2897.191621049 [20398-20411] ../src/vabackend.c: 710           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
      2897.191625059 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 11 (8) (nil) 0
      2897.191627979 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 11 (8) 0x7f16535a3b80 8
      2897.192566615 [20398-20411] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      2897.192573715 [20398-20411] ../src/vabackend.c: 647           nvCreateConfig got profile: 21 with 0 attributes
      2897.192576085 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (10) (nil) 0
      2897.192577565 [20398-20411] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (10) 0x7f16535a3040 8
      2897.193503481 [20398-20411] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      2897.193722451 [20398-20411] ../src/vabackend.c:2075              nvTerminate Terminating 0x7f167521e020
      2897.193771990 [20398-20411] ../src/vabackend.c:2089              nvTerminate Now have 0 (0 max) instances
libva info: VA-API version 1.18.0
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      2897.702473523 [20398-20453] ../src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      2897.702477353 [20398-20453] ../src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2897.702479653 [20398-20453] ../src/vabackend.c:2200       __vaDriverInit_1_0 Selecting Direct backend
      2897.702501153 [20398-20453] ../src/direct/nv-driver.c: 217            init_nvdriver Initing nvdriver...
[... cut ...]
      2898.229930233 [20398-20506] ../src/vabackend.c: 422          resolveSurfaces Surface 1 exported
      2898.244580646 [20398-20470] ../src/vabackend.c:1310             nvEndPicture Decoded frame successfully to idx: 9 (0x7f165368f480)
      2898.244598586 [20398-20506] ../src/vabackend.c: 398          resolveSurfaces Reading from queue: 13 14
      2898.244605366 [20398-20506] ../src/vabackend.c: 414          resolveSurfaces Mapping surface 9
      2898.244608506 [20398-20470] ../src/vabackend.c:2052    nvExportSurfaceHandle Exporting surface: 9 (0x7f165368f480)

```

elFarto, 2023-04-22:
I don't see anything wrong with your setup, but Firefox doesn't seem to even try using AV1, even though it probes it successfully.

Can you try playing [this](https://raw.githubusercontent.com/SPBTV/video_av1_samples/master/spbtv_sample_bipbop_av1_960x540_25fps.mp4) video back and see what it logs?

trofosila, 2023-04-22:
Video plays successfully. This is an excerpt of the logs:
```
[ERROR glean_core] Error setting metrics feature config: Json(Error("EOF while parsing a value", line: 1, column: 0))
libva info: VA-API version 1.18.0
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      4090.747918115 [27865-27886] ../src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4090.747921945 [27865-27886] ../src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
      4090.747923895 [27865-27886] ../src/vabackend.c:2200       __vaDriverInit_1_0 Selecting Direct backend
      4090.751556017 [27865-27886] ../src/direct/nv-driver.c: 217            init_nvdriver Initing nvdriver...
      4090.751561937 [27865-27886] ../src/direct/nv-driver.c: 222            init_nvdriver Got dev info: 2b00 1 2 6
      4090.751611838 [27865-27886] ../src/direct/nv-driver.c: 283            init_nvdriver NVIDIA kernel driver version: 530.41.03
libva info: va_openDriver() returns 0
      4090.800324105 [27865-27886] ../src/vabackend.c:1408      nvQueryImageFormats In nvQueryImageFormats
      4090.818262239 [27865-27886] ../src/vabackend.c: 647           nvCreateConfig got profile: 6 with 0 attributes
      4090.818268669 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      4090.818273519 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7f1a590fa040 8
      4090.819305372 [27865-27886] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4090.819316743 [27865-27886] ../src/vabackend.c: 647           nvCreateConfig got profile: 7 with 0 attributes
      4090.819321503 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      4090.819325023 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7f1a590fa7c0 8
      4090.820697464 [27865-27886] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4090.820708404 [27865-27886] ../src/vabackend.c: 647           nvCreateConfig got profile: 13 with 0 attributes
      4090.820713354 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      4090.820717394 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7f1a590fa1c0 8
      4090.822486204 [27865-27886] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4090.822492714 [27865-27886] ../src/vabackend.c: 647           nvCreateConfig got profile: 14 with 0 attributes
      4090.822495214 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 9 (8) (nil) 0
      4090.822497164 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 9 (8) 0x7f1a590fa040 8
      4090.823600519 [27865-27886] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      4090.823607439 [27865-27886] ../src/vabackend.c: 647           nvCreateConfig got profile: 19 with 0 attributes
      4090.823609889 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (8) (nil) 0
      4090.823611909 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (8) 0x7f1a590fa340 8
      4090.824592381 [27865-27886] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      4090.824599241 [27865-27886] ../src/vabackend.c: 647           nvCreateConfig got profile: 32 with 0 attributes
      4090.824601211 [27865-27886] ../src/vabackend.c: 710           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
      4090.824604932 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 11 (8) (nil) 0
      4090.824607792 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 11 (8) 0x7f1a590fa040 8
      4090.825590774 [27865-27886] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      4090.825598084 [27865-27886] ../src/vabackend.c: 647           nvCreateConfig got profile: 21 with 0 attributes
      4090.825601524 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (10) (nil) 0
      4090.825604224 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (10) 0x7f1a590fa1c0 8
      4090.826567746 [27865-27886] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      4090.843847745 [27865-27886] ../src/vabackend.c: 647           nvCreateConfig got profile: 32 with 0 attributes
      4090.843853475 [27865-27886] ../src/vabackend.c: 710           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
      4090.843856935 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 11 (8) (nil) 0
      4090.843858345 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 11 (8) 0x7f1a590fe1c0 8
      4090.844804246 [27865-27886] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      4090.844810326 [27865-27886] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 11 (8) (nil) 32538
[... cut ...]
      4090.884371517 [27865-27886] ../src/vabackend.c:1310             nvEndPicture Decoded frame successfully to idx: 3 (0x7f1a58ed2c00)
      4090.884387107 [27865-27900] ../src/vabackend.c: 398          resolveSurfaces Reading from queue: 3 4
      4090.884393907 [27865-27900] ../src/vabackend.c: 414          resolveSurfaces Mapping surface 3
      4090.884388587 [27865-27886] ../src/vabackend.c:2052    nvExportSurfaceHandle Exporting surface: 3 (0x7f1a58ed2c00)
      4090.884419558 [27865-27886] ../src/direct/direct-export-buf.c: 198 direct_allocateBackingImage Allocating BackingImages: 0x7f1a588fd7e0 960x540
```

trofosila, 2023-04-22:
I wonder if I'm missing some codecs. This is what I have installed:
```
dnf list --installed | grep gstreamer
PackageKit-gstreamer-plugin.x86_64                1.2.6-6.fc38                       @fedora                                           
gstreamer1.x86_64                                 1.22.2-1.fc38                      @updates                                          
gstreamer1-plugin-openh264.x86_64                 1.22.1-1.fc38                      @fedora-cisco-openh264                            
gstreamer1-plugins-bad-free.x86_64                1.22.2-1.fc38                      @updates                                          
gstreamer1-plugins-bad-free-fluidsynth.x86_64     1.22.2-1.fc38                      @updates                                          
gstreamer1-plugins-bad-free-wildmidi.x86_64       1.22.2-1.fc38                      @updates                                          
gstreamer1-plugins-bad-free-zbar.x86_64           1.22.2-1.fc38                      @updates                                          
gstreamer1-plugins-bad-freeworld.x86_64           1:1.22.1-1.fc38                    @rpmfusion-free                                   
gstreamer1-plugins-base.x86_64                    1.22.2-1.fc38                      @updates                                          
gstreamer1-plugins-good.x86_64                    1.22.2-1.fc38                      @updates                                          
gstreamer1-plugins-good-extras.x86_64             1.22.2-1.fc38                      @updates                                          
gstreamer1-plugins-good-gtk.x86_64                1.22.2-1.fc38                      @updates                                          
gstreamer1-plugins-good-qt.x86_64                 1.22.2-1.fc38                      @updates                                          
gstreamer1-plugins-ugly.x86_64                    1:1.22.1-1.fc38                    @rpmfusion-free                                   
gstreamer1-plugins-ugly-free.x86_64               1.22.2-1.fc38                      @updates                                          
pipewire-gstreamer.x86_64                         0.3.69-2.fc38                      @updates
```

elFarto, 2023-04-22:
I'm not sure what's going on here. Firefox has all the video codecs built in (it has it's own copy of FFMPEG) so it doesn't rely on what codecs you have installed. And it can clearly play back AV1 videos so the driver is working.

You could try running Firefox with `MOZ_LOG="PlatformDecoderModule:5"` set and see what it prints out. It might be an issue with FFMPEG or FFVPX. There are a couple of `about:config` options for controlling those that you could fiddle with, `media.ffvpx.enabled` and `media.ffmpeg.enabled`.

trofosila, 2023-04-22:
Thank you very much for putting the effort. This is the logging I'm getting:
```
MOZ_LOG="PlatformDecoderModule:5" firefox 
[ERROR glean_core] Error setting metrics feature config: Json(Error("EOF while parsing a value", line: 1, column: 0))
[Child 27582: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 27582: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 27582: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 27582: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 27582: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 27582: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 27582: Main Thread]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[RDD 28154: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[Utility 28155: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Utility process
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/avc'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[RDD 28154: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'video/theora'
[RDD 28154: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'video/theora'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[RDD 28154: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[RDD 28154: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/vorbis'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/vorbis'
[RDD 28154: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/vorbis'
[RDD 28154: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/vorbis'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/flac'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/flac'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/x-wav'
[RDD 28154: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/x-wav'
[RDD 28154: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/x-wav'
[RDD 28154: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/x-wav'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/avc'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[Utility 28155: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[... cut ...]
[Child 27582: MediaPDecoder #1]: V/PlatformDecoderModule AudioTrimmer[7fc300dad7e0] ::PrepareTrimmers: sample[534058,557278] no trimming information
[Utility 28155: MediaPDecoder #4]: V/PlatformDecoderModule ProcessDecode: mDuration=23220µs ; mTime=534058µs ; mTimecode=534058µs
[Utility 28155: MediaPDecoder #4]: V/PlatformDecoderModule ProcessDecode: mDuration=23220µs ; mTime=534058µs ; mTimecode=534058µs
[Child 27582: MediaPDecoder #2]: V/PlatformDecoderModule AudioTrimmer[7fc300dad7e0] ::HandleDecodedResult: sample[534058,557278] (decoded[534058,557277] no trimming needed
```

elFarto, 2023-04-22:
Was that log from the working AV1 video, or the broken Youtube one?

trofosila, 2023-04-22:
Logging is from the broke youtube ones.

This is logging from the mp4 link you sent:
```
MOZ_LOG="PlatformDecoderModule:5" firefox
[ERROR glean_core] Error setting metrics feature config: Json(Error("EOF while parsing a value", line: 1, column: 0))
[Parent 51592: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Chrome process
[Parent 51592: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/avc'
[Parent 51592: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[Parent 51592: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[Child 51854: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in Content process
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: Main Thread]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/avc
[Child 51854: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/av1
[Child 51854: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/av1
[Child 51854: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 51854: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 51854: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 51854: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder rejects requested type audio/mp4a-latm
[Child 51854: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox Utility Generic decoder supports requested type audio/mp4a-latm
[Child 51854: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/av1
[Child 51854: MediaSupervisor #1]: D/PlatformDecoderModule Sandbox RDD decoder supports requested type video/av1
[RDD 52442: Main Thread]: D/PlatformDecoderModule PDMInitializer, Init PDMs in RDD process
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/avc'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/avc'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp9'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/vp8'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'video/av1'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/theora'
[RDD 52442: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'video/theora'
[RDD 52442: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'video/theora'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mp4a-latm'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder supports requested type 'audio/mpeg'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/opus'
[RDD 52442: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[RDD 52442: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/opus'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/vorbis'
[RDD 52442: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'audio/vorbis'
[RDD 52442: Main Thread]: D/PlatformDecoderModule Agnostic decoder supports requested type 'audio/vorbis'
[... cut ...]
[av1 @ 0x7fca3df62000] Slice 3 data buffer (131 bytes) is 0x60.
[av1 @ 0x7fca3df62000] Slice 4 param buffer (40 bytes) is 0x61.
[av1 @ 0x7fca3df62000] Slice 4 data buffer (131 bytes) is 0x62.
[av1 @ 0x7fca3df62000] Slice 5 param buffer (40 bytes) is 0x63.
[av1 @ 0x7fca3df62000] Slice 5 data buffer (131 bytes) is 0x64.
[av1 @ 0x7fca3df62000] Slice 6 param buffer (40 bytes) is 0x65.
[av1 @ 0x7fca3df62000] Slice 6 data buffer (131 bytes) is 0x66.
[av1 @ 0x7fca3df62000] Slice 7 param buffer (40 bytes) is 0x67.
```

elFarto, 2023-04-22:
That looks very much like a Firefox bug. It just seems to be ignoring VA-API when playing that particular video. I can only recommend you raise this on Firefox's bug tracker to see what they say about it. There's not much I can do about it I'm afraid.

trofosila, 2023-04-22:
Thanks anyway for your patience. I'll see what I can do on Firefox side.

trofosila, 2023-04-22:
Hi @elFarto, I'm just back to thank you for the work you put in this project and to confirm everything works as expected.

Indeed there was nothing wrong in my configuration but I was expecting all youtube videos will have a AV1 version (and apparently they don't). So, I was trying time and time again to play videos that didn't have  a AV1 version (and I was disabling all other codecs using the `enhanced-h264ify` extension). That's why it now makes sense firefox was not even trying it.

Sorry for having wasted your time, hope at least will save you troubleshooting time in the future for people with the same _issue_.

elFarto, 2023-04-24:
Ah that would explain why it never attempted to initialise VA-API, it had nothing to play back.

closed 2023-04-24 by elFarto