elFarto/nvidia-vaapi-driver #202: have anyone got this to work in proxmox lxc container?

poohzaza166, 2023-04-27:


elFarto, 2023-04-27:
Can you run vainfo with the environment variable `NVD_LOG=1`, and paste the output here?

poohzaza166, 2023-04-28:
`error: can't connect to X server!
libva info: VA-API version 1.15.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     54006.512078197 [93996-93996] ../src/vabackend.c:2169       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
     54006.512084159 [93996-93996] ../src/vabackend.c:2178       __vaDriverInit_1_0 Now have 0 (0 max) instances
     54006.512086984 [93996-93996] ../src/vabackend.c:2201       __vaDriverInit_1_0 Selecting EGL backend
     54006.543950564 [93996-93996] ../src/backend-common.c:  13 checkModesetParameterFromFd ERROR: This driver requires the nvidia_drm.modeset kernel module parameter set to 1
libva error: /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit`

elFarto, 2023-04-28:
You need to set the `nvidia_drm.modeset` parameter for this driver to work correctly.

poohzaza166, 2023-04-30:
thanks i did that now vainfo return results

elFarto, 2023-04-30:
I'm not entire sure that this driver is compatible with Jellyfin. Does Jellyfin use VA-API?

poohzaza166, 2023-04-30:
jellyfin dose have VA-API as backend support but for amdgpu and intel igpu as far as i am aware this driver is not officially supported

elFarto, 2023-05-01:
While it does seem that Jellyfin supports VA-API, it also supports using NVDEC/NVENC directly. I would recommend that you use NVDEC direct rather than attempting to use it via this library.

poohzaza166, 2023-05-02:
u see it a gt1030

closed 2023-06-16 by elFarto