elFarto/nvidia-vaapi-driver #205: Switching to TTY mode causes Firefox process to crash

Monsterovich, 2023-05-09:
I have RTX 3060 Ti with NVidia driver 530.41.03 installed.

When I go to the text console (ex. CTRL+ALT+F2) and if decoding is performed, the browser crashes. If I minimize the browser window, decoding doesn't happen and therefore everything works.

I use direct backend because egl back doesn't work.

`NVD_BACKEND=direct NVD_LOG=1 firefox > vaapi.log`

[vaapi.log](https://github.com/elFarto/nvidia-vaapi-driver/files/11435510/vaapi.log)

If you don't use the GPU for decoding, everything works - the browser do not crash.

elFarto, 2023-05-10:
I wasn't able to get it to crash on my machine, although Firefox does stop using VA-API when I switch back so something isn't right. Can you repeat the crash with this set aswell `MOZ_LOG="PlatformDecoderModule:5 Dmabuf:5"`?

Monsterovich, 2023-05-10:
> I wasn't able to get it to crash on my machine, although Firefox does stop using VA-API when I switch back so something isn't right. Can you repeat the crash with this set aswell `MOZ_LOG="PlatformDecoderModule:5 Dmabuf:5"`?

This fixes the crash, what does this parameter do?

elFarto, 2023-05-11:
The options by themselve won't (shouldn't at least) fix anything. If you can capture the output with that MOZ_LOG option set and post it here, I can try and investigate it.

elFarto, 2023-05-11:
From what I can tell Firefox has stopped using this driver by the end and has fallen back to CPU decoding, so I'm not entirely sure the crash is the drivers fault (the last line logged by the driver is it exiting normally).

One odd thing is there is a lot of frames being held by Firefox, far more than should be needed. It's like its not releasing them when the VT is changed and running out of space. I would raise an issue with Firefox itself, as see if they pin point where it's crashing.

Monsterovich, 2023-05-11:
> One odd thing is there is a lot of frames being held by Firefox, far more than should be needed. It's like its not releasing them when the VT is changed and running out of space. I would raise an issue with Firefox itself, as see if they pin point where it's crashing.

Shouldn't the driver return an empty framebuffer (dummy frames)? I don't see the point of decoding frames (especially on CPU) when the driver is not initialized and you're in TTY mode.

elFarto, 2023-05-11:
Firefox is the one decode frames on the CPU, not this driver. We only decode frames on the GPU. The driver also has no idea what's going on with the rest of the system so we wouldn't know when to do this. From the logs you've posted I don't see any errors being logged from any calls that we're making, so there's not much we can do to even detect it.

Monsterovich, 2023-05-11:
> The driver also has no idea what's going on with the rest of the system so we wouldn't know when to do this.

Just like Firefox has absolutely no idea what the user is doing on the system. The user goes into TTY mode, but Firefox keeps sending frames to the VA-API driver and when that fails it switches back to CPU decoding mode - this mechanism is rather implemented only to switch to CPU mode once if there's an error in the VA-API driver and there's no functionality to maintain switching between graphics and non-graphics modes.

Monsterovich, 2023-05-11:
@elFarto Why deinitialize the driver when switching to TTY mode at all, can't this driver be used in a non-graphical environment?

elFarto, 2023-05-11:
Firefox is the one that deinitialises the driver and switches to CPU decoding, it's not something we have control over. In theory it can be used in a TTY environment, but this driver is only intended to be used from Firefox which does need a graphical environment (I'm not sure if there's a headless version of Firefox, but I'm not sure you'd need a GPU video decoder for that).

I have noticed a couple of errors in that log file after combing through it a bit more, but they're from Firefox, not the driver:
```
[vp9 @ 0x7fdbb6f79e00] get_buffer() failed
[vp9 @ 0x7fdbb6f79e00] thread_get_buffer() failed
[RDD 137847: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG: avcodec_send_packet error: Cannot allocate memory
```
It appears that Firefox isn't releasing frames while you're on the TTY, you can see the calls to `ReleaseDMABuf` stop then reappear after you switch back. But it's still continuing to decode new frames, and we only have so many buffers to put the decoded frame in.

As for why Intel's driver can handle this, that's likely because it's written by professionals who have documentation on how the chip works, rather than this driver that's written by a random person on the Internet :smiley:.

A somewhat more serious guess would be that the issue is more to do with the NVIDIA graphics driver not rendering anything while it's switched to a TTY causing Firefox to get backed up. The UI of Firefox stops, but the decoding continues.

But that's just a guess, there are many pieces in use while decoding videos and it's difficult to say what can happen if you rip the display away from the one at the end.

closed 2023-06-16 by elFarto