elFarto/nvidia-vaapi-driver #206: Driver (mostly) works under Ubuntu 20 LTS?

eddie-LA, 2023-05-10:
TL;DR: I get this in logs when trying to play some YT videos in Firefox. Video codec is AVC. Only this codec does not work and breaks te video.
```
[Child 7743, MediaDecoderStateMachine #5] WARNING: Decoder=7f7c05ef6700 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005) - Error no decoder found for audio/mp4a-latm: file /build/firefox-DD9c5S/firefox-112.0.2+build1/dom/media/MediaDecoderStateMachineBase.cpp:164
май 10 15:36:40 eddie-MS-7C91 gnome-shell[7743]: [Child 7743, MediaDecoderStateMachine #5] WARNING: Decoder=7f7c05ef6d00 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005) - Error no decoder found for audio/mp4a-latm: file /build/firefox-DD9c5S/firefox-112.0.2+build1/dom/media/MediaDecoderStateMachineBase.cpp:164
май 10 15:36:40 eddie-MS-7C91 gnome-shell[7743]: [Child 7743, MediaDecoderStateMachine #5] WARNING: Decoder=7f7c085f9900 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005) - Error no decoder found for audio/mp4a-latm: file /build/firefox-DD9c5S/firefox-112.0.2+build1/dom/media/MediaDecoderStateMachineBase.cpp:164
май 10 15:36:40 eddie-MS-7C91 gnome-shell[7743]: [Child 7743, MediaDecoderStateMachine #5] WARNING: Decoder=7f7c0fa49d00 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005) - Error no decoder found for audio/mp4a-latm: file /build/firefox-DD9c5S/firefox-112.0.2+build1/dom/media/MediaDecoderStateMachineBase.cpp:164
май 10 15:36:40 eddie-MS-7C91 gnome-shell[7743]: [Child 7743, MediaDecoderStateMachine #5] WARNING: Decoder=7f7bf896f600 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005) - Error no decoder found for video/avc: file /build/firefox-DD9c5S/firefox-112.0.2+build1/dom/media/MediaDecoderStateMachineBase.cpp:164
май 10 15:36:42 eddie-MS-7C91 gnome-shell[7743]: [Child 7743, MediaDecoderStateMachine #5] WARNING: Decoder=7f7c0729ba00 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005) - Error no decoder found for video/avc: file /build/firefox-DD9c5S/firefox-112.0.2+build1/dom/media/MediaDecoderStateMachineBase.cpp:164
```
I am on Ubuntu 20.o4 LTS, I compiled the driver manually successfully. I used latest version of meson from pip and compiled libva manually as it was slightly older than needed. 

I also get this when I try `vainfo`, so I can't look that up:
```
vainfo: symbol lookup error: /lib/x86_64-linux-gnu/libva-x11.so.2: undefined symbol: va_fool_postp
```
I have seen Steam users complaining from the same symbol lookup error but they solve it by installing stock libva-x11-2. Mine is already stock. 
Any help in solving either?

eddie-LA, 2023-05-11:
Sure, here it is:
```
linux-vdso.so.1 (0x00007ffd02c28000)
libva.so.2 => /lib/x86_64-linux-gnu/libva.so.2 (0x00007f5cc4d33000)
libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f5cc4bf6000)
libXext.so.6 => /lib/x86_64-linux-gnu/libXext.so.6 (0x00007f5cc4be1000)
libXfixes.so.3 => /lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f5cc4bd9000)
libdrm.so.2 => /lib/x86_64-linux-gnu/libdrm.so.2 (0x00007f5cc4bc3000)
libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f5cc4bbd000)
libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f5cc49c9000)
libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f5cc499f000)
/lib64/ld-linux-x86-64.so.2 (0x00007f5cc4d8f000)
libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f5cc4999000)
libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f5cc4991000)
libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f5cc4977000)
```

elFarto, 2023-05-11:
I think I see the issue. You say you rebuilt libva, did you include the option to enable the X11 backend? If not you would be using a libva.so you built, and a stock libva-x11.so, which wouldn't be compatible. Can you rebuild libva with the x11 backend enabled?

closed 2023-06-16 by elFarto

xeleven11, 2024-08-24:
i am having same issue. is rebuilding installing from source again and how can i enable x11 backend?