elFarto/nvidia-vaapi-driver #207: CUDA ERROR 'operation not supported'

akarsh1995, 2023-05-15:
I observed the nvidia-smi process in the terminal while running firefox. 

Compute against firefox process starts but it vanishes as soon as it starts.

**nvidia-smi output:** 

```
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 530.41.03              Driver Version: 530.41.03    CUDA Version: 12.1     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                  Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf            Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GTX 1650 S...    Off| 00000000:01:00.0  On |                  N/A |
| 35%   42C    P8                9W / 100W|    282MiB /  4096MiB |      5%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+

+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A       573      G   /usr/lib/Xorg                               145MiB |
|    0   N/A  N/A      1203      G   alacritty                                     8MiB |
|    0   N/A  N/A      2192      G   alacritty                                     8MiB |
|    0   N/A  N/A      7176      G   /usr/lib/firefox/firefox                    116MiB |
+---------------------------------------------------------------------------------------+

```

**Here is the NV_LOG=1 stdout attached.**
Please find the error in the logs.

```
       487.456351692 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       487.456360929 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
       487.456364218 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2197       __vaDriverInit_1_0 Selecting EGL backend
       487.456372632 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 147       findGPUIndexFromFd Looking for DRM device index: 0
       487.456381212 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 161       findGPUIndexFromFd Found 1 EGL devices
       487.456406259 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       487.456411991 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 176       findGPUIndexFromFd Found drmDeviceIndex: 0
       487.456415186 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
       487.456433778 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 277         egl_initExporter Driver supports 16-bit surfaces
       487.511256369 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:1408      nvQueryImageFormats In nvQueryImageFormats
       487.511450033 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2075              nvTerminate Terminating 0x7f8865e22980
       487.511457037 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c:  61      egl_releaseExporter Releasing exporter, 0 outstanding frames
       487.511459590 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c:  78      egl_releaseExporter Done releasing frames
       487.511462181 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2089              nvTerminate Now have 0 (0 max) instances
       487.530423944 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       487.530434553 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
       487.530439478 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2197       __vaDriverInit_1_0 Selecting EGL backend
       487.530448205 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 147       findGPUIndexFromFd Looking for DRM device index: 0
       487.530456536 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 161       findGPUIndexFromFd Found 1 EGL devices
       487.530480562 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       487.530485696 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 176       findGPUIndexFromFd Found drmDeviceIndex: 0
       487.530488276 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 208       findGPUIndexFromFd Selecting EGLDevice 0
       487.530506154 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 277         egl_initExporter Driver supports 16-bit surfaces
       487.588985837 [6886-7045] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:1408      nvQueryImageFormats In nvQueryImageFormats
       487.589155469 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2075              nvTerminate Terminating 0x7f8865e22980
       487.589165916 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c:  61      egl_releaseExporter Releasing exporter, 0 outstanding frames
       487.589169481 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c:  78      egl_releaseExporter Done releasing frames
       487.589172419 [6886-7047] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2089              nvTerminate Now have 0 (0 max) instances
       507.810164807 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       507.810175943 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
       507.810179556 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2197       __vaDriverInit_1_0 Selecting EGL backend
       507.810187860 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 147       findGPUIndexFromFd Looking for DRM device index: 0
       507.810203388 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 161       findGPUIndexFromFd Found 1 EGL devices
       507.810233864 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
[... 81 lines cut ...]
       507.947684601 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 352 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
       507.947688327 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7f8834a20100)
       507.947691075 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 352 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
       507.947694511 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7f8834a20060)
       507.947697517 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 352 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
       507.947701009 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7f8834a1f7a0)
       507.947703827 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 352 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
       507.947707613 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7f8834a1ff20)
       507.947710609 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c: 352 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
       507.947713892 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7f8834a1f700)
[... cut ...]
       507.947739787 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2075              nvTerminate Terminating 0x7f8865e234c0
       507.947742540 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c:  61      egl_releaseExporter Releasing exporter, 0 outstanding frames
       507.948805213 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/export-buf.c:  78      egl_releaseExporter Done releasing frames
       507.948862567 [6886-7074] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2089              nvTerminate Now have 0 (0 max) instances

```

akarsh1995, 2023-05-16:
I tried setting NVD_BACKEND=direct and h/w acceleration worked. 
Looks like for driver version `530.41.03` or 530 series direct backend seem to work.

elFarto, 2023-05-16:
Yes, the EGL backend has been broken by the NVIDIA drivers for a while now, so the direct backend needs to be used. They say they've got a fix for it and we were expecting it in the 530 series, but it hasn't appeared yet.

closed 2023-05-16 by elFarto