elFarto/nvidia-vaapi-driver #209: Anyone had any luck recently?

larryare, 2023-05-22:
Firefox spits out: `libva error: /usr/lib/dri/nvidia_drv_video.so init failed`. Recommended environment variables ant `about:config` changed.
Running `nvidia` 530.41.03 on rtx3070
`vainfo` output:
```
Trying display: wayland
Trying display: x11
vainfo: VA-API version: 1.18 (libva 2.18.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
```

Staudey, 2023-05-22:
Using the `nvidia-drm.modeset=1` kernel parameter?

elFarto, 2023-05-24:
Can you run firefox with the environment variable `NVD_LOG=1` set, and paste the output here?

larryare, 2023-05-26:
> Using the `nvidia-drm.modeset=1` kernel parameter?

yes

```bash
cat /proc/cmdline
initrd=\amd-ucode.img initrd=\initramfs-linux.img root=PARTUUID=7fe3a247-583d-4864-9ece-d1ca2fe36556 zswap.enabled=0 rootflags=subvol=@ rw rootfstype=btrfs quiet splash nvidia-drm.modeset=1
```

larryare, 2023-05-26:
> Can you run firefox with the environment variable `NVD_LOG=1` set, and paste the output here?

I'm embarrassed right now, was able to fix the issue using NVD_LOG. Thanks.

closed 2023-05-26 by larryare