elFarto/nvidia-vaapi-driver #210: Decoding not working in Firefox on Ubuntu 23.04

Steve-Tech, 2023-06-01:
Hi,

I'm using Firefox 114.0 installed from the [Mozilla PPA](https://launchpad.net/~mozillateam/+archive/ubuntu/ppa) (not snap), nvidia-vaapi-driver was installed via apt, Nvidia driver: 525.105.17.

I've done the `about:config` stuff in the readme, and added this to `/etc/environment`:
```
NVD_BACKEND=direct
MOZ_DISABLE_RDD_SANDBOX=1
EGL_PLATFORM=wayland
LIBVA_DRIVER_NAME=nvidia
```

I've also tried with `media.rdd-ffvpx.enabled` set to true and false, tried with both X11 and Wayland, and briefly tried driver 530 and attempted 510 but that got stuck booting.

`nvidia-drm.modeset=1` was also added to grub.

But YouTube videos still use software decoding, here's the output when running `NVD_LOG=1 firefox`:
```
[GFX1-]: FireTestProcess failed: Failed to spawn child process “/usr/lib/firefox/glxtest” (No such file or directory)

[GFX1-]: glxtest: ManageChildProcess failed

[GFX1-]: No GPUs detected via PCI

[Child 248322, MediaDecoderStateMachine #1] WARNING: Decoder=7fbf89724700 state=DECODING_METADATA Decode metadata failed, shutting down decoder: file /build/firefox-mMMMIM/firefox-114.0+build2/dom/media/MediaDecoderStateMachine.cpp:372
[Child 248322, MediaDecoderStateMachine #1] WARNING: Decoder=7fbf89724700 Decode error: NS_ERROR_DOM_MEDIA_METADATA_ERR (0x806e0006) - static MP4Metadata::ResultAndByteBuffer mozilla::MP4Metadata::Metadata(mozilla::ByteStream *): Cannot parse metadata: file /build/firefox-mMMMIM/firefox-114.0+build2/dom/media/MediaDecoderStateMachineBase.cpp:164
[Child 248322, MediaDecoderStateMachine #1] WARNING: Decoder=7fbf89724400 state=DECODING_METADATA Decode metadata failed, shutting down decoder: file /build/firefox-mMMMIM/firefox-114.0+build2/dom/media/MediaDecoderStateMachine.cpp:372
[Child 248322, MediaDecoderStateMachine #1] WARNING: Decoder=7fbf89724400 Decode error: NS_ERROR_DOM_MEDIA_METADATA_ERR (0x806e0006) - static MP4Metadata::ResultAndByteBuffer mozilla::MP4Metadata::Metadata(mozilla::ByteStream *): Cannot parse metadata: file /build/firefox-mMMMIM/firefox-114.0+build2/dom/media/MediaDecoderStateMachineBase.cpp:164
[Child 248322, MediaDecoderStateMachine #1] WARNING: Decoder=7fbf8a2bc500 state=DECODING_METADATA Decode metadata failed, shutting down decoder: file /build/firefox-mMMMIM/firefox-114.0+build2/dom/media/MediaDecoderStateMachine.cpp:372
[Child 248322, MediaDecoderStateMachine #1] WARNING: Decoder=7fbf8a2bc500 Decode error: NS_ERROR_DOM_MEDIA_METADATA_ERR (0x806e0006) - static MP4Metadata::ResultAndByteBuffer mozilla::MP4Metadata::Metadata(mozilla::ByteStream *): Cannot parse metadata: file /build/firefox-mMMMIM/firefox-114.0+build2/dom/media/MediaDecoderStateMachineBase.cpp:164
[Child 248322, MediaDecoderStateMachine #1] WARNING: Decoder=7fbf89724a00 state=DECODING_METADATA Decode metadata failed, shutting down decoder: file /build/firefox-mMMMIM/firefox-114.0+build2/dom/media/MediaDecoderStateMachine.cpp:372
[Child 248322, MediaDecoderStateMachine #1] WARNING: Decoder=7fbf89724a00 Decode error: NS_ERROR_DOM_MEDIA_METADATA_ERR (0x806e0006) - static MP4Metadata::ResultAndByteBuffer mozilla::MP4Metadata::Metadata(mozilla::ByteStream *): Cannot parse metadata: file /build/firefox-mMMMIM/firefox-114.0+build2/dom/media/MediaDecoderStateMachineBase.cpp:164
```

Interestingly .mp4 files don't have the all the `WARNING`s and the CPU usage isn't as high as YouTube, but Firefox isn't showing up in `nvidia-smi`, and I'm pretty sure it was before (I did have it working earlier, at least on 22.10, I don't know when it broke).

Thanks,
Steve

elFarto, 2023-06-01:
Can you run Firefox with this environment variable set: `NVD_LOG=1`, and post the output here?

Steve-Tech, 2023-06-01:
I thought I did that before, so I added `NVD_LOG=1` to `/etc/environment` and now there's no `WARNING`s but the CPU usage is still really high and no Firefox in `nvidia-smi`:
```
[GFX1-]: FireTestProcess failed: Failed to spawn child process “/usr/lib/firefox/glxtest” (No such file or directory)

[GFX1-]: glxtest: ManageChildProcess failed

[GFX1-]: No GPUs detected via PCI
```

Also here's `vainfo`, so vaapi and `NVD_LOG=1` are working:
```
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       687.728991567 [51653-51653] ../src/vabackend.c:1872       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 0x55c8fcd91060 40
       687.728998510 [51653-51653] ../src/vabackend.c:1894       __vaDriverInit_1_0 Now have 0 (0 max) instances
       687.729002417 [51653-51653] ../src/vabackend.c:1919       __vaDriverInit_1_0 Selecting Direct backend
       687.732183280 [51653-51653] ../src/direct/direct-export-buf.c:  94      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
       687.732189221 [51653-51653] ../src/direct/nv-driver.c: 190            init_nvdriver Initing nvdriver...
       687.732194361 [51653-51653] ../src/direct/nv-driver.c: 195            init_nvdriver Got dev info: 800 1 2 6
       687.732269563 [51653-51653] ../src/direct/nv-driver.c: 256            init_nvdriver NVIDIA kernel driver version: 525.105.17
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      <unknown profile>               :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
       687.794382368 [51653-51653] ../src/vabackend.c:1843              nvTerminate Terminating 0x55c8fcd91060
       687.794446580 [51653-51653] ../src/vabackend.c:1857              nvTerminate Now have 0 (0 max) instances
```

I also checked if there's anything relevant in dmesg and syslog, and there didn't seem to be.

Steve-Tech, 2023-06-02:
mpv also doesn't seem to work.

<details>
<summary>mpv log</summary>
<br>

```
$ NVD_LOG=1 mpv -v --msg-level=all=debug --hwdec=vaapi --gpu-debug --hwdec-codecs=all --vd-lavc-check-hw-profile=no output.mp4 
[cplayer] Command line options: '-v' '--msg-level=all=debug' '--hwdec=vaapi' '--gpu-debug' '--hwdec-codecs=all' '--vd-lavc-check-hw-profile=no' 'output.mp4'
[cplayer] mpv 0.35.1 Copyright © 2000-2023 mpv/MPlayer/mplayer2 projects
[cplayer]  built on UNKNOWN
[cplayer] FFmpeg library versions:
[cplayer]    libavutil       57.28.100
[cplayer]    libavcodec      59.37.100
[cplayer]    libavformat     59.27.100
[cplayer]    libswscale      6.7.100
[cplayer]    libavfilter     8.44.100
[cplayer]    libswresample   4.7.100
[cplayer] FFmpeg version: 5.1.2-3ubuntu1
[cplayer] 
[cplayer] Configuration: meson configure build -Dprefix=/usr -Dbuildtype=plain -Doptimization=plain
[cplayer] List of enabled features: alsa av-channel-layout caca cdda cplayer cplugins cuda-hwaccel cuda-interop dmabuf-interop-gl dmabuf-interop-pl dmabuf-wayland drm drm-is-kms dvdnav egl egl-drm egl-helpers egl-x11 ffmpeg ffnvcodec gbm gl gl-wayland glibc-thread-name glob glob-posix gpl iconv jack javascript jpeg jpegxl lcms2 libarchive libass libavdevice libbluray libdl libm libmpv-shared libplacebo libplacebo-next librt linux-fstatfs lua52 manpage-build memfd_create noexecstack pipewire plain-gl posix pulse rubberband rubberband-3 sdl2 sdl2-audio sdl2-gamepad sdl2-video sixel spirv-cross stdatomic threads uchardet vaapi vaapi-drm vaapi-egl vaapi-libplacebo vaapi-wayland vaapi-x-egl vaapi-x11 vdpau vector vt.h vulkan wayland wayland_protocols_1_24 x11 xv zimg zlib
[global] config path: '' -> '/home/stephen/.config/mpv'
[global] user path: '~~home/' -> '/home/stephen/.config/mpv'
[global] user path: '~~old_home/' -> '/home/stephen/.mpv'
[global] config path: 'encoding-profiles.conf' -/-> '/home/stephen/.config/mpv/encoding-profiles.conf'
[global] config path: 'encoding-profiles.conf' -/-> '/home/stephen/.mpv/encoding-profiles.conf'
[global] config path: 'encoding-profiles.conf' -> '/etc/mpv/encoding-profiles.conf'
[cplayer] Reading config file /etc/mpv/encoding-profiles.conf
[cplayer] Applying profile 'default'...
[global] config path: 'mpv.conf' -/-> '/home/stephen/.config/mpv/mpv.conf'
[global] config path: 'config' -/-> '/home/stephen/.config/mpv/config'
[global] config path: 'mpv.conf' -/-> '/home/stephen/.mpv/mpv.conf'
[global] config path: 'config' -/-> '/home/stephen/.mpv/config'
[global] config path: 'mpv.conf' -> '/etc/mpv/mpv.conf'
[global] config path: 'config' -/-> '/etc/mpv/config'
[cplayer] Reading config file /etc/mpv/mpv.conf
[cplayer] Applying profile 'default'...
[cplayer] Setting option 'hwdec' = 'vaapi' (flags = 4)
[cplayer] Setting option 'v' = '' (flags = 8)
[cplayer] Setting option 'msg-level' = 'all=debug' (flags = 8)
[cplayer] Setting option 'hwdec' = 'vaapi' (flags = 8)
[cplayer] Setting option 'gpu-debug' = '' (flags = 8)
[cplayer] Setting option 'hwdec-codecs' = 'all' (flags = 8)
[cplayer] Setting option 'vd-lavc-check-hw-profile' = 'no' (flags = 8)
[global] config path: 'input.conf' -/-> '/home/stephen/.config/mpv/input.conf'
[global] config path: 'input.conf' -/-> '/home/stephen/.mpv/input.conf'
[global] config path: 'input.conf' -/-> '/etc/mpv/input.conf'
[osc] Loading lua script @osc.lua...
[ytdl_hook] Loading lua script @ytdl_hook.lua...
[stats] Loading lua script @stats.lua...
[console] Loading lua script @console.lua...
[auto_profiles] Loading lua script @auto_profiles.lua...
[osc] loading mp.defaults
[ytdl_hook] loading mp.defaults
[console] loading mp.defaults
[stats] loading mp.defaults
[auto_profiles] loading mp.defaults
[global] config path: 'scripts' -/-> '/home/stephen/.config/mpv/scripts'
[global] config path: 'scripts' -/-> '/home/stephen/.mpv/scripts'
[global] config path: 'scripts' -/-> '/etc/mpv/scripts'
[cplayer] Waiting for scripts...
[osc] loading @osc.lua
[console] loading @console.lua
[auto_profiles] loading @auto_profiles.lua
[ytdl_hook] loading @ytdl_hook.lua
[stats] loading @stats.lua
[auto_profiles] Exiting...
[console] reading options for console 
[global] config path: 'script-opts/console.conf' -/-> '/home/stephen/.config/mpv/script-opts/console.conf'
[... 1313 lines cut ...]
[vo/gpu/x11] Enabling screensaver.
[cplayer] Terminating demuxers...
[cplayer] Done terminating demuxers.
[cplayer] finished playback, success (reason 3)
[cplayer] 
[cplayer] Exiting... (Quit)
[cplayer] Run command: change-list, flags=64, args=[name="shared-script-properties", operation="remove", value="osc-margins"]
[ytdl_hook] Exiting...
[cplayer] Set property: shared-script-properties -> 1
[console] Exiting...
[stats] Exiting...
[osc] Exiting...
[vo/gpu] flushing shader cache
[vo/gpu/libplacebo] Freeing slab of size 8640K
[vo/gpu/libplacebo] Freeing slab of size   12M
[vo/gpu/libplacebo] Freeing slab of size   24M
[vo/gpu/libplacebo] Freeing slab of size   49M
[vo/gpu/libplacebo] Freeing slab of size  256K
[vo/gpu/libplacebo] Freeing slab of size  256K
[vo/gpu/libplacebo] Freeing slab of size 2784K
[vo/gpu/libplacebo] Waiting for remaining commands...
[vo/gpu/x11] uninit ...
```

</details>

elFarto, 2023-06-02:
> I thought I did that before

My apologies, you did. I was looking at it and didn't see any output from the driver so I just assumed you hadn't supplied it.

There does seem to be something wrong with that build of Firefox, as it's not even attempting to initalise VA-API. You could try running Firefox with this aswell `MOZ_LOG="PlatformDecoderModule:5 Dmabuf:5"` to see if that sheds some light on what's happening.

Steve-Tech, 2023-06-02:
> There does seem to be something wrong with that build of Firefox

Seems like it, I just did an `apt upgrade` and it's fixed now haha.

closed 2023-06-02 by Steve-Tech

Vitaly76ASDgnaerkldbnak, 2024-05-06:
> Seems like it, I just did an `apt upgrade` and it's fixed now haha.
> 
> Edit: Thanks for your support though!

What version firefox is it?

Steve-Tech, 2024-05-06:
> What version firefox is it?

I wouldn't know now, but whatever was in Debian testing 11 months ago.

I've also moved to Intel Arc now, so I can't say if it still works for me.