elFarto/nvidia-vaapi-driver #213: Output from vainfo: isNvidiaDrmFd: Invalid driver for DRM device: i915

SebastianDavidson, 2023-06-09:
I have this issue after having installed `libva-nvidia-driver` from the AUR. Starting `vainfo` yields this:
```
$ NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
     75209.366429532 [73594-73594] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
     75209.366463469 [73594-73594] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
     75209.366479802 [73594-73594] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2197       __vaDriverInit_1_0 Selecting EGL backend
     75209.388133298 [73594-73594] ../nvidia-vaapi-driver-0.0.9/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```
This are my /etc/environment settings:
```
MOZ_DISABLE_RDD_SANDBOX=1
EGL_PLATFORM=wayland
LIBVA_DRIVER_NAME=nvidia
```
And these are my graphics cards, taken from doing `lspci -nn | grep '\[03'`.
```
$ lspci -nn | grep '\[03'
00:02.0 VGA compatible controller [0300]: Intel Corporation CoffeeLake-H GT2 [UHD Graphics 630] [8086:3e9b]
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP106M [GeForce GTX 1060 Mobile] [10de:1c20] (rev a1)
```

elFarto, 2023-06-10:
Firstly, you'll need to use the direct backend, with `NVD_BACKEND=direct` as the EGL backend is still broken with the recent NVIDIA drivers.

Secondly, it seems that your Intel card is set as the default one. Can you try running: `NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD129`. That should force it to use the NVIDIA card.

However, you may still have issues with Firefox, as it doesn't like being run on the non-default GPU, and I haven't yet found a way around that. You can try setting the `__EGL_VENDOR_LIBRARY_FILENAMES` environment variable, as detailed in the README. Firefox also has a `MOZ_DRM_DEVICE` environment variable that might help.

SebastianDavidson, 2023-06-10:
> Firstly, you'll need to use the direct backend, with `NVD_BACKEND=direct` as the EGL backend is still broken with the recent NVIDIA drivers.

Ah, sorry, I see now that I missed that in the README, my bad.

Running that command without changing anything, `NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD129` yields the following output:
```
Trying display: drm
      2802.273539657 [6542-6542] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      2802.273573312 [6542-6542] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2802.273597865 [6542-6542] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2197       __vaDriverInit_1_0 Selecting EGL backend
      2802.294494973 [6542-6542] ../nvidia-vaapi-driver-0.0.9/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```
And running the same command but after setting `NVD_BACKEND=direct` in /etc/environment yields a similar results:
```
Trying display: drm
      2883.849139806 [6590-6590] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      2883.849173478 [6590-6590] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2883.849189353 [6590-6590] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2200       __vaDriverInit_1_0 Selecting Direct backend
      2883.870277776 [6590-6590] ../nvidia-vaapi-driver-0.0.9/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

elFarto, 2023-06-10:
That's odd, /dev/dri/renderD129 is normally the secondary device, try it with /dev/dri/renderD128 instead. This would seem to imply that your Wayland compositor is running on the Intel card, which would be were vainfo is picking it up from.

SebastianDavidson, 2023-06-10:
It seems like that works. This is what it yields:
```
Trying display: drm
      6788.851690289 [13713-13713] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      6788.851723517 [13713-13713] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
      6788.851739670 [13713-13713] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2200       __vaDriverInit_1_0 Selecting Direct backend
      6788.872483902 [13713-13713] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 217            init_nvdriver Initing nvdriver...
      6788.872521621 [13713-13713] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 222            init_nvdriver Got dev info: 100 1 0 fe
      6788.872846482 [13713-13713] ../nvidia-vaapi-driver-0.0.9/src/direct/nv-driver.c: 283            init_nvdriver NVIDIA kernel driver version: 530.41.03
vainfo: VA-API version: 1.18 (libva 2.18.2)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      6789.189671974 [13713-13713] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2075              nvTerminate Terminating 0x55b92874b8e0
      6789.189862282 [13713-13713] ../nvidia-vaapi-driver-0.0.9/src/vabackend.c:2089              nvTerminate Now have 0 (0 max) instances
```
Yeah, sorry that I didn't come knowing beforehand that my Wayland compositor was running on my Intel card. I think I didn't set it up properly, then.

elFarto, 2023-06-11:
Sorry, I wasn't clear before. Using `vainfo --display drm ...` forces the device to use. Without that it will ask Wayland for the default device. The order of the devices (card0/1, renderD128/129) is basically random, based on the order the drivers are loaded, so card0/renderD128 isn't necessarily the default device.

Unfortunetly I don't have much Wayland experience.