elFarto/nvidia-vaapi-driver #214: Works great in firefox, fails in beta and nightly

pallaswept, 2023-06-12:
Hi there elFarto, first off thanks for sharing this with us, it's nice to have working hardware acceleration in the browser. I thought that I should share this with you in case something is changing in future versions, so you could be ahead of the curve, but perhaps I'm just doing it wrong.

I've been enjoying this in the current release version of Firefox (113) but had some totally unrelated problems with my browser (missing scroll buttons in the all tabs list, seriously unrelated), and since I used to use ff nightly, I thought maybe I'd see if those problems were fixed in that version, so I installed it and noticed that this driver wasn't functioning. I checked by running nvidia-smi and looking for the 'C' and sure enough, not there. I thought maybe something had changed in nightly so I gave beta a shot and it's not behaving there, either. I'm using the same FF profile in each case, with the necessary environment variables in my /usr/etc/environment file and necessary Firefox settings in a user.js file, so I'm sure my settings are correct. I've been through the issues to see if anything similar has been reported and didn't see anything of the sort other than #182, but I already have the setting applied from that issue, as per your readme.

One thing which immediately struck me was that when running nvidia-smi, in the working release version, I see two entries for firefox, one of which has the 'C+G' marking, and obviously is a separate process for decoding the video. When running either nightly or beta versions, I only see a single entry for firefox in that list, with the 'G' but no 'C'. It's as though firefox isn't even trying to use the decoder at all.

I ran nightly with NVD_LOG=1 as you had requested in other threads, perhaps this information will be of some use:
```

> NVD_LOG=1 firefox-nightly 
[GFX1-]: Wayland protocol error: warning: queue 7efee45042b0 destroyed while proxies still attached:

[GFX1-]: Wayland protocol error:   wl_registry@47 still attached

[fluent] Missing message in locale en-GB: identity-credential-policy-description
[fluent] Missing message in locale en-GB: identity-credential-urlbar-anchor
[fluent] Missing message in locale en-GB: identity-credential-urlbar-anchor
[fluent] Missing message in locale en-GB: identity-credential-urlbar-anchor
[fluent] Missing message in locale en-GB: identity-credential-urlbar-anchor
[fluent] Missing message in locale en-GB: identity-credential-urlbar-anchor
Validation error without device target: No suitable adapter found
Validation error without device target: No suitable adapter found
Validation error without device target: No suitable adapter found
[GFX1-]: Wayland protocol error: warning: queue 7efee4504400 destroyed while proxies still attached:

[GFX1-]: Wayland protocol error:   zwp_linux_dmabuf_v1@53 still attached

> NVD_LOG=1 firefox --allow-downgrade 
warning: queue 0x56173816f4b0 destroyed while proxies still attached:
  wl_registry@16 still attached
kf.i18n: KLocalizedString: Using an empty domain, fix the code. msgid: "Mozilla Firefox" msgid_plural: "" msgctxt: ""
[Child 25133, Main Thread] WARNING: JSWindowActorChild::SendRawMessage (Conduits, ConduitClosed) not sent: !CanSend() || !mManager || !mManager->CanSend(): file /home/abuild/rpmbuild/BUILD/firefox-113.0.2/dom/ipc/jsactor/JSWindowActorChild.cpp:56  
```
Edit: The above warning is thrown some thousands of times, I have removed them for ease of reading.
```
[ERROR glean_core] Error setting metrics feature config: Json(Error("EOF while parsing a value", line: 1, column: 0))
libva info: VA-API version 1.18.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     11388.121957567 [27955-27961] ../src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
     11388.121961437 [27955-27961] ../src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
     11388.121963757 [27955-27961] ../src/vabackend.c:2200       __vaDriverInit_1_0 Selecting Direct backend
     11388.124647378 [27955-27961] ../src/direct/nv-driver.c: 217            init_nvdriver Initing nvdriver...
     11388.124653078 [27955-27961] ../src/direct/nv-driver.c: 222            init_nvdriver Got dev info: 700 1 2 6
     11388.124708518 [27955-27961] ../src/direct/nv-driver.c: 283            init_nvdriver NVIDIA kernel driver version: 525.116.04
libva info: va_openDriver() returns 0
     11388.167472922 [27955-27961] ../src/vabackend.c:1408      nvQueryImageFormats In nvQueryImageFormats
     11388.181213578 [27955-27961] ../src/vabackend.c: 647           nvCreateConfig got profile: 6 with 0 attributes
     11388.181223558 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) 0
     11388.181226958 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7fd1beb77100 8
     11388.181950636 [27955-27961] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     11388.181958516 [27955-27961] ../src/vabackend.c: 647           nvCreateConfig got profile: 7 with 0 attributes
     11388.181962386 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) 0
     11388.181964976 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7fd1beb77100 8
     11388.182686334 [27955-27961] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     11388.182693314 [27955-27961] ../src/vabackend.c: 647           nvCreateConfig got profile: 13 with 0 attributes
     11388.182697044 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) (nil) 0
     11388.182699514 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 4 (8) 0x7fd1beb771c0 8
     11388.183427113 [27955-27961] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     11388.183433943 [27955-27961] ../src/vabackend.c: 647           nvCreateConfig got profile: 14 with 0 attributes
     11388.183437163 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 9 (8) (nil) 0
     11388.183439703 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 9 (8) 0x7fd1beb77100 8
     11388.184287162 [27955-27961] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
     11388.184294312 [27955-27961] ../src/vabackend.c: 647           nvCreateConfig got profile: 19 with 0 attributes
     11388.184297583 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (8) (nil) 0
     11388.184300153 [27955-27961] ../src/vabackend.c:1762 nvQuerySurfaceAttributes with 10 (8) 0x7fd1beb77280 8
     11388.185027721 [27955-27961] ../src/vabackend.c:1835 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
     11388.185034921 [27955-27961] ../src/vabackend.c: 647           nvCreateConfig got profile: 32 with 0 attributes
     11388.185038151 [27955-27961] ../src/vabackend.c: 710           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
[... 166 lines cut ...]
     11388.800493583 [27955-27984] ../src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd1bffb3a60)
     11388.800496153 [27955-27984] ../src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd1bffb39c0)
     11388.800498083 [27955-27984] ../src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd1bffb3ce0)
     11388.800500194 [27955-27984] ../src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd1bffb3920)
     11388.800502094 [27955-27984] ../src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd1bffb37e0)
     11388.800504274 [27955-27984] ../src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd1bffb3c40)
     11388.800505624 [27955-27984] ../src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd1bffb3f60)
     11388.800507394 [27955-27984] ../src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd1bffb3740)
     11388.800509634 [27955-27984] ../src/vabackend.c: 964        nvDestroySurfaces Destroying surface -1 (0x7fd1bffb3ba0)
[... cut ...]
warning: queue 0x56173816f4b0 destroyed while proxies still attached:
  zwp_linux_dmabuf_v1@30 still attached
  
```
In the first instance, running firefox-nightly (116), I see only a single firefox process in nvidia-smi, and hardware decoding appears not to be working.
In the following instance, I see two, and it appears to be working.

If thre's any information I can provide to help you out, please let me know. For now I'll continue using the release version of FF and enjoying the working hardware decode, so thank you :)

elFarto, 2023-06-16:
It's probably worth raising this with Mozilla, as there's nothing I can do about the Wayland errors. The GBM errors look a bit more concerning, I'll need to look into those.

elFarto, 2023-06-20:
I've just tested this on X, and everything seems fine, so it's probably a Wayland related issue.

elFarto, 2023-06-24:
I've finally got around to retesting it under Wayland, and everything seems to work for me. Which Wayland compositor are you using?

pallaswept, 2023-06-26:
> I've finally got around to retesting it under Wayland, and everything seems to work for me. Which Wayland compositor are you using?

That's interesting. I'm using kwin. I have since had a version upgrade from 113 to 114 in vanilla FF, and it's working fine, so I'm starting to wonder if it's something to do with the patches that go into the distro (opensuse) build of FF that sort it out. I plan to try and build a FF version from the nightly tree but with those patches and see how it goes, but perhaps the compositor may be playing a role too. I'll be sure to let you know once I get that working (I'm struggling with problems with the package building toolchain at the moment so I'm a bit delayed)

pallaswept, 2023-07-09:
Just an update, I've just been upgraded to 115 from the distro and that's the same version as the failing beta earlier - and it's working fine. This pretty much confirms that it's something to do with OpenSUSE's packaging of firefox built from source, compared to the pre-built beta/nightly builds from mozilla. They apply quite a few patches, notably one specifying libavcodec versions, so I guess one of those patches makes the difference.

It might be appropriate to close this issue, I'm not entirely sure so I'll leave it up to you for now. If I'm able to get a self-built nightly from Mozilla's nightly sources but with OpenSUSE's patches, and it works, I'll close this one myself.

closed 2023-12-17 by elFarto

pallaswept, 2023-12-18:
Thanks @elFarto . I wasn't ignoring this, but I've been quite ill lately and as a result I'm taking way too long to get things done. I am sorry for my delay. FWIW, I was [working on this](https://build.opensuse.org/package/show/home:pallaswept:branches:home:pallaswept/MozillaFirefox) just last night. I didn't mean any disrespect by taking so long, my apologies.