elFarto/nvidia-vaapi-driver #215: Direct backend no longer working on NVIDIA propreitary driver 535.54.03

FAYZER77, 2023-06-19:
Distro: Arch

Direct backend no longer working on NVIDIA propreitary driver 535.54.03

vainfo output EGL:

Trying display: wayland
Trying display: x11
vainfo: VA-API version: 1.18 (libva 2.18.2)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD

output for direct:
Trying display: wayland
Trying display: x11
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit

elFarto, 2023-06-19:
Can you try the latest version from master?

FAYZER77, 2023-06-19:
Oh, my apologies, I didn't realize the repo I installed it from was out of date.  Building and installing from master made vainfo run successfully using the direct backend.  Thank you so much for your quick response!

closed 2023-06-19 by FAYZER77

marcin-rzeznicki, 2023-06-19:
Sorry to hijack a closed issue, but perhaps a release is in order given that Arch already has 535.54.03 in core - to give a chance to update AUR packages that build from released tag. Thanks