elFarto/nvidia-vaapi-driver #216: Youtube videos lagging/stuttering after pause and resume. Plays fine after jumping to another video position

mmbossoni, 2023-06-19:
Whenever playing youtube videos, if I pause and then resume the video it starts stuttering, or in other words, playback is not smooth.
If I skip the video (by pressing the back arrow to go back a few seconds or jump to another position) video plays smoothly again.
Don't know if that is a general vaapi or firefox issue, but since it just happens when I enable vaapi, starting it here.

Operating System: Arch Linux 
KDE Plasma Version: 5.27.5
KDE Frameworks Version: 5.107.0
Qt Version: 5.15.10
Kernel Version: 6.3.8-x64v1-xanmod1-1 (64-bit)
Graphics Platform: Wayland
Processors: 16 × AMD Ryzen 7 5700X 8-Core Processor
Memory: 15.5 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1070/PCIe/SSE2
Manufacturer: Micro-Star International Co., Ltd.
Product Name: MS-7A37
System Version: 1.0

elFarto, 2023-06-24:
I've not noticed any sort of issue like that. It might be worth using the [Firefox profiler](https://profiler.firefox.com/), with the 'Media' setting to get a profile of the issue.

mmbossoni, 2023-06-24:
https://share.firefox.dev/44zCRcD
Actions go like:
1 - Started viewing a video
2 - Started recording
3 - Count to 5 and pause
4 - Count to 5 and resume
5 - Count to 5 and pressed back

After that firefox drop's me to not accelerated? (I can't see the C+G on nvidia-smi if I keep video paused for too long)

mmbossoni, 2023-06-24:
I'll try to get a video

elFarto, 2023-06-24:
Could you also capture the output of Firefox with NVD_LOG=1 set with this happening?

elFarto, 2023-06-24:
I think I see the issue with that trace. I see references to the dav1d decoder, which means you're watching an AV1 video, which you 1070 can't decode, so it is falling back. I recommend setting the `media.av1.enabled` setting to `false` to prevent AV1 videos from being played back at all.

mmbossoni, 2023-06-24:
I started video playback
Opened a new tab
checked on nvidia-smi that firefox was only with G (probably because new tab firefox was "saving" resources)
switched back to video tab
paused and resumed (got the stuttering)
pressed back and suttering was gone

https://share.firefox.dev/3pmEhIm
[nvdlog.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/11857221/nvdlog.txt)

elFarto, 2023-06-24:
I can't quite figure out what's going on. It does seem to be taking a bit too long to initialise CUDA, but I don't see any obvious problems. I don't suppose you've got any CUDA debugging environment variables set (I see a number of references to 'dbg' in the stack traces)?

At this point, I would raise it with Mozilla, and see what they say.

nahuel3pro, 2023-09-15:
I've got the same problem, could you find any fix?

Glitch31415, 2023-10-04:
same issue here, but with an AMD graphics card. RX 6800

mmbossoni, 2023-10-04:
OK, finally opened a bug on mozilla bugtracker
https://bugzilla.mozilla.org/show_bug.cgi?id=1857106

closed 2023-12-17 by elFarto