elFarto/nvidia-vaapi-driver #218: vainfo return's nvidia_drv_video.so init failed

Bob-YsPan, 2023-06-28:
My graphic card is GTX 750 Ti, on the Ubuntu 23.04
I tried build the `v0.0.10` version from the source and install it, `vainfo` get the error like below:

```
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
     32217.921807778 [27813-27813] ../src/vabackend.c: 138                     init CUDA ERROR 'system not yet initialized' (802)

libva info: Found init function __vaDriverInit_1_0
     32217.921878746 [27813-27813] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     32217.921884097 [27813-27813] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
     32217.921888286 [27813-27813] ../src/vabackend.c:2203       __vaDriverInit_1_0 Selecting EGL backend
     32217.929438379 [27813-27813] ../src/export-buf.c: 132       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
     32217.929468163 [27813-27813] ../src/export-buf.c: 149       findGPUIndexFromFd Looking for GPU index: 0
     32217.932000992 [27813-27813] ../src/export-buf.c: 161       findGPUIndexFromFd Found 3 EGL devices
     32217.934431381 [27813-27813] ../src/export-buf.c: 196       findGPUIndexFromFd No EGL_CUDA_DEVICE_NV support for EGLDevice 0
     32217.934470649 [27813-27813] ../src/export-buf.c: 196       findGPUIndexFromFd No EGL_CUDA_DEVICE_NV support for EGLDevice 1
     32217.934475629 [27813-27813] ../src/export-buf.c: 199       findGPUIndexFromFd No DRM device file for EGLDevice 2
     32217.934486546 [27813-27813] ../src/export-buf.c: 202       findGPUIndexFromFd No match found, falling back to default device
     32217.934494058 [27813-27813] ../src/vabackend.c:2231       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```

asp345, 2023-06-29:
Same error on RTX3060 debian 12 and firefox esr does not enable vaapi.
But `sudo vainfo` works fine.

elFarto, 2023-07-01:
Do you have the latest NVIDIA 470 series driver installed? 

Something is very wrong with that output as EGL_CUDA_DEVICE_NV should be supported on the NVIDIA card, so either the driver isn't installed correctly, or it's not being used. Maybe nouveau is being used instead?

Bob-YsPan, 2023-07-02:
I check the package, it has both Nvidia and Nouveau driver installed, But I use `lsmod` to check which module loaded(maybe include driver?), I not seem any about `nouveau` loaded.

```
ys-pan@bob-dorm-ubuntuMATE /etc/modprobe.d » lsmod | grep xf
xfrm_interface         28672  0
xfrm4_tunnel           16384  0
tunnel4                16384  3 xfrm_interface,xfrm4_tunnel,ip_vti
xfrm6_tunnel           16384  2 xfrm_interface,ipcomp6
xfrm_ipcomp            16384  2 ipcomp6,ipcomp
tunnel6                16384  2 xfrm_interface,xfrm6_tunnel
xfrm_user              61440  2
xfrm_algo              20480  6 esp6,esp4,xfrm_ipcomp,ah6,ah4,xfrm_user
ys-pan@bob-dorm-ubuntuMATE /etc/modprobe.d » lsmod | grep nou
ys-pan@bob-dorm-ubuntuMATE /etc/modprobe.d » lsmod | grep nvi                                                                                                                 1 ↵
nvidia_uvm           1536000  0
nvidia_drm             86016  2
nvidia_modeset       1159168  3 nvidia_drm
nvidia              40706048  84 nvidia_uvm,nvidia_modeset
drm_kms_helper        249856  1 nvidia_drm
drm                   692224  6 drm_kms_helper,nvidia,nvidia_drm
video                  73728  1 nvidia_modeset
```
This is my apt output

```
ys-pan@bob-dorm-ubuntuMATE /etc/modprobe.d » apt search nvidia | grep "installed"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libffmpeg-nvenc-dev/lunar,lunar,now 11.1.5.2-1 all [installed]
libnvidia-cfg1-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-common-515/lunar,lunar,now 515.105.01-0ubuntu1 all [installed,automatic]
libnvidia-compute-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-decode-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-egl-wayland1/lunar,now 1:1.1.10-1 amd64 [installed,automatic]
libnvidia-encode-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-extra-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-fbc1-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
libnvidia-gl-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
libvdpau-dev/lunar,now 1.5-2 amd64 [installed]
libvdpau1/lunar,now 1.5-2 amd64 [installed]
libxnvctrl0/lunar,now 510.47.03-0ubuntu1 amd64 [installed,automatic]
mate-optimus/lunar,lunar,now 21.04.0-1 all [installed,automatic]
mate-sensors-applet/lunar,now 1.26.0-1 amd64 [installed,automatic]
nvidia-compute-utils-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
nvidia-dkms-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
nvidia-driver-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed]
nvidia-kernel-common-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
nvidia-kernel-source-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
nvidia-prime/lunar,lunar,now 0.8.17.1 all [installed,automatic]
nvidia-settings/lunar,now 510.47.03-0ubuntu1 amd64 [installed]
nvidia-utils-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
psensor/lunar,now 1.1.5-1.3ubuntu2 amd64 [installed]
screen-resolution-extra/lunar,lunar,now 0.18.3 all [installed,automatic]
ubuntu-drivers-common/lunar,now 1:0.9.7.1 amd64 [installed,automatic]
vdpauinfo/lunar,now 1.5-2 amd64 [installed]
xserver-xorg-video-nouveau/lunar,now 1:1.0.17-2build1 amd64 [installed,automatic]
xserver-xorg-video-nvidia-515/lunar,now 515.105.01-0ubuntu1 amd64 [installed,automatic]
```

```
ys-pan@bob-dorm-ubuntuMATE /etc/modprobe.d » apt search nouveau | grep "installed"

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libdrm-nouveau2/lunar,now 2.4.114-1 amd64 [installed,automatic]
xserver-xorg-video-nouveau/lunar,now 1:1.0.17-2build1 amd64 [installed,automatic]
```

elFarto, 2023-07-08:
Can you paste the output of `nvidia-smi`?

Bob-YsPan, 2023-07-12:
The `nvidia-smi`'s output seems like normal
```
ys-pan@bob-dorm-ubuntuMATE ~ » nvidia-smi                         
Wed Jul 12 21:16:43 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.54.03              Driver Version: 535.54.03    CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GTX 750 Ti      Off | 00000000:01:00.0 Off |                  N/A |
| 28%   40C    P8               1W /  38W |      6MiB /  2048MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A    110516      G   /usr/lib/xorg/Xorg                            2MiB |
+---------------------------------------------------------------------------------------+
```

Bob-YsPan, 2023-07-12:
I seem like found the cause of the problem.
The Ubuntu 23.04's `libva` library was updated to 2.17.0 and included some DRI3 things.

Today I found this discussion mentions a temporary solution about recompile the `libva` library, and I download Debian's source package, apply the patch it writes, and then recompile and install it.

Seems like the problem was solved, and I test the MPV can normally play video without any ERROR prints.

But the Vivaldi(Chromium based) freezes when click the page including videos it freezes, I look at the README seems like it is normal(Maybe it is just not capable)

The output when I installed the patched `libva-x11-2` (From the `libva` source package)

```
ys-pan@bob-dorm-ubuntuMATE ~ » env LIBVA_DRIVER_NAME=nvidia vainfo             
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
ys-pan@bob-dorm-ubuntuMATE ~ » env LIBVA_DRIVER_NAME=nvidia mpv --hwdec=vaapi ~/CTS_TEST.mp4
 (+) Video --vid=1 (*) (h264 1920x1080 30.000fps)
 (+) Audio --aid=1 (*) (aac 2ch 44100Hz)
AO: [pipewire] 44100Hz stereo 2ch floatp
VO: [gpu] 1920x1080 yuv420p
AV: 00:01:06 / 00:04:27 (25%) A-V:  0.000
```

elFarto, 2023-07-13:
I'm not sure if mpv is using VA-API in your test, as the egl backend isn't currently working due to an NVIDIA driver issue. You can add `NVD_LOG=1` to the mpv command to see if it is actually using it.

Bob-YsPan, 2023-07-14:
Oh yeah... If I use `NVD_LOG=1`, the vainfo seems like normal, but the MPV still have problem.

`vainfo` command
```
ys-pan@bob-dorm-ubuntuMATE ~ » env LIBVA_DRIVER_NAME=nvidia NVD_LOG=1 vainfo
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       497.628854557 [15914-15914] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
       497.628893973 [15914-15914] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
       497.628899173 [15914-15914] ../src/vabackend.c:2203       __vaDriverInit_1_0 Selecting EGL backend
       497.638709179 [15914-15914] ../src/export-buf.c: 132       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
       497.638734144 [15914-15914] ../src/export-buf.c: 149       findGPUIndexFromFd Looking for GPU index: 0
       497.641689969 [15914-15914] ../src/export-buf.c: 161       findGPUIndexFromFd Found 4 EGL devices
       497.641809315 [15914-15914] ../src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       497.641815976 [15914-15914] ../src/export-buf.c: 191       findGPUIndexFromFd Selecting EGLDevice 0
       497.643110778 [15914-15914] ../src/export-buf.c: 260         egl_initExporter Driver supports 16-bit surfaces
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
       497.700511137 [15914-15914] ../src/vabackend.c:2081              nvTerminate Terminating 0x55a86325ef50
       497.700538060 [15914-15914] ../src/export-buf.c:  61      egl_releaseExporter Releasing exporter, 0 outstanding frames
       497.700558140 [15914-15914] ../src/export-buf.c:  78      egl_releaseExporter Done releasing frames
       497.700563222 [15914-15914] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances
```

`mpv` prints a lot of error
```
ys-pan@bob-dorm-ubuntuMATE ~ » env NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia mpv --hwdec=vaapi ~/CTS_TEST.mp4
 (+) Video --vid=1 (*) (h264 1920x1080 30.000fps)
 (+) Audio --aid=1 (*) (aac 2ch 44100Hz)
       502.904328201 [15963-15974] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
       502.904342159 [15963-15974] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
       502.904346791 [15963-15974] ../src/vabackend.c:2203       __vaDriverInit_1_0 Selecting EGL backend
       502.904382592 [15963-15974] ../src/export-buf.c: 132       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
       502.904387835 [15963-15974] ../src/export-buf.c: 149       findGPUIndexFromFd Looking for GPU index: 0
       502.905462148 [15963-15974] ../src/export-buf.c: 161       findGPUIndexFromFd Found 4 EGL devices
       502.905626784 [15963-15974] ../src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
       502.905633337 [15963-15974] ../src/export-buf.c: 191       findGPUIndexFromFd Selecting EGLDevice 0
       502.906916491 [15963-15974] ../src/export-buf.c: 260         egl_initExporter Driver supports 16-bit surfaces
       502.937163138 [15963-15974] ../src/vabackend.c:1409      nvQueryImageFormats In nvQueryImageFormats
       502.962339786 [15963-15974] ../src/vabackend.c: 648           nvCreateConfig got profile: 0 with 0 attributes
       502.962386064 [15963-15974] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 1 (8) (nil) 0
       502.962394979 [15963-15974] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 1 (8) 0x7efde4d23e40 8
       502.964868026 [15963-15974] ../src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
       502.964905938 [15963-15974] ../src/vabackend.c: 931        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7efde4a41470)
       502.964916146 [15963-15974] ../src/vabackend.c:1503            nvDeriveImage In nvDeriveImage
       502.964938095 [15963-15974] ../src/vabackend.c: 931        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7efde4b6dc60)
       502.964944036 [15963-15974] ../src/vabackend.c:1503            nvDeriveImage In nvDeriveImage
       502.965020047 [15963-15974] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: -1 (0x7efde4b6dc60)
       502.965026291 [15963-15974] ../src/export-buf.c: 528       egl_realiseSurface No free surfaces found
       502.965248604 [15963-15974] ../src/export-buf.c: 440 egl_allocateBackingImage Presenting frame -1 128x128 (0x7efde4b6dc60, 0x7efde4b6dd00, 0x7efde4b70ac0)
       502.965257425 [15963-15974] ../src/export-buf.c: 441 egl_allocateBackingImage CUDA ERROR 'invalid resource handle' (400)

[... cut ...]
       502.965813858 [15963-15974] ../src/export-buf.c: 441 egl_allocateBackingImage CUDA ERROR 'unknown error' (999)

       502.965835535 [15963-15974] ../src/export-buf.c:  95                reconnect Reconnecting to stream
       502.965882047 [15963-15974] ../src/export-buf.c: 115                reconnect CUDA ERROR 'operation not supported' (801)

...(Seems like repeat many times)
```
I changed my motherboard recently, and now I have a intergated GPU(HD 530), but I tried to run the command in the Nvidia only mode and Prime Offload mode, the problem prints still the same.

[image]

elFarto, 2023-07-14:
Set `NVD_BACKEND=direct`, the EGL backend is not currently working due to an NVIDIA driver issue.

Bob-YsPan, 2023-07-14:
Hmm... seems like the log print is normal now.
But I'm not very sure how to check the CUDA's usage to ensure the GPU is hardware decoding?

```
 (+) Video --vid=1 (*) (h264 1920x1080 30.000fps)
 (+) Audio --aid=1 (*) (aac 2ch 44100Hz)
       197.611467498 [4411-4422] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
       197.611503747 [4411-4422] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
       197.611525152 [4411-4422] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
       197.611631900 [4411-4422] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
       197.611670164 [4411-4422] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
       197.611699923 [4411-4422] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
       197.611722651 [4411-4422] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 100 1 0 fe
       197.611749735 [4411-4422] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 535.54.03, major version: 535
       197.647679028 [4411-4422] ../src/vabackend.c:1409      nvQueryImageFormats In nvQueryImageFormats
       197.675283356 [4411-4422] ../src/vabackend.c: 648           nvCreateConfig got profile: 0 with 0 attributes
       197.675313473 [4411-4422] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 1 (8) (nil) 0
       197.675321430 [4411-4422] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 1 (8) 0x7f7cd8ab9100 8
       197.676851685 [4411-4422] ../src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
       197.676887550 [4411-4422] ../src/vabackend.c: 931        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7f7cd87d7d30)
       197.676896839 [4411-4422] ../src/vabackend.c:1503            nvDeriveImage In nvDeriveImage
       197.676906304 [4411-4422] ../src/vabackend.c: 931        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7f7cd89035b0)
       197.676913076 [4411-4422] ../src/vabackend.c:1503            nvDeriveImage In nvDeriveImage
       197.676958436 [4411-4422] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: -1 (0x7f7cd89035b0)
       197.676966324 [4411-4422] ../src/direct/direct-export-buf.c: 198 direct_allocateBackingImage Allocating BackingImages: 0x7f7cd8903650 128x128
       197.676973276 [4411-4422] ../src/direct/nv-driver.c: 433              alloc_image Calculated GOB size: 64x128 (0x4)
       197.676979591 [4411-4422] ../src/direct/nv-driver.c: 442              alloc_image Aligned image size: 128x128 = 16384
       197.677158210 [4411-4422] ../src/direct/nv-driver.c: 518              alloc_image created image: 128x128 3000000004fe014 128 4000
       197.677167868 [4411-4422] ../src/direct/nv-driver.c: 433              alloc_image Calculated GOB size: 64x64 (0x3)
       197.677174251 [4411-4422] ../src/direct/nv-driver.c: 442              alloc_image Aligned image size: 128x64 = 8192
       197.677267641 [4411-4422] ../src/direct/nv-driver.c: 518              alloc_image created image: 64x64 3000000004fe013 128 2000
       197.677276552 [4411-4422] ../src/direct/direct-export-buf.c: 204 direct_allocateBackingImage Importing images
       197.677283212 [4411-4422] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 128x128 = 4000
       197.677398011 [4411-4422] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 64x64 = 2000
       197.677549186 [4411-4422] ../src/vabackend.c:2068    nvExportSurfaceHandle Exporting with 128 128 0 128 3000000004fe014 0 128 3000000004fe013
       197.677579602 [4411-4422] ../src/direct/direct-export-buf.c:  42                    debug [EGL] eglCreateImageKHR: createImageFromDmaBufs failed
       197.677600932 [4411-4422] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7f7cd89035b0)
       197.677607710 [4411-4422] ../src/direct/direct-export-buf.c: 243      destroyBackingImage Destroying BackingImage: 0x7f7cd8903650
       197.677739063 [4411-4422] ../src/vabackend.c: 931        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x7f7cd8906820)
       197.677760033 [4411-4422] ../src/vabackend.c:1503            nvDeriveImage In nvDeriveImage
       197.677779438 [4411-4422] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: -1 (0x7f7cd8906820)
       197.677786890 [4411-4422] ../src/direct/direct-export-buf.c: 198 direct_allocateBackingImage Allocating BackingImages: 0x7f7cd8903250 128x128
       197.677793380 [4411-4422] ../src/direct/nv-driver.c: 433              alloc_image Calculated GOB size: 64x128 (0x4)
       197.677799643 [4411-4422] ../src/direct/nv-driver.c: 442              alloc_image Aligned image size: 256x128 = 32768
       197.677921921 [4411-4422] ../src/direct/nv-driver.c: 518              alloc_image created image: 128x128 3000000004fe014 256 8000
       197.677931382 [4411-4422] ../src/direct/nv-driver.c: 433              alloc_image Calculated GOB size: 64x64 (0x3)
       197.677937653 [4411-4422] ../src/direct/nv-driver.c: 442              alloc_image Aligned image size: 256x64 = 16384
       197.678029674 [4411-4422] ../src/direct/nv-driver.c: 518              alloc_image created image: 64x64 3000000004fe013 256 4000
[... cut ...]
       197.708651126 [4411-4422] ../src/direct/direct-export-buf.c: 243      destroyBackingImage Destroying BackingImage: 0x7f7cd8906880
       197.708817312 [4411-4422] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7f7cd890a440)
       197.708828623 [4411-4422] ../src/vabackend.c:2081              nvTerminate Terminating 0x7f7cd82d0ad0
       197.708904254 [4411-4422] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances
AO: [pipewire] 44100Hz stereo 2ch floatp
VO: [gpu] 1920x1080 yuv420p
```

Bob-YsPan, 2023-07-14:
The chrome's decode output like this:
Seems like chrome's VA-API support actually have some bugs...
```
...(Too huge of the log prints)

       480.638156550 [4925-5191] ../src/direct/direct-export-buf.c: 198 direct_allocateBackingImage Allocating BackingImages: 0xa0805749880 1920x1088
       480.638172882 [4925-5191] ../src/direct/nv-driver.c: 433              alloc_image Calculated GOB size: 64x128 (0x4)
       480.638185812 [4925-5191] ../src/direct/nv-driver.c: 442              alloc_image Aligned image size: 1920x1152 = 2211840
       480.639820530 [4925-5191] ../src/direct/nv-driver.c: 518              alloc_image created image: 1920x1088 3000000004fe014 1920 21c000
       480.639844791 [4925-5191] ../src/direct/nv-driver.c: 433              alloc_image Calculated GOB size: 64x128 (0x4)
       480.639849378 [4925-5191] ../src/direct/nv-driver.c: 442              alloc_image Aligned image size: 1920x640 = 1228800
       480.640095002 [4925-5191] ../src/direct/nv-driver.c: 518              alloc_image created image: 960x544 3000000004fe014 1920 12c000
       480.640101688 [4925-5191] ../src/direct/direct-export-buf.c: 204 direct_allocateBackingImage Importing images
       480.640106548 [4925-5191] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 1920x1088 = 21c000
       480.640412833 [4925-5191] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 960x544 = 12c000
       480.640765879 [4925-5191] ../src/vabackend.c: 423          resolveSurfaces Surface 0 exported
       480.640829479 [4925-5191] ../src/vabackend.c: 399          resolveSurfaces Reading from queue: 1 5
       480.640833941 [4925-5191] ../src/vabackend.c: 415          resolveSurfaces Mapping surface 1
       480.640911955 [4925-4925] ../src/vabackend.c:1397             nvPutSurface In nvPutSurface
[4925:4925:0714/225454.750513:ERROR:vaapi_wrapper.cc(2703)] vaPutSurface failed, VA error: the requested function is not implemented
[4925:4925:0714/225454.751062:ERROR:vaapi_video_decode_accelerator.cc(287)] Failed putting surface into pixmap
       480.641993383 [4925-5191] ../src/vabackend.c: 419          resolveSurfaces Mapped surface 1 to 0x503020000 (2048)
       480.642176605 [4925-5191] ../src/direct/direct-export-buf.c: 198 direct_allocateBackingImage Allocating BackingImages: 0xa08057496c0 1920x1088
       480.642674893 [4925-5191] ../src/direct/nv-driver.c: 433              alloc_image Calculated GOB size: 64x128 (0x4)
       480.642742282 [4925-5191] ../src/direct/nv-driver.c: 442              alloc_image Aligned image size: 1920x1152 = 2211840
       480.643076849 [4925-5191] ../src/direct/nv-driver.c: 518              alloc_image created image: 1920x1088 3000000004fe014 1920 21c000
       480.643344542 [4925-5191] ../src/direct/nv-driver.c: 433              alloc_image Calculated GOB size: 64x128 (0x4)
       480.643948576 [4925-5191] ../src/direct/nv-driver.c: 442              alloc_image Aligned image size: 1920x640 = 1228800
       480.644151355 [4925-5191] ../src/direct/nv-driver.c: 518              alloc_image created image: 960x544 3000000004fe014 1920 12c000
       480.644157080 [4925-5191] ../src/direct/direct-export-buf.c: 204 direct_allocateBackingImage Importing images
       480.644162281 [4925-5191] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 1920x1088 = 21c000
       480.644471655 [4925-5191] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 960x544 = 12c000
       480.644754484 [4925-5191] ../src/vabackend.c: 423          resolveSurfaces Surface 1 exported
       480.644764084 [4925-5191] ../src/vabackend.c: 399          resolveSurfaces Reading from queue: 2 5
       480.644768192 [4925-5191] ../src/vabackend.c: 415          resolveSurfaces Mapping surface 2
       480.644826542 [4925-5191] ../src/vabackend.c: 419          resolveSurfaces Mapped surface 2 to 0x503020000 (2048)
       480.644833024 [4925-5191] ../src/direct/direct-export-buf.c: 198 direct_allocateBackingImage Allocating BackingImages: 0xa080574a300 1920x1088

...(Too huge of the log prints)
```

elFarto, 2023-07-14:
The first output looks like it's having trouble importing the image:
```
eglCreateImageKHR: createImageFromDmaBufs failed
```
Which likely means that mpv is picking the wrong EGL instance (the Intel card) and trying to import the video there, which isn't supported.

The chromium output ...seems to be working..? The `vaPutSurface` error is expected, but the rest of it looks like it's decoding, which is unexpected as it shouldn't be working on Chromium.

Bob-YsPan, 2023-07-14:
OK, I will try to change the BIOS setting and boot the system by the Nvidia Card (And plug the monitor cable to that card) and see will make the difference or not.

Bob-YsPan, 2023-07-14:
The mpv look like a lot of different than last test:
[image]

It's print the log constantly and... wala, now have the video processing usage.

Bob-YsPan, 2023-07-14:
But the Firefox still can't use the VA-API decode... weird...
(The Chromium's problem still the same, seem like actually it just shouldn't fit the Chromium)
Although I only use the Nvidia Card
Maybe I open the new Issue to trace this problem?

```
ys-pan@bob-dorm-ubuntuMATE ~ » env NVD_LOG=1 NVD_BACKEND=direct MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json firefox
Gtk-Message: 00:31:10.749: Failed to load module "appmenu-gtk-module"
Gtk-Message: 00:31:10.749: Not loading module "atk-bridge": The functionality is provided by GTK natively. Please try to not load it.
[GFX1-]: glxtest: libEGL no display
[GFX1-]: vaapitest: ERROR
[GFX1-]: vaapitest: VA-API test failed: failed to open renderDeviceFD.

ys-pan@bob-dorm-ubuntuMATE ~ » env NVD_LOG=1 NVD_BACKEND=direct MOZ_DISABLE_RDD_SANDBOX=1 LIBVA_DRIVER_NAME=nvidia firefox                              130 ↵
Gtk-Message: 00:31:36.675: Failed to load module "appmenu-gtk-module"
Gtk-Message: 00:31:36.676: Not loading module "atk-bridge": The functionality is provided by GTK natively. Please try to not load it.
[GFX1-]: vaapitest: ERROR
[GFX1-]: vaapitest: VA-API test failed: failed to initialise VAAPI connection.
```

OmegaRogue, 2023-09-04:
I'm on arch linux with `libva-nvidia-driver` installed, and `NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia VDPAU_DRIVER=nvidia vainfo` outputs this:
```
Trying display: wayland
Trying display: x11
     48253.419681238 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c: 138                     init CUDA ERROR 'unknown error' (999)

     48253.419699618 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     48253.419703368 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
     48253.419706718 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
     48253.425473611 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
     48253.425480231 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
     48253.425485601 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 2700 1 2 6
     48253.425514052 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 535.104.05, major version: 535
     48253.425608253 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/direct/direct-export-buf.c:  23       findGPUIndexFromFd CUDA ERROR 'initialization error' (3)

     48253.425612423 [352904-352904] ../nvidia-vaapi-driver-0.0.10/src/vabackend.c:2236       __vaDriverInit_1_0 CUDA ERROR 'initialization error' (3)

libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

Bob-YsPan, 2023-09-05:
> I seem like found the cause of the problem. The Ubuntu 23.04's `libva` library was updated to 2.17.0 and included some DRI3 things.
> 
> Today I found this [Arch Linux discussion](https://bbs.archlinux.org/viewtopic.php?pid=2080054#p2080054) mentions a temporary solution about recompile the `libva` library, and I download Debian's source package, apply the patch it writes, and then recompile and replace the `libva-x11-2` package(Force remove, Remove changelog.Debian.gz, and then install Patched package).
> 
> Seems like the problem was solved, and I test the MPV can normally play video without any ERROR prints.
> 
> But the Vivaldi(Chromium based) freezes when click the page including videos, I look at the README seems like it is normal(Maybe it is just not capable)
> 
> The output when I installed the patched `libva-x11-2` (From the `libva` source package)
> 
> ```
> ys-pan@bob-dorm-ubuntuMATE ~ » env LIBVA_DRIVER_NAME=nvidia vainfo             
> libva info: VA-API version 1.17.0
> libva info: User environment variable requested driver 'nvidia'
> libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
> libva info: Found init function __vaDriverInit_1_0
> libva info: va_openDriver() returns 0
> vainfo: VA-API version: 1.17 (libva 2.12.0)
> vainfo: Driver version: VA-API NVDEC driver [egl backend]
> vainfo: Supported profile and entrypoints
>       VAProfileMPEG2Simple            :	VAEntrypointVLD
>       VAProfileMPEG2Main              :	VAEntrypointVLD
>       VAProfileVC1Simple              :	VAEntrypointVLD
>       VAProfileVC1Main                :	VAEntrypointVLD
>       VAProfileVC1Advanced            :	VAEntrypointVLD
>       VAProfileH264Main               :	VAEntrypointVLD
>       VAProfileH264High               :	VAEntrypointVLD
>       VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
> ys-pan@bob-dorm-ubuntuMATE ~ » env LIBVA_DRIVER_NAME=nvidia mpv --hwdec=vaapi ~/CTS_TEST.mp4
>  (+) Video --vid=1 (*) (h264 1920x1080 30.000fps)
>  (+) Audio --aid=1 (*) (aac 2ch 44100Hz)
> AO: [pipewire] 44100Hz stereo 2ch floatp
> VO: [gpu] 1920x1080 yuv420p
> AV: 00:01:06 / 00:04:27 (25%) A-V:  0.000
> ```

@OmegaRogue Have you tried the DRI patche's way?
The ArchLinux seems like need to patches the driver at the source code download from AUR, then install it.
But I'm not sure about the new version of driver already contains this patch, you may need to check to ensure that

OmegaRogue, 2023-09-05:
> @OmegaRogue Have you tried the DRI patche's way? The ArchLinux seems like need to patches the driver at the source code download from AUR, then install it. But I'm not sure about the new version of [...]

I havent, is there a premade AUR Package with the patch or do i have to modify it myself?

Bob-YsPan, 2023-09-05:
I writes a shell script with refrence to this [forum](https://bbs.archlinux.org/viewtopic.php?pid=2080054#p2080054):
Clone the libva's AUR repository and patch by the below script, and them make the package & install, the driver may works
(I not very sure because the error is not quite the same as mine, but worth give it a try)

```
#!/bin/bash

# You may need to change "./libva-2.17.0/va/x11/va_x11.c" to your location of "va_x11.c"
sed -i "s/va_DRI3_Get/va_DRI2_Get/g" ./libva-2.17.0/va/x11/va_x11.c
```

OmegaRogue, 2023-09-05:
after a reboot it does seem to work, but audio seems to be cutting out every few seconds

Bob-YsPan, 2023-09-05:
> The mpv look like a lot of different than last test: [image]
> 
> It's print the log constantly and... wala, now have the video processing usage.
> 
> The partial log print:
> 
> ```
>       1314.416322067 [12368-12379] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: 3 (0x562e726a0d10)
>       1314.416380863 [12368-12379] ../src/vabackend.c:2068    nvExportSurfaceHandle Exporting with 1920 1088 0 1920 3000000004fe014 0 1920 3000000004fe014
>       1314.445592418 [12368-12368] ../src/vabackend.c:1311             nvEndPicture Decoded frame successfully to idx: 4 (0x562e726a0780)
>       1314.445712414 [12368-12383] ../src/vabackend.c: 399          resolveSurfaces Reading from queue: 8 9
>       1314.445755216 [12368-12383] ../src/vabackend.c: 415          resolveSurfaces Mapping surface 4
>       1314.447910835 [12368-12383] ../src/vabackend.c: 419          resolveSurfaces Mapped surface 4 to 0x506020000 (2048)
>       1314.448046190 [12368-12383] ../src/vabackend.c: 423          resolveSurfaces Surface 4 exported
>       1314.449606966 [12368-12379] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: 2 (0x562e726a12a0)
>       1314.449656196 [12368-12379] ../src/vabackend.c:2068    nvExportSurfaceHandle Exporting with 1920 1088 0 1920 3000000004fe014 0 1920 3000000004fe014
>       1314.466883909 [12368-12368] ../src/vabackend.c:1311             nvEndPicture Decoded frame successfully to idx: 3 (0x562e726a0d10)
>       1314.467008362 [12368-12383] ../src/vabackend.c: 399          resolveSurfaces Reading from queue: 9 10
>       1314.467045249 [12368-12383] ../src/vabackend.c: 415          resolveSurfaces Mapping surface 3
>       1314.469175366 [12368-12383] ../src/vabackend.c: 419          resolveSurfaces Mapped surface 3 to 0x506020000 (2048)
>       1314.469305249 [12368-12383] ../src/vabackend.c: 423          resolveSurfaces Surface 3 exported
>       1314.482946418 [12368-12379] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: 1 (0x562e726a1830)
>       1314.483004057 [12368-12379] ../src/vabackend.c:2068    nvExportSurfaceHandle Exporting with 1920 1088 0 1920 3000000004fe014 0 1920 3000000004fe014
>       1314.516831672 [12368-12379] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: 4 (0x562e726a0780)
>       1314.516894567 [12368-12379] ../src/vabackend.c:2068    nvExportSurfaceHandle Exporting with 1920 1088 0 1920 3000000004fe014 0 1920 3000000004fe014
>       1314.516930043 [12368-12368] ../src/vabackend.c:1311             nvEndPicture Decoded frame successfully to idx: 1 (0x562e726a1830)
>       1314.517027820 [12368-12383] ../src/vabackend.c: 399          resolveSurfaces Reading from queue: 10 11
>       1314.517052084 [12368-12383] ../src/vabackend.c: 415          resolveSurfaces Mapping surface 1
>       1314.519162431 [12368-12383] ../src/vabackend.c: 419          resolveSurfaces Mapped surface 1 to 0x506020000 (2048)
>       1314.519271766 [12368-12383] ../src/vabackend.c: 423          resolveSurfaces Surface 1 exported
>       1314.541490141 [12368-12368] ../src/vabackend.c:1311             nvEndPicture Decoded frame successfully to idx: 2 (0x562e726a12a0)
>       1314.541572669 [12368-12383] ../src/vabackend.c: 399          resolveSurfaces Reading from queue: 11 12
>       1314.541611162 [12368-12383] ../src/vabackend.c: 415          resolveSurfaces Mapping surface 2
>       1314.543726014 [12368-12383] ../src/vabackend.c: 419          resolveSurfaces Mapped surface 2 to 0x506020000 (2048)
>       1314.543913502 [12368-12383] ../src/vabackend.c: 423          resolveSurfaces Surface 2 exported
>       1314.549677042 [12368-12379] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: 3 (0x562e726a0d10)
>       1314.549727158 [12368-12379] ../src/vabackend.c:2068    nvExportSurfaceHandle Exporting with 1920 1088 0 1920 3000000004fe014 0 1920 3000000004fe014
>       1314.583304949 [12368-12379] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: 1 (0x562e726a1830)
>       1314.583376944 [12368-12368] ../src/vabackend.c:1311             nvEndPicture Decoded frame successfully to idx: 4 (0x562e726a0780)
>       1314.583484630 [12368-12383] ../src/vabackend.c: 399          resolveSurfaces Reading from queue: 12 13
>       1314.583569166 [12368-12383] ../src/vabackend.c: 415          resolveSurfaces Mapping surface 4
>       1314.583357105 [12368-12379] ../src/vabackend.c:2068    nvExportSurfaceHandle Exporting with 1920 1088 0 1920 3000000004fe014 0 1920 3000000004fe014
>       1314.585620183 [12368-12383] ../src/vabackend.c: 419          resolveSurfaces Mapped surface 4 to 0x506020000 (2048)
>       1314.585727472 [12368-12383] ../src/vabackend.c: 423          resolveSurfaces Surface 4 exported
>       1314.616277978 [12368-12368] ../src/vabackend.c:1311             nvEndPicture Decoded frame successfully to idx: 3 (0x562e726a0d10)
>       1314.616355524 [12368-12379] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: 2 (0x562e726a12a0)
>       1314.616357058 [12368-12383] ../src/vabackend.c: 399          resolveSurfaces Reading from queue: 13 14
>       1314.616693956 [12368-12383] ../src/vabackend.c: 415          resolveSurfaces Mapping surface 3
>       1314.616661455 [12368-12379] ../src/vabackend.c:2068    nvExportSurfaceHandle Exporting with 1920 1088 0 1920 3000000004fe014 0 1920 3000000004fe014
[... cut ...]
>       1314.717232109 [12368-12368] ../src/vabackend.c: 301           destroyContext Waiting for resolve thread to exit
>       1314.717250218 [12368-12383] ../src/vabackend.c: 428          resolveSurfaces [RT] Resolve thread for 0x562e7291f670 exiting
>       1314.717427489 [12368-12368] ../src/vabackend.c: 303           destroyContext pthread_timedjoin_np finished with 0
> ```

@OmegaRogue Maybe you can append the `NVD_LOG=1` to ensure the driver is working
If the log prints like mines posts at top, the driver may be working correctly.

OmegaRogue, 2023-09-05:
> @OmegaRogue Maybe you can append the `NVD_LOG=1` to ensure the driver is working If the log prints like mines posts at top, the driver may be working correctly.

where should i append that?

Bob-YsPan, 2023-09-05:
Append to the front of program you want to run:
e.g. `env NVD_LOG=1 firefox`(or the `env` can be ignored if you use the Terminal)

elFarto, 2023-12-17:
Is this still an issue?

Bob-YsPan, 2023-12-18:
Maybe it is fixed,
but Firefox can't use the hw decoding at few months ago, maybe it will some settings at Firefox causes.

If it still have a problem, I will search the issue exists or open another issue.

closed 2023-12-18 by Bob-YsPan

mirh, 2024-02-02:
Yes this is still an issue, and libva has nothing to do with it.
It's a regression with older drivers that specifically happened after bb7752d37412b05f0151e60904f5e743d08ec297