elFarto/nvidia-vaapi-driver #219: va_openDriver() returns -1 in Firefox

billzhonggz, 2023-07-01:
Hi, I am using RTX 4090 with Ubuntu 23.04 and NVIDIA driver 530. I installed `nvidia-vaapi-driver` from master branch. I am using X11, direct back end, and everything is up-to-date.

The output of `NVD_LOG=1 vainfo` looks normal.

```
$ NVD_LOG=1 vainfo
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      1057.861263864 [13287-13287] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      1057.861268707 [13287-13287] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1057.861270828 [13287-13287] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
      1057.864235230 [13287-13287] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      1057.864239029 [13287-13287] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
      1057.864241139 [13287-13287] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 100 1 2 6
      1057.864254918 [13287-13287] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 530.41.03, major version: 530
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      1057.910796947 [13287-13287] ../src/vabackend.c:2081              nvTerminate Terminating 0x5638eeef1060
      1057.910823059 [13287-13287] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances
```

But I got this output from Firefox and the hardware decode is not working when watching Youtube (AV1 codec). CPU rate is high and no *compute* task shows in `nvidia-smi`.

```
$ NVD_LOG=1 firefox
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
ATTENTION: default value of option mesa_glthread overridden by environment.
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /snap/firefox/2517/gnome-platform/usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
```

Can anyone helps?

elFarto, 2023-07-01:
You're using the Snap version of Firefox, but have the driver installed outside of it, which it is unable to access.

mikemocha, 2023-07-01:
I have this issue as well and can't figure out where the problem is.  I am also using X11.  There's a couple of links about this in various places:

[https://bugzilla.mozilla.org/show_bug.cgi?id=1766517](url)
[https://bugs.launchpad.net/ubuntu/+source/firefox/+bug/1947115](url)
[https://bugzilla.mozilla.org/show_bug.cgi?id=1751363](url)

There's a few more threads on Reddit.  I'm searching for words like "snap firefox vaapi nvidia"

It's still not clear to me if this is a snap bug.  Changing environment variables to point to the location of the nvidia_drv_video.so doesn't change the behavior.  I'm not sure what to do about this.

billzhonggz, 2023-07-02:
> You're using the Snap version of Firefox, but have the driver installed outside of it, which it is unable to access.

Thanks for the reply! I uninstalled the Firefox that come with Ubuntu and manually installed from the `tar.bz` downloaded from Firefox website. The problem is solved now.

The default Firefox on Ubuntu is a snap version anyway, include the one from `apt`.

closed 2023-07-02 by billzhonggz

philipl, 2023-07-02:
I'd recommend the ppa to get packages: https://launchpad.net/~mozillateam/+archive/ubuntu/ppa