elFarto/nvidia-vaapi-driver #222: Request: support encode profiles via NVENC..

oscarbg, 2023-07-14:
Hi,

I know primary interest is to support Firefox..

but from a theoretical view would be nice if this project can evolve to include also VAAPI encoder profiles using NVENC..
don't know how much difficult would be to add support for..

one simple use case is for uses having ffmpeg "transcode" scripts using libva as they work on Intel and AMD GPUs already..

./ffmpeg -init_hw_device vaapi=hw:/dev/dri/renderD128 -hwaccel_output_format vaapi -hwaccel vaapi -i a.mp4 -vcodec h265_vaapi -acodec aac b.mp4

heck even using cartwheel-ffmpeg (https://github.com/intel/cartwheel-ffmpeg) it provides FFMPEG libva (VAAPI) AV1 encode support and NVENC supports also AV1 on Ada GPUs..

so in brief, it could be nice if this driver supported H264,HEVC,AV1 encode profiles also..
thanks..

elFarto, 2023-07-14:
We already have an issue open for this, #116. However, it's not an easy task as the two APIs are very different, and information on using the encode side of VA-API is quite thin on the ground.

closed 2023-07-14 by elFarto

oscarbg, 2023-07-14:
ok sorry.. lazy to search..