elFarto/nvidia-vaapi-driver #223: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)

RafaelLinux, 2023-07-20:
A week ago, I installed for first time RPM version of `nvidia-vaapi-driver` from [this repository](https://build.opensuse.org/package/binaries/home:awerlang/nvidia-vaapi-driver/openSUSE_Tumbleweed).  I reboot and [vainfo](url) output was correct, recognizing vaapi installed and working.

However, two days ago, I checked again `vainfo` output, cause I felt Firefox playing videos like before install `nvidia-vaapi-driver` , and now it's failing.
```
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.18.0
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
vaInitialize failed with error code -1 (unknown libva error),exit

```

Only change I remember I did was a kernel upgrade (see below info about my system) and in fact, I upgraded again kernel and even a new NVidia propietary driver version, but `vainfo` check still failing.

```
Operating System: openSUSE Tumbleweed 20230718
KDE Plasma Version: 5.27.6
KDE Frameworks Version: 5.108.0
Qt Version: 5.15.10
Kernel Version: 6.4.3-1-default (64-bit)
Graphics Platform: X11
Graphics Processor: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2
```

And this is the `nvidia-smi` output

```
Thu Jul 20 23:37:14 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.86.05              Driver Version: 535.86.05    CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GTX 1050 Ti     Off | 00000000:02:00.0  On |                  N/A |
| 35%   48C    P0              N/A /  75W |   1912MiB /  4096MiB |     27%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A     20345      G   /usr/bin/Xorg.bin                           661MiB |
|    0   N/A  N/A     20502      G   /usr/bin/kwalletd5                            1MiB |
|    0   N/A  N/A     20659      G   /usr/bin/ksmserver                            1MiB |
|    0   N/A  N/A     20663      G   /usr/bin/kded5                                1MiB |
|    0   N/A  N/A     20664      G   /usr/bin/kwin_x11                           828MiB |
|    0   N/A  N/A     20696      G   /usr/bin/plasmashell                        124MiB |
|    0   N/A  N/A     20707      G   ...c/polkit-kde-authentication-agent-1        1MiB |
|    0   N/A  N/A     20795      G   /usr/bin/knotes                               1MiB |
|    0   N/A  N/A     20811      G   /usr/bin/kaccess                              1MiB |
|    0   N/A  N/A     20815      G   /usr/bin/kalendarac                           1MiB |
|    0   N/A  N/A     20886      G   /usr/bin/akonadi_control                      1MiB |
|    0   N/A  N/A     21075      G   /usr/bin/akonadi_akonotes_resource            1MiB |
|    0   N/A  N/A     21076      G   /usr/bin/akonadi_archivemail_agent            1MiB |
|    0   N/A  N/A     21077      G   /usr/bin/akonadi_birthdays_resource           1MiB |
|    0   N/A  N/A     21078      G   /usr/bin/akonadi_contacts_resource            1MiB |
|    0   N/A  N/A     21079      G   .../bin/akonadi_followupreminder_agent        1MiB |
|    0   N/A  N/A     21080      G   /usr/bin/akonadi_google_resource              1MiB |
|    0   N/A  N/A     21082      G   /usr/bin/akonadi_imap_resource                1MiB |
|    0   N/A  N/A     21084      G   /usr/bin/akonadi_indexing_agent               1MiB |
|    0   N/A  N/A     21085      G   /usr/bin/akonadi_maildir_resource             1MiB |
|    0   N/A  N/A     21086      G   /usr/bin/akonadi_maildir_resource             1MiB |
|    0   N/A  N/A     21088      G   /usr/bin/akonadi_maildispatcher_agent         1MiB |
|    0   N/A  N/A     21093      G   /usr/bin/akonadi_mailfilter_agent             1MiB |
|    0   N/A  N/A     21095      G   /usr/bin/akonadi_mailmerge_agent              1MiB |
|    0   N/A  N/A     21097      G   /usr/bin/akonadi_migration_agent              1MiB |
|    0   N/A  N/A     21100      G   /usr/bin/akonadi_newmailnotifier_agent        1MiB |
|    0   N/A  N/A     21102      G   /usr/bin/akonadi_notes_agent                  1MiB |
|    0   N/A  N/A     21105      G   /usr/bin/akonadi_sendlater_agent              1MiB |
|    0   N/A  N/A     21107      G   /usr/bin/akonadi_unifiedmailbox_agent         1MiB |
|    0   N/A  N/A     21486      G   /usr/lib64/firefox/firefox                  233MiB |
|    0   N/A  N/A     21525      G   /usr/lib/mozilla/kmozillahelper               1MiB |
|    0   N/A  N/A     21537      G   /usr/libexec/xdg-desktop-portal-kde           1MiB |
|    0   N/A  N/A     21984      G   /usr/bin/konsole                              1MiB |
+---------------------------------------------------------------------------------------+
```
So, is it neccesary to get a new RPM each time kernel or nvidia drivers change? 

Thank you

R1chterScale, 2023-07-21:
Any NVIDIA driver version that is 525 or later requires the direct backend, use that and it should be fine

RafaelLinux, 2023-07-21:
I read [direct backend](https://github.com/elFarto/nvidia-vaapi-driver#direct-backend) in your page, but really I don't know what to do. It seems that info is related to driver compilation. As I (and many other) are using a precompiled RPM file, should we ask to have two distinct controllers available to download?

R1chterScale, 2023-07-21:
No, that's not what it says at all, you just need to set the environment variable.

elFarto, 2023-07-21:
The `NVD_BACKEND` environment variable is a runtime option, not a compile time one. And it's required to set it to `direct` until NVIDIA fix their drivers.

RafaelLinux, 2023-07-21:
I tried `NVD_BACKEND=direct` and even `export NVD_BACKEND=direct` and then I execute `vainfo`, but no changes about.
I tried `NVD_BACKEND=direct vainfo` too, but same result.
Must I do something more?

elFarto, 2023-07-21:
Hmm, It looks like libva is having an issue working out which driver to use. Try setting `LIBVA_DRIVER_NAME=nvidia`. This shouldn't really be needed with the version of libva you've got, so I'm not sure what's changed.

RafaelLinux, 2023-07-21:
Thanks. Something worked. Output of vainfo is now:

```
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.18.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit

```
while after was:

```
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.18.0
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
vaInitialize failed with error code -1 (unknown libva error),exit
```

elFarto, 2023-07-21:
You'll need `NVD_BACKEND=direct` set aswell. You can also set `NVD_LOG=1` to have the driver logs printed out.

RafaelLinux, 2023-07-21:
Both variables was exported previously:
```
env | grep -E "NVD|LIBVA"
LIBVA_DRIVER_NAME=nvidia
NVD_BACKEND=direct
```
Now, after  `export NVD_LOG=1` , ouput of `vainfo` is:
```
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.18.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
     22120.720833359 [11077-11077] ../src/vabackend.c:2165       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     22120.720846792 [11077-11077] ../src/vabackend.c:2174       __vaDriverInit_1_0 Now have 0 (0 max) instances
     22120.720856157 [11077-11077] ../src/vabackend.c:2200       __vaDriverInit_1_0 Selecting Direct backend
     22120.731952360 [11077-11077] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
     22120.731965315 [11077-11077] ../src/direct/nv-driver.c: 217            init_nvdriver Initing nvdriver...
     22120.731973098 [11077-11077] ../src/direct/nv-driver.c: 222            init_nvdriver Got dev info: 200 1 0 fe
     22120.731998506 [11077-11077] ../src/direct/nv-driver.c:  33          nv_alloc_object nv_alloc_object failed: -1 0 22
     22120.732004671 [11077-11077] ../src/direct/nv-driver.c: 243            init_nvdriver nv_alloc_object NV01_ROOT_CLIENT failed
     22120.732009776 [11077-11077] ../src/direct/nv-driver.c: 305            init_nvdriver Got error initing
     22120.732028936 [11077-11077] ../src/direct/nv-driver.c:  76            nv_rm_control nv_rm_control failed: -1 0 25
libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit

```

elFarto, 2023-07-21:
That issue is fixed in 0.0.10, and you're running 0.0.9. You will need to nudge the maintainer of that RPM to build the latest version, or build the driver yourself.

closed 2023-07-21 by RafaelLinux