elFarto/nvidia-vaapi-driver #224: [GFX1-]: vaapitest: VA-API test failed: failed to initialise VAAPI connection.

Bali10050, 2023-07-21:
`vainfo` says it's can decode, but nothing is using the gpu.

[image]

Please help.

elFarto, 2023-07-21:
It seems your GPU doesn't support AV1, which is the default codec Youtube will use. You can use `media.av1.enabled` Firefox config option to disable it, which should force Youtube to give you something else that your card supports.

rkoot, 2023-07-23:
You seem to use the EGL backend. You should use the direct backend instead as the EGL backend is broken atm.
append 'export NVD_BACKEND=direct' to your .bashrc, log out and back in and try again.

strahe, 2023-07-24:
Same issue.

Dirleye, 2023-07-24:
Same here, except on Wayland and with the direct backend. All described environment variables are set, along with options in about:config. This started a week or two ago and I can't remember if it was after a Firefox or an Nvidia driver update.

Bali10050, 2023-07-24:
I think I found the problem, `ffmpeg` doesn't have the `--enable-vaapi` flag on archlinux. Can I get help with that?

Bali10050, 2023-07-26:
Yes, ffmpeg was the problem.

DoraTheExploder, 2023-08-02:
I'm running up against this issue now as well. Am on Arch (-based Garuda) and looked at the package repos for ffmpeg and the flag in question seems to have never been there. (even though the driver formerly worked for me???) Is the prevailing advice just to rebuild ffmpeg with the flag --enable-vaapi? Appreciate any help, thanks.

RobertoAlda, 2023-12-14:
I have the same issue on:

- Ubuntu 22.04.3 LTS
- kernel version 5.15.0-91-generic
- Mozilla Firefox 120.0.1
- Nvidia T1000 4Gb
- Nvidia drivers 525.147.05 (installed with apt)

My vainfo output is:
```
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.17.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
```

Enviroments variables are:
```
MOZ_DISABLE_RDD_SANDBOX=1 
LIBVA_DRIVER_NAME=nvidia 
NVD_BACKEND=direct 
MOZ_X11_EGL=1 
```

Firefox flags are:
```
media.ffmpeg.vaapi.enabled=true
media.rdd-ffmpeg.enabled=true
media.av1.enabled=false
gfx.x11-egl.force-enabled=true
widget.dmabuf.force-enabled=true
```

but still i get this error:
```
[GFX1-]: vaapitest: ERROR
[GFX1-]: vaapitest: VA-API test failed: failed to initialise VAAPI connection.
```
 please help me...

Bali10050, 2023-12-14:
Did you build ffmpeg with the `--enable-vaapi` flag?

RobertoAlda, 2023-12-14:
I didn't built ffmpeg, i just installed it with apt and if i run ffmpeg it shows the --enable-vaapi flag.
It's not enough?

This is ffmpeg outputs:
```
ffmpeg version n4.3.1 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.5.0-3ubuntu1~18.04)
  configuration: --prefix= --prefix=/usr --disable-debug --disable-doc --disable-static --enable-cuda --enable-cuda-sdk --enable-cuvid --enable-libdrm --enable-ffplay --enable-gnutls --enable-gpl --enable-libass --enable-libfdk-aac --enable-libfontconfig --enable-libfreetype --enable-libmp3lame --enable-libnpp --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopus --enable-libpulse --enable-sdl2 --enable-libspeex --enable-libtheora --enable-libtwolame --enable-libv4l2 --enable-libvorbis --enable-libvpx --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxvid --enable-nonfree --enable-nvenc --enable-omx --enable-openal --enable-opencl --enable-runtime-cpudetect --enable-shared --enable-vaapi --enable-vdpau --enable-version3 --enable-xlib
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
Hyper fast Audio and Video encoder
usage: ffmpeg [options] [[infile options] -i infile]... {[outfile options] outfile}...
```
Do i have to clone ffmpeg git repo and  and build it with by myself? If so any hints to do it?
Thank you

Bali10050, 2023-12-14:
Try using these environment variables:
```
MOZ_DISABLE_RDD_SANDBOX=1
LIBVA_DRIVER_NAME=nvidia
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json
NVD_BACKEND=direct
MOZ_ENABLE_WAYLAND=1
```

RobertoAlda, 2023-12-15:
> Try using these environment variables:
> 
> ```
> MOZ_DISABLE_RDD_SANDBOX=1
> LIBVA_DRIVER_NAME=nvidia
> __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json
> NVD_BACKEND=direct
> MOZ_ENABLE_WAYLAND=1
> ```

I tried these env vars, but i stil get errors:
```
[GFX1-]: glxtest: libEGL no displayVENDOR
[GFX1-]: vaapitest: ERROR
[GFX1-]: vaapitest: VA-API test failed: failed to open renderDeviceFD.

[Parent 3579, Main Thread] WARNING: Failed to call GetIdletime(): GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.gnome.Mutter.IdleMonitor was not provided by any .service files
: 'glib warning', file /build/firefox/parts/firefox/build/toolkit/xre/nsSigHandlers.cpp:187

** (firefox:3579): WARNING **: 08:16:15.600: Failed to call GetIdletime(): GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.gnome.Mutter.IdleMonitor was not provided by any .service files
```

I can't figure out..

Bali10050, 2023-12-15:
Try setting `gfx.x11-egl.force-enabled` to `False`, if that doesn't help, I'll reopen the issue, because that's out of the range of my expertiese

RobertoAlda, 2023-12-15:
I tried  gfx.x11-egl.force-enabled to False, still the same output:

[GFX1-]: glxtest: libEGL no displayVENDOR
[GFX1-]: vaapitest: ERROR
[GFX1-]: vaapitest: VA-API test failed: failed to open renderDeviceFD.

[Parent 7014, Main Thread] WARNING: Failed to call GetIdletime(): GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.gnome.Mutter.IdleMonitor was not provided by any .service files
: 'glib warning', file /build/firefox/parts/firefox/build/toolkit/xre/nsSigHandlers.cpp:187

** (firefox:7014): WARNING **: 12:12:31.412: Failed to call GetIdletime(): GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.gnome.Mutter.IdleMonitor was not provided by any .service files

```

elFarto, 2023-12-17:
Can you check if `media.ffvpx.enabled` is enabled in `about:config`? That can help work around ffmpeg issues by using Firefox's own copy.

mariachini, 2023-12-19:
Got the same issue. Used commands from main repo, these from @Bali10050 , `media.ffvpx.enabled` is enabled and nothing. 
Latest Arch.

RobertoAlda, 2023-12-20:
I tried `media.ffvpx.enabled=true` but there's no active hw acceleration yet.
If helps my Firefox version is:

120.0.1 (64-bit)
Mozilla Firefox Snap for Ubuntu
canonical-002 - 1.0

satmandu, 2025-03-12:
I'm getting the same issues with firefox inside a wayland session using the nvidia 470 drivers. (The newest my kepler card supports.)

After disabling the wayland forcing in /usr/bin/firefox I see this:
```
MOZ_ENABLE_WAYLAND=0 MOZ_X11_EGL=0  NVD_BACKEND=direct MOZ_DISABLE_RDD_SANDBOX=1 __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json firefox
[GFX1-]: glxtest: libEGL no display
[GFX1-]: vaapitest: ERROR
[GFX1-]: vaapitest: VA-API test failed: failed to open renderDeviceFD.
```

However vainfo does show support:
```
 vainfo
Trying display: wayland
libva info: VA-API version 1.22.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.22 (libva 2.22.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
```

satmandu, 2025-03-12:
Hardware decoding DOES work when I startup in an Xorg session though.

I wonder if the larger issue here is from broken egl streams support in the Nvidia driver as per https://github.com/NVIDIA/egl-wayland/issues/167

satmandu, 2025-03-12:
(So maybe firefox somehow sets NVD_BACKEND=egl when checking for vaapi profiles when loaded with EGL?)