elFarto/nvidia-vaapi-driver #226: Firefox try to play video through your driver but fail

ElvenEleven11, 2023-07-27:
Hello Mr. elFarto!

Trying to make your driver work. I have Pascal Nvidia card (GP106) on Ubuntu 22.04.2 LTS so had to build driver by myself. I was missing package libdrm-dev, suggest to add it to ReadMe Building-apt-packages section. After installing libdrm-dev build was successful. Vainfo found driver and show codecs it support. I added environment variables, Firefox config keys, restarted PC. Than i start Firefox Nightly (ordinary one in stable branch says HW acceleration not supported in release or beta) with logging on: ~/Programs/firefox-nightly/firefox -MOZ_LOG="PlatformDecoderModule:5" and drag-and-drop there h264.mp4 file. It plays video but with software decoding, I see it with nvidia-settings app and with "nvidia-smi dmon". FF stdout (see attached log file) says it successfully see driver, try to decode video with it but something fails and it switches to software decoding. Closer to the end of file there is a string: [RDD 9971: MediaSupervisor #1]: D/PlatformDecoderModule FFMPEG: VAAPI is disabled by parent decoder module.

Also i have some errors in about:config (see attached log file), one of them is DMABUF = Blocklisted by gfxInfo =  and there is link to bug#1788573 at https://bugzilla.mozilla.org/show_bug.cgi?id=1788573
Another error is HARDWARE_VIDEO_DECODING  = Blocklisted by gfxInfo =  FEATURE_HARDWARE_VIDEO_DECODING_NO_LINUX_NVIDIA

If you need any additional info just say. Thanks in advance!

[image]

[firefox-nightly-stdout.log.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/12182984/firefox-nightly-stdout.log.txt)

[firefox-nightly-about-support.log.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/12182989/firefox-nightly-about-support.log.txt)

elFarto, 2023-07-30:
Do you have `widget.dmabuf.force-enabled` set to `true` in Firefox? That's required to work around the Firefox devs disabling it.

ElvenEleven11, 2023-07-31:
Yes I do. In **Firefox** i have some _Force disabled by gfxInfo_  but in **Firefox Nightly** only some _Blocklisted by gfxInfo_ . I wonder, blocklist mean disable or this is just a warning. 

 At FF Nightly start:
```
oleg@ubuntu:~$ ~/Programs/firefox-nightly/firefox -MOZ_LOG="PlatformDecoderModule:5, Dmabuf:5"
[Parent 6595: Main Thread]: D/Dmabuf DMABufDevice::Configure()
[Parent 6595: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 6595: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Parent 6595: Main Thread]: D/Dmabuf DMABuf is enabled

```
At h.264 play attempt:
```
[RDD 7069: MediaSupervisor #1]: D/PlatformDecoderModule FFMPEG: FFmpegVideoDecoder::FFmpegVideoDecoder MIME video/avc Codec ID 27
[Child 7047: RemVidChild]: D/PlatformDecoderModule RemoteMediaDataDecoder[7f8ff3b93510] ::RemoteMediaDataDecoder: 7f8ff3b93510 is created
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG: Initialising VA-API FFmpeg decoder
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   codec h264 : H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
[RDD 7069: MediaPDecoder #1]: D/Dmabuf DMABufDevice::Configure()
[RDD 7069: MediaPDecoder #1]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[RDD 7069: MediaPDecoder #1]: D/Dmabuf Using DRM device /dev/dri/renderD128
[RDD 7069: MediaPDecoder #1]: D/Dmabuf DMABuf is enabled
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
[Utility 6841: Main Thread]: D/PlatformDecoderModule FFmpeg decoder rejects requested type 'video/avc'

blah-blah supported codecs...

[Utility 6841: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG: Initialising FFmpeg decoder
[Utility 6841: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   codec aac : AAC (Advanced Audio Coding)
libva info: Found init function __vaDriverInit_1_0
[fluent] Missing message in locale ru: translations-panel-intro-description
[fluent] Missing message in locale ru: translations-panel-learn-more-link
[fluent] Missing message in locale ru: translations-panel-learn-more-link
[Utility 6841: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   FFmpeg decoder init successful.
[Child 7047: RemVidChild]: D/PlatformDecoderModule RemoteMediaDataDecoder[7f8ff3fa9eb0] ::operator(): 7f8ff3fa9eb0 RemoteDecoderChild has been initialized - description: ffmpeg audio decoder (Utility Generic remote), process: utility+audioDecoder_Generic, codec: aac
[Child 7047: MediaPDecoder #1]: V/PlatformDecoderModule AudioTrimmer[7f8ff1f823d0] ::PrepareTrimmers: sample[{0,48000},{1024,48000}] no trimming information
[Utility 6841: MediaPDecoder #1]: V/PlatformDecoderModule ProcessDecode: mDuration=21333µs ; mTime=0µs ; mTimecode=0µs
[Utility 6841: MediaPDecoder #1]: V/PlatformDecoderModule ProcessDecode: mDuration=21333µs ; mTime=0µs ; mTimecode=0µs
[Utility 6841: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG: Packet decoded: [{0,48000}, {1024,48000}] (21333us, 1024 frames)
[Child 7047: MediaPDecoder #1]: V/PlatformDecoderModule AudioTrimmer[7f8ff1f823d0] ::HandleDecodedResult: sample[{0,48000}, {1024,48000}] (decoded[{0,48000}, {1024,48000}] no trimming needed
libva info: va_openDriver() returns 0
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG: FFmpegVideoDecoder::GetAcceleratedFormats()
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   Profile H264Main:
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:     codec h264 format nv12              3         12
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:     codec h264 format p010le            3         15
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   Profile H264High:
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:     codec h264 format nv12              3         12
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:     codec h264 format p010le            3         15
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   Profile H264ConstrainedBaseline:
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:     codec h264 format nv12              3         12
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:     codec h264 format p010le            3         15
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   Supported accelerated formats:
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:       h264
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG:   VA-API FFmpeg init successful

blah-blah not very interesting

[RDD 7069: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG: Frame decode finished, time 128.04 ms averange decode time 128.04 ms decoded 1 frames
[RDD 7069: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   slow decode: failed to decode in time, frame duration 33.37 ms, decode time 128.04
[RDD 7069: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   frames: all decoded 1 late decoded 1 over averange 1
[RDD 7069: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG: VA-API Got one frame output with pts=0 dts=33366 duration=33366 opaque=-9223372036854775808
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG: ProcessFlush()
[RDD 7069: MediaPDecoder #1]: D/PlatformDecoderModule FFMPEG: FFmpegDataDecoder: flushing buffers
[... cut ...]
[RDD 7069: MediaSupervisor #1]: D/PlatformDecoderModule FFMPEG: FFmpegVideoDecoder::FFmpegVideoDecoder MIME video/avc Codec ID 27
[RDD 7069: MediaSupervisor #1]: D/PlatformDecoderModule FFMPEG: VAAPI is disabled by parent decoder module.

```
and than it switches to SW decoder. Is it ffmpeg --enable-vaapi thing? If yes, would be nice to get a guide / link to how-to reconfigure ffmpeg.

And thanks for your work, I think it's really cool! 

Full log here:
[firefox-nightly-stdout.log2.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/12212126/firefox-nightly-stdout.log2.txt)

elFarto, 2023-07-31:
Interesting, it seems ok up to this part of the log:
```
[RDD 7069: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG: Frame decode finished, time 128.04 ms averange decode time 128.04 ms decoded 1 frames
[RDD 7069: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   slow decode: failed to decode in time, frame duration 33.37 ms, decode time 128.04
[RDD 7069: MediaPDecoder #2]: D/PlatformDecoderModule FFMPEG:   frames: all decoded 1 late decoded 1 over averange 1
```
Can you try running that again, but with `NVD_LOG=1`? 128ms is extremely long for a decode.

ElvenEleven11, 2023-08-01:
Yup
[firefox-nightly-stdout.log3.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/12226709/firefox-nightly-stdout.log3.txt)

elFarto, 2023-08-05:
You need to set `NVD_BACKEND=direct` as it's currently trying to use the EGL backend which doesn't work.
```
218.915772474 [4388-4416] ../src/vabackend.c:2203       __vaDriverInit_1_0 Selecting EGL backend
```

ElvenEleven11, 2023-08-05:
Finally! It works. But only with h264 codec. My card GeForce GTX 1060 also support decoding of h265 and VP9 (8bit). h265 should work out of the box (vainfo says it support 3 HEVC profiles). It seems it doesn't even try to use HW acceleration when i drag and drop h265.mp4, see log.
[firefox-nightly-stdout.log4hevc.txt](https://github.com/elFarto/nvidia-vaapi-driver/files/12266814/firefox-nightly-stdout.log4hevc.txt)

Anyway I don't care much of h265, but VP9 is what I want for YouTube. As you see in screenshot in first post I do not have `gstreamer-codecparsers-1.0` that is, you say, needed for VP9. So I have to install `libgstreamer-plugins-bad1.0-dev` which pulls 101 new packages and 330Mb and rebuild driver. Or use `h264ify` FF addon. OK, will think what I prefer. Any chance to install just those codecparsers without other 100 packages?

By the way, it works only in FF Nightly, FF stable says it can't connect to VA-API and HW_decoding force disabled by gfxinfo.

elFarto, 2023-08-05:
I'm afraid not. I'm pretty sure `gstreamer-codecparsers-1.0` requires gstreamer itself, so there's not getting around it.

I'm not sure why it wouldn't be working in FF stable. Can you paste the `HARDWARE_VIDEO_DECODING` part of `about:support`.

ElvenEleven11, 2023-08-05:
[image]

Also there is text in WebGPU adapter by default:
{
  "navigator.gpu.requestAdapter({})": {
    "not_supported": "WebGPU is not yet available in Release or Beta builds."
  }
}

marceldev89, 2023-08-15:
@ElvenEleven11 I had the same issue, try with MOZ_DISABLE_RDD_SANDBOX=1. Works for me now. I don't remember where but I read somewhere that it wasn't required anymore but I guess it is. 🙈

ElvenEleven11, 2023-08-16:
@marceldev89 I have this in my `/etc/environment`, and I see it in `about:support - Environment` as well. But thanks!

LocalPhysicsEnthusiast, 2024-04-22:
I'm still having the same issue. It says this in my about:support
[image]
This is the log when i start nightly:

```[Parent 17315: Main Thread]: D/Dmabuf DMABufDevice::Configure()
[Parent 17315: Main Thread]: D/Dmabuf Loading DMABuf system library libgbm.so.1 ...
[Parent 17315: Main Thread]: D/Dmabuf Using DRM device /dev/dri/renderD128
[Parent 17315: Main Thread]: D/Dmabuf wl_drm is available.
[Parent 17315: Main Thread]: D/Dmabuf zwp_linux_dmabuf_v1 is available.
[Parent 17315: Main Thread]: D/Dmabuf DMABuf is enabled 
```

This is when i try to watch a video on youtube
```
[RDD 17686: MediaPDecoder #1]: D/Dmabuf VideoFrameSurface: VAAPI releasing dmabuf surface UID 31 FFMPEG ID 0xe aForFrameRecycle 1 mLib 7cd3d4e0b3e0 mAVHWFrameContext 7cd397595b80 mHWAVBuffer 7cd397595c80
[RDD 17686: MediaPDecoder #1]: D/Dmabuf DMABufSurfaceYUV::ReleaseSurface() UID 31
[RDD 17686: MediaPDecoder #1]: D/Dmabuf DMABufSurfaceYUV::ReleaseTextures() UID 31
[RDD 17686: MediaPDecoder #1]: D/Dmabuf DMABufSurface::ReleaseDMABuf() UID 31
[RDD 17686: MediaPDecoder #1]: D/Dmabuf Using VA-API DMABufSurface UID 31 FFMPEG ID 0x9
[RDD 17686: MediaPDecoder #1]: D/Dmabuf Used HW surface UID 29 FFMPEG ID 0x10
[RDD 17686: MediaPDecoder #1]: D/Dmabuf Used HW surface UID 30 FFMPEG ID 0x11
[RDD 17686: MediaPDecoder #1]: D/Dmabuf Used HW surface UID 32 FFMPEG ID 0xd
[RDD 17686: MediaPDecoder #1]: D/Dmabuf Used HW surface UID 33 FFMPEG ID 0xb
[RDD 17686: MediaPDecoder #1]: D/Dmabuf Used HW surface UID 34 FFMPEG ID 0x12
[RDD 17686: MediaPDecoder #1]: D/Dmabuf Used HW surface UID 35 FFMPEG ID 0xc
[RDD 17686: MediaPDecoder #1]: D/Dmabuf Surface pool size 8 used copied 0 used ffmpeg 6 (max 18) free ratio 0.666667
[RDD 17686: MediaPDecoder #1]: D/Dmabuf DMABufSurfaceYUV::UpdateYUVData() UID 31 copy 0
[RDD 17686: MediaPDecoder #1]: D/Dmabuf DMABufSurfaceYUV::ImportPRIMESurfaceDescriptor() UID 31
[RDD 17686: MediaPDecoder #1]: D/Dmabuf     plane 0 size 2560 x 1440 format 20203852
[RDD 17686: MediaPDecoder #1]: D/Dmabuf     plane 1 size 1280 x 720 format 38384752
[RDD 17686: MediaPDecoder #1]: D/Dmabuf VideoFrameSurface: VAAPI locking dmabuf surface UID 31 FFMPEG ID 0x9 mAVHWFrameContext 7cd397595b80 mHWAVBuffer 7cd397595c80
[RDD 17686: MediaPDecoder #1]: D/Dmabuf DMABUFSurfaceImage::DMABUFSurfaceImage (7cd3b35d98d0) aSurface 7cd392455ac0 UID 31
[RDD 17686: MediaSupervisor #5]: D/Dmabuf DMABufSurfaceYUV::Serialize() UID 31
[RDD 17686: MediaSupervisor #5]: D/Dmabuf DMABufSurfaceYUV::Serialize() UID 31
[Parent 17315: Compositor]: D/Dmabuf DMABufSurfaceYUV::ImportSurfaceDescriptor() UID 31
[Parent 17315: Compositor]: D/Dmabuf     plane 0 fd 263 size 2560 x 1440 format 20203852
[Parent 17315: Compositor]: D/Dmabuf     plane 1 fd 264 size 1280 x 720 format 38384752
[Parent 17315: Renderer]: D/Dmabuf DMABufSurfaceYUV::CreateTexture() UID 32 plane 0
[Parent 17315: Renderer]: D/Dmabuf DMABufSurfaceYUV::CreateEGLImage() UID 32 plane 0
[Parent 17315: Renderer]: D/Dmabuf   Success.
[Parent 17315: Renderer]: D/Dmabuf DMABufSurfaceYUV::CreateTexture() UID 32 plane 1
[Parent 17315: Renderer]: D/Dmabuf DMABufSurfaceYUV::CreateEGLImage() UID 32 plane 1
[Parent 17315: Renderer]: D/Dmabuf   Success.
[Child 18815: MediaPDecoder #8]: D/PlatformDecoderModule AudioTrimmer[775912df4390] ::Decode: AudioTrimmer::Decode
[Child 18815: MediaPDecoder #8]: V/PlatformDecoderModule AudioTrimmer[775912df4390] ::PrepareTrimmers: sample[{28641000,1000000},{28661000,1000000}] no trimming information
[Utility 17534: MediaPDecoder #2]: V/PlatformDecoderModule ProcessDecode: mDuration=20000µs ; mTime=28641000µs ; mTimecode=28641000µs
[Utility 17534: MediaPDecoder #2]: V/PlatformDecoderModule ProcessDecode: mDuration=20000µs ; mTime=28641000µs ; mTimecode=28641000µs
[Child 18815: MediaPDecoder #7]: D/PlatformDecoderModule AudioTrimmer[775912df4390] ::HandleDecodedResult: HandleDecodedResults: 1 decoded data, 1 trimmers
[Child 18815: MediaPDecoder #7]: V/PlatformDecoderModule AudioTrimmer[775912df4390] ::HandleDecodedResult: decoded buffer [{28641000,1000000}, {28661000,1000000}] doesn't need trimming
[Child 18815: MediaPDecoder #8]: D/PlatformDecoderModule AudioTrimmer[775912df4390] ::Decode: AudioTrimmer::Decode
[Child 18815: MediaPDecoder #8]: V/PlatformDecoderModule AudioTrimmer[775912df4390] ::PrepareTrimmers: sample[{28661000,1000000},{28681000,1000000}] no trimming information
[Utility 17534: MediaPDecoder #1]: V/PlatformDecoderModule ProcessDecode: mDuration=20000µs ; mTime=28661000µs ; mTimecode=28661000µs
[Utility 17534: MediaPDecoder #1]: V/PlatformDecoderModule ProcessDecode: mDuration=20000µs ; mTime=28661000µs ; mTimecode=28661000µs
```

When I look at my GPU usage it doesnt change at all, and since the about:support still returns that it wont use the nvidia driver it makes me think this is just some nightly issue? Since the other person in this thread also used nightly.

mirh, 2024-07-13:
I feel like you hijacked an issue that was probably about forgetting `MOZ_DISABLE_RDD_SANDBOX=1`, with one that is probably about wayland

littleblack111, 2025-07-11:
1 year later, the problem is still here... (although for some reason the chromium bug is fixed?)

closed 2025-08-07 by elFarto