elFarto/nvidia-vaapi-driver #233: HW Acceleration on Ubuntu 22.04

xphoniex, 2023-08-11:
I was looking at my mpv logs and noticed HW Acceleration is not working for me:

```bash
[vd] Opening decoder h264
[vd] Looking at hwdec h264-vaapi...
[vo/gpu] Loading hwdec driver 'vaapi-egl'
[vo/gpu/vaapi-egl] using VAAPI EGL interop
[vo/gpu/vaapi-egl] Trying to open a x11 VA display...
[vo/gpu/vaapi-egl/vaapi] libva: VA-API version 1.14.0
[vo/gpu/vaapi-egl/vaapi] libva: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
[vo/gpu/vaapi-egl/vaapi] libva: va_openDriver() returns -1
[vo/gpu] Loading failed.
[vo/gpu] Loading hwdec driver 'cuda-nvdec'
[vo/gpu] Loading hwdec driver 'drmprime-drm'
[vo/gpu/drmprime-drm] Failed to retrieve DRM fd from native display.
[vo/gpu] Loading failed.
[vd] Could not create device.
[vd] No hardware decoding available for this codec.
[vd] Using software decoding.
```

`vainfo` was returning something similar: `va_openDriver() returns -1`

I'm on Ubuntu 22.04 LTS Jammy, which the package for, does not exist. You have it for 22.10 but not 22.04.

So I built it from source, (also had to `apt install libdrm-dev`, maybe add it to list?) and now:

```bash
$ vainfo
libva info: VA-API version 1.14.0
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
```

I have installed my driver using `NVIDIA-Linux-x86_64-525.85.05.run` file, and don't have anything else.

now in mpv, I get:

```bash
[vd] Opening decoder h264
[vd] Looking at hwdec h264-vaapi...
[vo/gpu] Loading hwdec driver 'vaapi-egl'
[vo/gpu/vaapi-egl] using VAAPI EGL interop
[vo/gpu/vaapi-egl] Trying to open a x11 VA display...
[vo/gpu/vaapi-egl/vaapi] libva: VA-API version 1.14.0
[vo/gpu/vaapi-egl/vaapi] libva: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
[vo/gpu/vaapi-egl/vaapi] libva: Found init function __vaDriverInit_1_0
[vo/gpu/vaapi-egl/vaapi] libva: va_openDriver() returns 0
[vo/gpu/vaapi-egl/vaapi] Initialized VAAPI: version 1.14
[ffmpeg] AVHWDeviceContext: VAAPI driver: VA-API NVDEC driver [egl backend].
[ffmpeg] AVHWDeviceContext: Driver not found in known nonstandard list, using standard behaviour.
[vo/gpu/vaapi-egl] Going to probe surface formats (may log bogus errors)...
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
[vo/gpu/vaapi-egl] Supported formats:
[vo/gpu/vaapi-egl] Done probing surface formats.
[vo/gpu] Loading failed.
[vo/gpu] Loading hwdec driver 'cuda-nvdec'
[vo/gpu] Loading hwdec driver 'drmprime-drm'
[vo/gpu/drmprime-drm] Failed to retrieve DRM fd from native display.
[vo/gpu] Loading failed.
[vd] Could not create device.
[vd] No hardware decoding available for this codec.
[vd] Using software decoding.
```

here's nvidia-smi:

```bash
$ nvidia-smi
Fri Aug 11 23:22:20 2023
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.85.05    Driver Version: 525.85.05    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0  On |                  Off |
| 34%   41C    P8    22W / 450W |    941MiB / 24564MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+

+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1183      G   /usr/lib/xorg/Xorg                242MiB |
|    0   N/A  N/A      1986      G   alacritty                          15MiB |
|    0   N/A  N/A      6223      G   ...7/usr/lib/firefox/firefox      179MiB |
|    0   N/A  N/A     22890      G   ...735348111746672520,262144       57MiB |
|    0   N/A  N/A    169356    C+G   mpv                               401MiB |
|    0   N/A  N/A    234712      G   alacritty                          15MiB |
|    0   N/A  N/A    238420      G   alacritty                          23MiB |
+-----------------------------------------------------------------------------+
```

Given the line `[vo/gpu/drmprime-drm] Failed to retrieve DRM fd from native display.`, I thought this was a drm issue, so I rebooted and added `nvidia_drm.modeset=1` at the end of my `linux ...` line in grub.

I tried both `nvidia_drm.modeset` and `nvidia-drm.modeset`, but mpv is still using software decoding, and drm seems to be enabled:

```console
$ sudo cat /sys/module/nvidia_drm/parameters/modeset
Y
```

what's missing? what do I need to do?

xphoniex, 2023-08-13:
Do I need to download CUDA for this to work?

xphoniex, 2023-08-13:
I can set `NVD_BACKEND=direct` and get this:

```bash
[vd] Opening decoder h264
[vd] Looking at hwdec h264-vaapi...
[vo/gpu] Loading hwdec driver 'vaapi-egl'
[vo/gpu/vaapi-egl] using VAAPI EGL interop
[vo/gpu/vaapi-egl] Trying to open a x11 VA display...
[vo/gpu/vaapi-egl/vaapi] libva: VA-API version 1.14.0
[vo/gpu/vaapi-egl/vaapi] libva: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
[vo/gpu/vaapi-egl/vaapi] libva: Found init function __vaDriverInit_1_0
[vo/gpu/vaapi-egl/vaapi] libva: va_openDriver() returns 0
[vo/gpu/vaapi-egl/vaapi] Initialized VAAPI: version 1.14
[ffmpeg] AVHWDeviceContext: VAAPI driver: VA-API NVDEC driver [direct backend].
[ffmpeg] AVHWDeviceContext: Driver not found in known nonstandard list, using standard behaviour.
[vo/gpu/vaapi-egl] Going to probe surface formats (may log bogus errors)...
[vo/gpu/vaapi-egl] Supported formats:
[vo/gpu/vaapi-egl]  nv12
[vo/gpu/vaapi-egl]  p010
[vo/gpu/vaapi-egl]  yuv444p
[vo/gpu/vaapi-egl] Done probing surface formats.
[vo/gpu] Loading hwdec driver 'cuda-nvdec'
[vo/gpu] Loading hwdec driver 'drmprime-drm'
[vo/gpu/drmprime-drm] Failed to retrieve DRM fd from native display.
[vo/gpu] Loading failed.
[vd] Trying hardware decoding via h264-vaapi.
[vd] Selected codec: h264 (H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10)
[vf] User filter list:
[vf]   (empty)
[ad] Codec list:
[ad]     aac - AAC (Advanced Audio Coding)
[ad]     aac_fixed (aac) - AAC (Advanced Audio Coding)
[ad] Opening decoder aac
[ad] Requesting 1 threads for decoding.
[ad] Selected codec: aac (AAC (Advanced Audio Coding))
[af] User filter list:
[af]   (empty)
[cplayer] Starting playback...
[af] [in] 44100Hz stereo 2ch floatp
[af] [userspeed] 44100Hz stereo 2ch floatp
[af] [userspeed] (disabled)
[af] [convert] 44100Hz stereo 2ch floatp
[vd] Pixel formats supported by decoder: vdpau cuda vaapi_vld yuv420p
[vd] Codec profile: High (0x64)
[vd] Requesting pixfmt 'vaapi_vld' from decoder.
[ffmpeg/video] h264: Reinit context to 1280x640, pix_fmt: vaapi_vld
[vd] Using hardware decoding (vaapi).
```

but the screen is just green. (I can also see Video Engine Utilization at 0% - 1%)

elFarto, 2023-08-13:
It's pointless to use this driver with MPV, as that supports using NVDEC directly which is far more efficient.

xphoniex, 2023-08-13:
But any idea why VA-API is not working?

In mpv, if I set `hwdec=auto`, I get NVDEC -- it's slower than CPU though :disappointed: .

Marking the issue as complete, but it'd still be nice to know what I did wrong, when setting up VA-API, in case some one comes across this issue via Google, etc.

closed 2023-08-13 by xphoniex

elFarto, 2023-08-13:
You could try running it with `NVD_LOG=1` set and see if there's anything that points to an issues. But it's probably a question for the MPV devs.

xphoniex, 2023-08-13:
<details><summary>mpv logs with NVD_LOG=1</summary>

```bash
[vd] Opening decoder h264
[vd] Looking at hwdec h264-vaapi...
[vo/gpu] Loading hwdec driver 'vaapi-egl'
[vo/gpu/vaapi-egl] using VAAPI EGL interop
[vo/gpu/vaapi-egl] Trying to open a x11 VA display...
[vo/gpu/vaapi-egl/vaapi] libva: VA-API version 1.14.0
[vo/gpu/vaapi-egl/vaapi] libva: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
[vo/gpu/vaapi-egl/vaapi] libva: Found init function __vaDriverInit_1_0
     25194.658334156 [104862-104873] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     25194.658337500 [104862-104873] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
     25194.658338768 [104862-104873] ../src/vabackend.c:2203       __vaDriverInit_1_0 Selecting EGL backend
     25194.658344725 [104862-104873] ../src/export-buf.c: 132       findGPUIndexFromFd Defaulting to CUDA GPU ID 0. Use NVD_GPU to select a specific CUDA GPU
     25194.658347028 [104862-104873] ../src/export-buf.c: 149       findGPUIndexFromFd Looking for GPU index: 0
     25194.658497991 [104862-104873] ../src/export-buf.c: 161       findGPUIndexFromFd Found 3 EGL devices
     25194.658534345 [104862-104873] ../src/export-buf.c: 170       findGPUIndexFromFd Got EGL_CUDA_DEVICE_NV value '0' for EGLDevice 0
     25194.658535958 [104862-104873] ../src/export-buf.c: 191       findGPUIndexFromFd Selecting EGLDevice 0
     25194.659030167 [104862-104873] ../src/export-buf.c: 260         egl_initExporter Driver supports 16-bit surfaces
[vo/gpu/vaapi-egl/vaapi] libva: va_openDriver() returns 0
[vo/gpu/vaapi-egl/vaapi] Initialized VAAPI: version 1.14
     25194.699817867 [104862-104873] ../src/vabackend.c:1409      nvQueryImageFormats In nvQueryImageFormats
[ffmpeg] AVHWDeviceContext: VAAPI driver: VA-API NVDEC driver [egl backend].
[ffmpeg] AVHWDeviceContext: Driver not found in known nonstandard list, using standard behaviour.
[vo/gpu/vaapi-egl] Going to probe surface formats (may log bogus errors)...
     25194.706796338 [104862-104873] ../src/vabackend.c: 648           nvCreateConfig got profile: 0 with 0 attributes
     25194.706802788 [104862-104873] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 1 (8) (nil) 0
     25194.706804600 [104862-104873] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 1 (8) 0x7fa6590e5d80 8
     25194.707285648 [104862-104873] ../src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
     25194.707294425 [104862-104873] ../src/vabackend.c: 931        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7fa658fee690)
     25194.707297827 [104862-104873] ../src/vabackend.c:1503            nvDeriveImage In nvDeriveImage
     25194.707329466 [104862-104873] ../src/export-buf.c:  56                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x1)

     25194.707333391 [104862-104873] ../src/export-buf.c:  56                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7fa65852e840)

     25194.707335320 [104862-104873] ../src/export-buf.c:  56                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7fa658a95950)

     25194.707338190 [104862-104873] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: -1 (0x7fa658fee690)
     25194.707339986 [104862-104873] ../src/export-buf.c: 528       egl_realiseSurface No free surfaces found
     25194.707397040 [104862-104873] ../src/export-buf.c: 440 egl_allocateBackingImage Presenting frame -1 128x128 (0x7fa658fee690, 0x7fa658ff0a20, 0x7fa658ff14f0)
     25194.707399807 [104862-104873] ../src/export-buf.c: 441 egl_allocateBackingImage CUDA ERROR 'invalid resource handle' (400)

     25194.707401341 [104862-104873] ../src/export-buf.c:  95                reconnect Reconnecting to stream
     25194.707435932 [104862-104873] ../src/export-buf.c: 115                reconnect CUDA ERROR 'operation not supported' (801)

     25194.707437775 [104862-104873] ../src/export-buf.c: 533       egl_realiseSurface Unable to realise surface: 0x7fa658fee690 (-1)
     25194.707439259 [104862-104873] ../src/vabackend.c:2060    nvExportSurfaceHandle Unable to export surface
[vo/gpu/vaapi-egl] vaExportSurfaceHandle() failed (resource allocation failed)
     25194.707449688 [104862-104873] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fa658fee690)
     25194.707451300 [104862-104873] ../src/export-buf.c: 335 egl_detachBackingImageFromSurface Cannot detach NULL BackingImage from Surface
     25194.707454873 [104862-104873] ../src/vabackend.c: 931        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x7fa658fef670)
     25194.707456572 [104862-104873] ../src/vabackend.c:1503            nvDeriveImage In nvDeriveImage
     25194.707463940 [104862-104873] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: -1 (0x7fa658fef670)
     25194.707465301 [104862-104873] ../src/export-buf.c: 528       egl_realiseSurface No free surfaces found
     25194.707491716 [104862-104873] ../src/export-buf.c: 440 egl_allocateBackingImage Presenting frame -1 128x128 (0x7fa658fef670, 0x7fa658ff0690, 0x7fa658ff5ac0)
     25194.707516110 [104862-104873] ../src/export-buf.c: 441 egl_allocateBackingImage CUDA ERROR 'unknown error' (999)

[... cut ...]
[vf] [in] 1280x640 yuv420p bt.709/bt.709/bt.1886/limited/display SP=1.000000 CL=mpeg2/4/h264
[vf] [userdeint] 1280x640 yuv420p bt.709/bt.709/bt.1886/limited/display SP=1.000000 CL=mpeg2/4/h264
[vf] [userdeint] (disabled)
[vf] [autorotate] 1280x640 yuv420p bt.709/bt.709/bt.1886/limited/display SP=1.000000 CL=mpeg2/4/h264
[vf] [autorotate] (disabled)
[vf] [convert] 1280x640 yuv420p bt.709/bt.709/bt.1886/limited/display SP=1.000000 CL=mpeg2/4/h264
[vf] [convert] (disabled)
[vf] [out] 1280x640 yuv420p bt.709/bt.709/bt.1886/limited/display SP=1.000000 CL=mpeg2/4/h264
```

</details>

elFarto, 2023-08-13:
You'll need `NVD_BACKEND=direct` set too.

xphoniex, 2023-08-13:
<details><summary>mpv logs with NVD_BACKEND=direct NVD_LOG=1</summary>

```bash
[vd] Opening decoder h264
[vd] Looking at hwdec h264-vaapi...
[vo/gpu] Loading hwdec driver 'vaapi-egl'
[vo/gpu/vaapi-egl] using VAAPI EGL interop
[vo/gpu/vaapi-egl] Trying to open a x11 VA display...
[vo/gpu/vaapi-egl/vaapi] libva: VA-API version 1.14.0
[vo/gpu/vaapi-egl/vaapi] libva: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
[vo/gpu/vaapi-egl/vaapi] libva: Found init function __vaDriverInit_1_0
     28632.695734047 [126019-126030] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
     28632.695737146 [126019-126030] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
     28632.695738846 [126019-126030] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
     28632.695750408 [126019-126030] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
     28632.695752089 [126019-126030] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
     28632.695754317 [126019-126030] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 100 1 2 6
     28632.695765034 [126019-126030] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: , major version: 0
[vo/gpu/vaapi-egl/vaapi] libva: va_openDriver() returns 0
[vo/gpu/vaapi-egl/vaapi] Initialized VAAPI: version 1.14
     28632.726787365 [126019-126030] ../src/vabackend.c:1409      nvQueryImageFormats In nvQueryImageFormats
[ffmpeg] AVHWDeviceContext: VAAPI driver: VA-API NVDEC driver [direct backend].
[ffmpeg] AVHWDeviceContext: Driver not found in known nonstandard list, using standard behaviour.
[vo/gpu/vaapi-egl] Going to probe surface formats (may log bogus errors)...
     28632.734805982 [126019-126030] ../src/vabackend.c: 648           nvCreateConfig got profile: 0 with 0 attributes
     28632.734812650 [126019-126030] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 1 (8) (nil) 0
     28632.734815255 [126019-126030] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 1 (8) 0x7f3f7d0268c0 8
     28632.735280323 [126019-126030] ../src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4080, height: 16 - 4080
     28632.735288067 [126019-126030] ../src/vabackend.c: 931        nvCreateSurfaces2 Creating surface 128x128, format 1 (0x7f3f7d027930)
     28632.735290760 [126019-126030] ../src/vabackend.c:1503            nvDeriveImage In nvDeriveImage
     28632.735323648 [126019-126030] ../src/direct/direct-export-buf.c:  42                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x1)

     28632.735326717 [126019-126030] ../src/direct/direct-export-buf.c:  42                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7f3f7c52cf50)

     28632.735329035 [126019-126030] ../src/direct/direct-export-buf.c:  42                    debug [EGL] eglDestroyImageKHR: EGL_BAD_PARAMETER error: In eglDestroyImage: Invalid EGLImage (0x7f3f7c9d31d0)

     28632.735331755 [126019-126030] ../src/vabackend.c:2055    nvExportSurfaceHandle Exporting surface: -1 (0x7f3f7d027930)
     28632.735333952 [126019-126030] ../src/direct/direct-export-buf.c: 198 direct_allocateBackingImage Allocating BackingImages: 0x7f3f7cf30be0 128x128
     28632.735335753 [126019-126030] ../src/direct/nv-driver.c: 423              alloc_image Calculated GOB size: 64x128 (0x4)
     28632.735337176 [126019-126030] ../src/direct/nv-driver.c: 432              alloc_image Aligned image size: 128x128 = 16384
     28632.735394443 [126019-126030] ../src/direct/nv-driver.c: 508              alloc_image created image: 128x128 300000000606014 128 4000
     28632.735396085 [126019-126030] ../src/direct/nv-driver.c: 423              alloc_image Calculated GOB size: 64x64 (0x3)
     28632.735397165 [126019-126030] ../src/direct/nv-driver.c: 432              alloc_image Aligned image size: 128x64 = 8192
     28632.735426301 [126019-126030] ../src/direct/nv-driver.c: 508              alloc_image created image: 64x64 300000000606013 128 2000
     28632.735427702 [126019-126030] ../src/direct/direct-export-buf.c: 204 direct_allocateBackingImage Importing images
     28632.735428839 [126019-126030] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 128x128 = 4000
     28632.735482498 [126019-126030] ../src/direct/direct-export-buf.c: 118           import_to_cuda importing memory size: 64x64 = 2000
     28632.735531727 [126019-126030] ../src/vabackend.c:2068    nvExportSurfaceHandle Exporting with 128 128 0 128 300000000606014 0 128 300000000606013
     28632.735671564 [126019-126030] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7f3f7d027930)
     28632.735673289 [126019-126030] ../src/direct/direct-export-buf.c: 243      destroyBackingImage Destroying BackingImage: 0x7f3f7cf30be0
     28632.735730972 [126019-126030] ../src/vabackend.c: 931        nvCreateSurfaces2 Creating surface 128x128, format 100 (0x7f3f7cf32060)
     28632.735733049 [126019-126030] ../src/vabackend.c:1503            nvDeriveImage In nvDeriveImage
[... cut ...]
[vo/gpu/opengl] [ 29] color.rgb = mat3(colormatrix) * color.rgb + colormatrix_c;
[vo/gpu/opengl] [ 30] color.a = 1.0;
[vo/gpu/opengl] [ 31] // color mapping
[vo/gpu/opengl] [ 32] color.rgb *= vec3(1.000000);
[vo/gpu/opengl] [ 33] color.rgb *= vec3(1.000000);
[vo/gpu/opengl] [ 34] out_color = color;
[vo/gpu/opengl] [ 35] }
[vo/gpu/opengl] shader link log (status=1): 
[vo/gpu/opengl] after rendering: OpenGL error INVALID_OPERATION.
[cplayer] first video frame after restart shown
[cplayer] audio ready
```

</details>

elFarto, 2023-08-13:
It seems you have an EGL issue. What version of MPV do you have. You need at least v0.35.0 for this to work.

xphoniex, 2023-08-13:
Yeah, I'm on mpv v0.34.1

--

I'm somewhat disappointed as even with NVDEC, which you said is more efficient than VA-API, I still get frame drops and jumps when I seek to another position when playing x265/HEVC file.

I have 13900k/4090, which is the best consumer grade hardware, and still can't get a smooth playback. Is there anything else I can tune?

elFarto, 2023-08-13:
v0.34.1 doesn't have the patch necessary for it to work with this driver.

As for your performance issue, I'm not sure what could be causing that. Hardware that powerful should have no problem playing back any video. My 5900X/1060 can easily do multiple 4k HEVC streams without issue.

Do you the same issue if you use software decoding? If so, then your issue may be on the presentation side of things, rather than the decode.

xphoniex, 2023-08-13:
I don't even do 4k, mine is 720p x265. I can play just fine, my issue is when I move to another position on the timeline, e.g. jumping a minute ahead/behind. That's when it slows down, starts to drop frames, and jumps to compensate, until it becomes smooth again.

asynapse, 2023-10-07:
I encountered the same unfortunateness with video playback in mpv, where when starting a video, or after jumping forward or back, playback would slow down for a while, and eventually become smooth.

I started having the same issue with mpv after refreshing my desktop, going from an Nvidia Geforce 1060 to using amdgpu (integrated graphics on my new processor). I had upgraded from ubuntu 20.04 to 22.04 as part of the hardware refresh, thinking that I needed to in order to get a more recent version of amdgpu. I initially attributed the mpv issue to be due to using the integrated graphics, as mpv playback was fine previously, but I had the same mpv playback issue after throwing the 1060 back in my refreshed system, running 22.04.

I installed the mpv and libcdio18 packages from Ubuntu 20.04 on my 22.04 system (obtained by running 'apt download <package>' for each package on a 20.04 system, and installed with 'dpkg -i <packageFile.deb>), and playback is smooth again like it was on 20.04, I marked the mpv package as held to prevent it from being upgraded, I'll probably take a crack at compiling a newer version in the near future:
```
$ dpkg -l mpv libcdio18
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name            Version         Architecture Description
+++-===============-===============-============-======================================
ii  libcdio18:amd64 2.0.0-2         amd64        library to read and control CD-ROM
hi  mpv             0.32.0-1ubuntu1 amd64        video player based on MPlayer/mplayer2
```

I think the version of mpv (0.34.1) that comes with ubuntu 22.04 is wonky :-|

xphoniex, 2023-10-07:
@asynapse Thanks for sharing this, I'll give your solution a try.

I noticed a few weeks ago that this issue is less prevalent on Windows, so I tried the updated mpv from snap (0.36) but that didn't help either..