elFarto/nvidia-vaapi-driver #234: Doesn't work on a thinkpad P14 with nvidia quadro T500

sebastiangomezlopez, 2023-08-14:
Hi I've installed everything ans set all the env vars according to the installation manual. 

Here's what vainfo returns
```shell
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints

``` 

If I understand the output correctly, it should be working just fine. However the GPU utilization is 0% when i play youtube videos in firefox

elFarto, 2023-08-14:
Try setting `NVD_BACKEND=direct`, this is required as the NVIDIA drivers have a bug that prevents the EGL backend from working.

sebastiangomezlopez, 2023-08-14:
Thanks for answering so quickly. Unfortunately, doing that doesn't solve it.

elFarto, 2023-08-14:
Can you run `NVD_LOG=1 NVD_BACKEND=direct vainfo` and paste the output here?

sebastiangomezlopez, 2023-08-14:
Here's the output:

```shell
libva info: VA-API version 1.14.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      1247.164643813 [8250-8250] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      1247.164657121 [8250-8250] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1247.164661762 [8250-8250] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
      1247.181624873 [8250-8250] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
      1247.181670817 [8250-8250] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
      1247.181676295 [8250-8250] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
      1247.181683192 [8250-8250] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 100 1 2 6
      1247.181752196 [8250-8250] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 535.86.05, major version: 535
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      1247.292735087 [8250-8250] ../src/vabackend.c:2081              nvTerminate Terminating 0x557bc7af4510
      1247.292800549 [8250-8250] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances
```

rkoot, 2023-08-15:
Maybe your version of libva is too old?  I'm using libva 2.19.0 which works fine. (vaapi 1.19)

elFarto, 2023-12-17:
The issue is that the Quadro T500 is not supported by NVDEC[1], therefore this driver will not work with it:

[1] https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new

closed 2023-12-17 by elFarto