elFarto/nvidia-vaapi-driver #235: How to restore symlink to VDPAU after removing VA-API?

Scotty-Trees, 2023-08-14:
According to the removal documentation on this GitHub:

> By default the driver installs itself as /usr/lib64/dri/nvidia_drv_video.so (this might be /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so on some distros). To uninstall the driver, simply remove this file. In addition, this file is usually symlinked to /usr/lib64/dri/vdpau_drv_video.so (or /usr/lib/x86_64-linux-gnu/dri/vdpau_drv_video.so) if the VDPAU to VA-API driver is installed, so this symlink will need to be restored for that driver to work normally again.

I'm on Fedora 38 Gnome, and tested a few things on here, but didn't get the results I was looking for. Just trying to undo everything I did when I was testing. Is there anyway to restore the symlink like you mentioned?

elFarto, 2023-08-14:
Run this command from the `/usr/lib64/dri/` directory:
```
sudo ln -s vdpau_drv_video.so nvidia_drv_video.so
```

Scotty-Trees, 2023-08-14:
Thanks for the super fast reply, appreciate your quick help! All good to go.

closed 2023-08-14 by Scotty-Trees