elFarto/nvidia-vaapi-driver #239: Having issues in laptop with NVIDIA T1200 Laptop GPU

apozsuse, 2023-08-24:
Hi,

I'm having some issue with video playback on a laptop with openSUSE Tumbleweed up to date. Videos get frozen and de synchronized from audio.

Here's the output of vainfo:

```
$ vainfo 
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.19 (libva 2.19.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
```

And the log of firefox when playing a yt video:

```
NVD_BACKEND=direct firefox -MOZ_LOG="PlatformDecoderModule:5" -MOZ_LOG_FILE=all.log
kf.i18n: KLocalizedString: Using an empty domain, fix the code. msgid: "Mozilla Firefox" msgid_plural: "" msgctxt: ""
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
[AVHWDeviceContext @ 0x7f4aa5f87440] Format 0x3231564e -> nv12.
[AVHWDeviceContext @ 0x7f4aa5f87440] Format 0x30313050 -> p010le.
[AVHWDeviceContext @ 0x7f4aa5f87440] Format 0x32313050 -> p012le.
[AVHWDeviceContext @ 0x7f4aa5f87440] Format 0x36313050 -> unknown.
[AVHWDeviceContext @ 0x7f4aa5f87440] Format 0x50343434 -> yuv444p.
[AVHWDeviceContext @ 0x7f4aa5f87440] VAAPI driver: VA-API NVDEC driver [direct backend].
[AVHWDeviceContext @ 0x7f4aa5f87440] Driver not found in known nonstandard list, using standard behaviour.
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
[AVHWDeviceContext @ 0x7f4a7dd84380] Format 0x3231564e -> nv12.
[AVHWDeviceContext @ 0x7f4a7dd84380] Format 0x30313050 -> p010le.
[AVHWDeviceContext @ 0x7f4a7dd84380] Format 0x32313050 -> p012le.
[AVHWDeviceContext @ 0x7f4a7dd84380] Format 0x36313050 -> unknown.
[AVHWDeviceContext @ 0x7f4a7dd84380] Format 0x50343434 -> yuv444p.
[AVHWDeviceContext @ 0x7f4a7dd84380] VAAPI driver: VA-API NVDEC driver [direct backend].
[AVHWDeviceContext @ 0x7f4a7dd84380] Driver not found in known nonstandard list, using standard behaviour.
libva info: VA-API version 1.19.0
[... 136 lines cut ...]
[vp9 @ 0x7f4a9149bc00] Decode to surface 0x12.
[vp9 @ 0x7f4a9149bc00] Param buffer (type 0, 92 bytes) is 0x1c.
[vp9 @ 0x7f4a9149bc00] Slice 0 param buffer (316 bytes) is 0x1d.
[vp9 @ 0x7f4a9149bc00] Slice 0 data buffer (19 bytes) is 0x1e.
[vp9 @ 0x7f4a9149bc00] Decode to surface 0x11.
[vp9 @ 0x7f4a9149bc00] Param buffer (type 0, 92 bytes) is 0x1f.
[vp9 @ 0x7f4a9149bc00] Slice 0 param buffer (316 bytes) is 0x20.
[vp9 @ 0x7f4a9149bc00] Slice 0 data buffer (39 bytes) is 0x21.
[vp9 @ 0x7f4a9149bc00] Decode to surface 0x10.
[vp9 @ 0x7f4a9149bc00] Param buffer (type 0, 92 bytes) is 0x22.
[vp9 @ 0x7f4a9149bc00] Slice 0 param buffer (316 bytes) is 0x23.
[vp9 @ 0x7f4a9149bc00] Slice 0 data buffer (72 bytes) is 0x24.
[vp9 @ 0x7f4a9149bc00] Decode to surface 0xf.
[vp9 @ 0x7f4a9149bc00] Param buffer (type 0, 92 bytes) is 0x25.
[vp9 @ 0x7f4a9149bc00] Slice 0 param buffer (316 bytes) is 0x26.
[vp9 @ 0x7f4a9149bc00] Slice 0 data buffer (87 bytes) is 0x27.

```

The issue might be related to this line?
```
[AVHWFramesContext @ 0x7f4a7e560700] Direct mapping disabled: deriving image does not work: 1 (operation failed).
```

Thanks!

elFarto, 2023-09-17:
Sorry for taking so long to get to this. Can you paste the logs when running with NVD_LOG=1 set?

I don't think the error you've highlighted is the cause of your issue, that's just ffmpeg seeing is vaDeriveImage works.

I'm honestly surprised it's working, as most laptop users have a lot of issues getting Firefox to only use the NVIDIA device. Do you have X/Wayland running only on the NVIDIA GPU and not on the integrated GPU?

apozsuse, 2023-09-18:
Hi!

Thanks for checking this out!

I think everything is running on the NVIDIA GPU (I think back in the day I just left the NVIDIA GPU active by means of a BIOS setting).

The output of the commands:

```
NVD_LOG=1 vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
    447634.899676365 [18968-18968] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
    447634.899679058 [18968-18968] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
    447634.899680209 [18968-18968] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
    447634.904568863 [18968-18968] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
    447634.904575471 [18968-18968] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
    447634.904578671 [18968-18968] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 100 1 2 6
    447634.904594169 [18968-18968] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 535.104.05, major version: 535
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.19 (libva 2.19.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
    447634.954549173 [18968-18968] ../src/vabackend.c:2081              nvTerminate Terminating 0x5567708a4580
    447634.954597730 [18968-18968] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances
```

And I attach the log files for firefox reproducing some video. The UX is that all the audio is played, but the video is (mostly) frozen untill the audio ends...

[firefox_stdout_yt_play.log](https://github.com/elFarto/nvidia-vaapi-driver/files/12651542/firefox_stdout_yt_play.log)
[all.log.child-7.log](https://github.com/elFarto/nvidia-vaapi-driver/files/12651556/all.log.child-7.log)
[all.log.child-10.log](https://github.com/elFarto/nvidia-vaapi-driver/files/12651557/all.log.child-10.log)
[all.log.child-9.log](https://github.com/elFarto/nvidia-vaapi-driver/files/12651555/all.log.child-9.log)

Thank you very much in advance!

elFarto, 2023-09-28:
I can't spot anything in the log files that suggests there was an error playing the video. I would recommend raising this issue with Mozilla directly as I don't think it's a driver problem.

apozsuse, 2023-10-03:
I just realized that the video only is stuttering when the audio is on. With 0 volume, the video plays smooth. I guess it must be related to pipewire... I'll keep on investigating.

Thank you very much!

closed 2023-12-17 by elFarto