elFarto/nvidia-vaapi-driver #241: Firefox high GPU utilization

evgeniy-harchenko, 2023-09-08:
There is a high GPU utilization (up to 70%) when quick scrolling page in Firefox with a lot of videos with animated preview. 
To reproduce:
1. Open YouTube main page.
2. Put cursor on any video to start preview.
3. Start scrolling down.

<details>
<summary>Example (big GIF)</summary>

[image]
</details>

Tested on Manjaro 23.0, Ubuntu 22.04. Firefox 117, Firefox 102.

elFarto, 2023-09-17:
I'm not sure this library is responsible for the high usage, as there's basically no Video Engine Utilisation. Do you get the same high usage with this library disabled and using software decoding?

evgeniy-harchenko, 2023-09-17:
So I checked it with uninstalled library, there is no problem. Also I tested it on Intel HD Graphics 4600 (libva-intel-driver) and there is high utilization (up to 60%) too. But I think for this GPU it`s normal and there are no lags. On Nvidia cursor moves choppy while scrolling. I tested GPU with benchmarks and stress tests, there are no lags on 99% GPU utilization. I don’t know if this is important, but there are no such problems on Windows. I also found out that you don’t have to scroll the page, you can just move the cursor over the videos and get this problem.

elFarto, 2023-09-28:
Could you possibly get a profile of Firefox with this occurring, using [this](https://profiler.firefox.com/)? Make sure to set it to Media.

evgeniy-harchenko, 2023-09-30:
Sure, here it is: https://share.firefox.dev/3EYr7p2 (moving cursor around the videos)
And here it is profile when I'm just watching one preview: https://share.firefox.dev/3F4FFmZ. On the first 1-2 seconds there is high GPU utilization (near 50%) and then it decreases to 5-7%. Also there is very low Video Engine Utilization on previews, as a rule <1%.

elFarto, 2023-10-10:
From the stack traces in there, it looks like there some sort of internal error in CUDA that it's trying to report. I'm not sure why that would cause increased GPU usage though. I'm honestly not sure what's going on here.

closed 2024-02-11 by evgeniy-harchenko