elFarto/nvidia-vaapi-driver #245: libva error: vaGetDriverNames() failed with unknown libva error

bruce-awareit, 2023-09-18:
running "vainfo" command, it returns this error on archlinux.

linux kernel 6.5.3-arch1-1
nvidia-dkms 535.104.05-1
libva libva 2.19.0
libva-nvidia-driver 0.0.10-1

detiam, 2023-09-20:
It's libva 2.20.0 update, just say vaGetDriverNames() failed to get driver name, if you set `LIBVA_DRIVER_NAME=nvidia` it will work still. I set this env var all the time, otherwise it won't work even in libva 2.19.
https://github.com/intel/libva/blob/8575b2bbc6d5cbf31690e682f4fa268d742fcb62/va/va.c#L683
Unknown why `vainfo` still says it's libva 2.19.0

NuLL3rr0r, 2023-11-29:
@detiam thank you! You saved the day. I was having issues running CLion and your suggestion solved it.

denji, 2023-12-16:
@detiam The internal libva code truncates the `nvidia` driver name string to `nvid`.

elFarto, 2023-12-17:
I've updated the readme to better suggest that this setting is needed as the newer versions of libva have changed how driver detection works.

closed 2023-12-17 by elFarto

mirh, 2024-01-15:
The error (and even the need to force drivers names) can be solved by compiling libva with the `-Dwith_legacy=nvctrl` meson build option. 
Otherwise there's nothing else instating the nvidia name to the roaster.

duncanyoyo1, 2024-07-20:
I am getting this error again, not sure what caused it. Would be happy to diag.