elFarto/nvidia-vaapi-driver #247: This driver works perfectly in X11 and drm display mode, but not wayland display mode when using optimus.

CosmicFusion, 2023-09-28:
I have an Optimus Laptop

When in X11 LIBVA works in X11 mode and works perfectly

When in wayland it doesn't work as it tries to load the driver on the i915 drm

When forcing vaapi to use drm mode via ```vainfo --display drm```  it works again

Maybe hardcode vaapi to drm mode

or add an NVD env-var to specify a drm device.

elFarto, 2023-09-28:
This driver can't control which display mode libva picks. The issue here is that libva asks the wayland display driver which driver it should use, and it responds with the Intel card.

You could try setting the `LIBVA_DRIVER_NAME` environment variable to `nvidia`, that should at least force it to use the correct driver, not sure if it'll work though.

CosmicFusion, 2023-09-28:
> This driver can't control which display mode libva picks. The issue here is that libva asks the wayland display driver which driver it should use, and it responds with the Intel card.
> 
> You could try setting the `LIBVA_DRIVER_NAME` environment variable to `nvidia`, that should at least force it to use the correct driver not sure if it'll work though, optimus stuff is very hit or miss.

I set driver name already from readme, because without it would use iris xe libva

When that is set it loads the nvidia va driver, but when using wayland display mode, it tries to run the driver on the intel drm node

This just doesn't happen on x11 and drm display modes, there it just works

elFarto, 2023-10-10:
Can you try setting `NVD_GPU` to `0` (assuming you only have one NVIDIA GPU) and see if that fixes the issue?

CosmicFusion, 2023-10-12:
> Can you try setting `NVD_GPU` to `0` (assuming you only have one NVIDIA GPU) and see if that fixes the issue?

Indeed that fixes the issue

offeex, 2023-10-13:
having similar issue, when trying to `LIBVA_DRIVER_NAME=nvidia`
By default, `vainfo` tries to use AMD (iGPU), works like a charm in case i have `libva-mesa-driver` installed, but fails to load with `LIBVA_DRIVER_NAME=nvidia`. Using `--display drm --device /dev/dri/renderD128` fixes the issue, but how can i make it use nvidia dGPU by default?

Using Hyprland (Wayland compositor), hybrid AMD/Nvidia GPU mode with optimus (prime-run)

env vars:
```
# Firefox
env = MOZ_DISABLE_RDD_SANDBOX, 1
env = MOZ_ENABLE_WAYLAND, 1
env = MOZ_DRM_DEVICE, /dev/dri/renderD128
env = LIBVA_DRIVER_NAME, nvidia
env = NVD_BACKEND, direct
```

CosmicFusion, 2023-10-13:
> having similar issue, when trying to `LIBVA_DRIVER_NAME=nvidia`
> By default, `vainfo` tries to use AMD (iGPU), works like a charm in case i have `libva-mesa-driver` installed, but fails to load with [...]

Add the env var `NVD_GPU=0` or whatever your gpu number is

On my iris xe main + nvidia dedicated `NVD_GPU=0` works

offeex, 2023-10-13:
just tried `VD_GPU=0 prime-run vainfo` command (both with and without `prime-run`), observing same result: 
```
Trying display: wayland
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```

CosmicFusion, 2023-10-13:
> just tried `VD_GPU=0 prime-run vainfo` command (both with and without `prime-run`), observing same result: 
> ```
> Trying display: wayland
> libva error: /usr/lib/dri/nvidia_drv_video.so init failed
> vaInitialize failed with error code 1 (operation failed),exit
> ```

Try `NVD_GPU=1` or any other value

offeex, 2023-10-13:
> > just tried `VD_GPU=0 prime-run vainfo` command (both with and without `prime-run`), observing same result:
> > ```
> > Trying display: wayland
> > libva error: /usr/lib/dri/nvidia_drv_video.so init failed
> > vaInitialize failed with error code 1 (operation failed),exit
> > ```
> 
> Try `NVD_GPU=1` or any other value

well, my bad. I mistyped it. I tried `VD_GPU` instead `NVD_GPU`

It works like a charm now with `NVD_GPU=0`

In case anyone wondering, i used `nvtop` package to list all my `GPUs`

closed 2023-12-17 by elFarto