elFarto/nvidia-vaapi-driver #248: Firefox tab crashes on higher and lower playback speeds

OmegaRogue, 2023-09-30:
When setting the playback speed different from 1 (higher or lower, the tab instantly crashes. This error is in the console when that happens:
```
OMP: Error #100: Fatal system error detected.
OMP: System error #1: Operation not permitted
Redirecting call to abort() to mozalloc_abort
```
I'm using `firefox-vaapi` from the AUR.
`vainfo`:
```
Trying display: wayland
Trying display: x11
libva error: vaGetDriverNames() failed with unknown libva error
vainfo: VA-API version: 1.20 (libva 2.20.0)
vainfo: Driver version: VA-API NVDEC driver [egl backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
```

zbikk, 2023-10-02:
the egl backend is currently broken on newer driver versions, set it to direct via the environment variable NVD_BACKEND=direct and try again.

elFarto, 2023-12-17:
Does this happen on any video, or is it a specific video? I've spent a lot of time watching videos at faster speeds without any issues.

OmegaRogue, 2024-01-02:
> Does this happen on any video, or is it a specific video? I've spent a lot of time watching videos at faster speeds without any issues.

any video, so much so that ive for now stopped using this as i like to watch videos on higher speeds

OmegaRogue, 2024-01-02:
> the egl backend is currently broken on newer driver versions, set it to direct via the environment variable NVD_BACKEND=direct and try again.

that doesn't change anything

thesword53, 2024-02-04:
I also don't have any issue with slower or faster video speed.
What version of the `nvidia-vaapi-driver` and NVIDIA drivers are you using ?