elFarto/nvidia-vaapi-driver #250: Watch 4K videos on bilibili using Firefox.

tutict, 2023-10-05:
[image]

there is my vainfo output:

```
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.21.0
libva error: vaGetDriverNames() failed with unknown libva error
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      3240.497191293 [30681-30681] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      3240.497195722 [30681-30681] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
      3240.497198154 [30681-30681] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
      3240.504883329 [30681-30681] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      3240.504892558 [30681-30681] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
      3240.504896666 [30681-30681] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 100 1 2 6
      3240.504920152 [30681-30681] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 535.113.01, major version: 535
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.21 (libva 2.21.0.pre1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      3240.609882188 [30681-30681] ../src/vabackend.c:2081              nvTerminate Terminating 0x5596df7ba520
      3240.609950816 [30681-30681] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances

```
 The RDD (Remote Data Decoder) Process is a component of Firefox responsible for handling certain media decoding tasks in an isolated process.How can I use hardware acceleration for 4K HEVC videos in Firefox?

tutict, 2023-10-12:
[image]
I've already set it in my ~/.zshrc,but where is its log file?

samduanx, 2023-10-25:
Some videos using AV1 codec on bilibili could not work due to unknown reason. I haven't saved the log at that time, but I remember that there were errors about the video's authentication key or something like that. If needed I could provide logs on that.
As for the logs, you could use terminal to run Firefox.

*Chinese alert*
我之前在b站测试av1（似乎b站会在自动解码器模式下给4k视频优先上av1），有一部分能正常加速，有一部分不能，不能的我没存log，但我记得提示似乎是视频内没有解码key，估计是阿B的问题。
log的话，直接在终端里开firefox就可以了。

elFarto, 2023-10-29:
I'm not sure any video with any DRM/authentication key will work with Firefox. It certainly won't work with this driver. 

To log the output of the driver, start Firefox with the `NVD_LOG=1` environment variable set, for example:
```
NVD_LOG=1 firefox 2>&1 | tee nvd-firefox.log
```

tutict, 2023-10-29:
@Impeta  The font I use is Clerical Script, which is a type of font that utilizes Chinese calligraphy.

tutict, 2023-10-29:
> I'm not sure any video with any DRM/authentication key will work with Firefox. It certainly won't work with this driver.
> 
> To log the output of the driver, start Firefox with the `NVD_LOG=1` environment variable set, for example:
> 
> ```
> NVD_LOG=1 firefox 2>&1 | tee nvd-firefox.log
> ```
It is unable to use, this is the output I got from Firefox's about:logging

> [RDD 27434: MediaPDecoder #3]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101617000µs ; mTimecode=101617000µs
> [RDD 27434: MediaPDecoder #2]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101625000µs ; mTimecode=101625000µs
> [RDD 27434: MediaPDecoder #4]: V/PlatformDecoderModule ProcessDecode: mDuration=9000µs ; mTime=101633000µs ; mTimecode=101633000µs
> [RDD 27434: MediaPDecoder #1]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101642000µs ; mTimecode=101642000µs
> [RDD 27434: MediaPDecoder #3]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101650000µs ; mTimecode=101650000µs
> [RDD 27434: MediaPDecoder #2]: V/PlatformDecoderModule ProcessDecode: mDuration=9000µs ; mTime=101658000µs ; mTimecode=101658000µs
> [RDD 27434: MediaPDecoder #4]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101667000µs ; mTimecode=101667000µs
> [RDD 27434: MediaPDecoder #2]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101675000µs ; mTimecode=101675000µs
> [RDD 27434: MediaPDecoder #3]: V/PlatformDecoderModule ProcessDecode: mDuration=9000µs ; mTime=101683000µs ; mTimecode=101683000µs
> [RDD 27434: MediaPDecoder #1]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101692000µs ; mTimecode=101692000µs
> [RDD 27434: MediaPDecoder #4]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101700000µs ; mTimecode=101700000µs
> [RDD 27434: MediaPDecoder #2]: V/PlatformDecoderModule ProcessDecode: mDuration=9000µs ; mTime=101708000µs ; mTimecode=101708000µs
> [RDD 27434: MediaPDecoder #3]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101717000µs ; mTimecode=101717000µs
> [RDD 27434: MediaPDecoder #1]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101725000µs ; mTimecode=101725000µs
> [RDD 27434: MediaPDecoder #4]: V/PlatformDecoderModule ProcessDecode: mDuration=9000µs ; mTime=101733000µs ; mTimecode=101733000µs
> [RDD 27434: MediaPDecoder #2]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101742000µs ; mTimecode=101742000µs
> [RDD 27434: MediaPDecoder #3]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101750000µs ; mTimecode=101750000µs
> [RDD 27434: MediaPDecoder #1]: V/PlatformDecoderModule ProcessDecode: mDuration=9000µs ; mTime=101758000µs ; mTimecode=101758000µs
> [RDD 27434: MediaPDecoder #4]: V/PlatformDecoderModule ProcessDecode: mDuration=8000µs ; mTime=101767000µs ; mTimecode=101767000µs

and:

> [Parent 26166: Main Thread]: I/Logger Flushing old log files
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
[... cut ...]
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
> [Parent 26166: Unnamed thread 7ff2c80b2280]: E/cubeb stream.rs:1063: Trigger user callback with output buffer size=16384, read_offset=0
> [Parent 26166: cubeb_log]: E/cubeb stream.rs:383: Output callback to be written buffer size 16384
>

samduanx, 2023-10-30:
### Self-test conclusion
I've checked 4k videos with AV1 codecs on bilibili just now, and I found that the problem seemed to be more severe where the driver exits with a "Due to no RTFormat specified."

My desktop has hardware as follows: i7-12700k, RTX 4090 (NVIDIA Driver fully closed-source 535.113.01), openSUSE Tumbleweed with kernel 6.5.9, KDE Plasma 5.27.9 (tested both on X11 and Wayland). Two videos were used in the test (results were almost the same so logs of playing the first one only will be uploaded here. Neither video is encrypted):
https://www.bilibili.com/video/av746613551/
https://www.bilibili.com/video/av534934565/

### Tests
The test comprised two parts, first to force using AV1 codec in the Playing Strategy (播放策略) setting in the settings menu on the lower right corner of the player (as shown in the picture), then force using HEVC codec (if not possible the player would fallback to H264, which it did).

[image]
(Press "更多播放设置" here, which literally means "More player settings"
[image]

### Logs
Logs are provided as is:
[nvd-firefox-av1.log](https://github.com/elFarto/nvidia-vaapi-driver/files/13202225/nvd-firefox-av1.log)
[nvd-firefox-h264.log](https://github.com/elFarto/nvidia-vaapi-driver/files/13202226/nvd-firefox-h264.log)

@elFarto Do favour contacting me if you want to reproduce the issue, since bilibili only provides Simplified Chinese pages only online. It is possible that the problem comes from the player used by bilibili.
@tutict 我现在有这个问题是你b播放器的问题的感觉（

flwwater, 2023-11-04:
> i

on bilibili，use linux + firefox + intel gpu ，play av1 video that still cannot using hardware decode，but when use chrome it can.
so, 这个锅得 firefox 背。

elFarto, 2023-12-17:
Can you see if you have `media.ffvpx.enabled` enabled in `about:config`? I've tested locally and without that set Firefox just seems to ignore AV1 support in the driver. With it set it uses the driver.

samduanx, 2023-12-19:
Re-adjusting my settings in about:config with `media.ffvpx.enabled` seems to be working on my side, which I could observe video engine utilization fluctuating when decoding AV1 videos from bilibili in X11. Situation in Wayland hasn't been tested.

closed 2025-08-07 by elFarto