elFarto/nvidia-vaapi-driver #251: VAAPI stopped working

Axisar, 2023-10-07:
After nvidia driver update it just stop working
here is the output
```
NVD_BACKEND=direct NVD_LOG=1 IBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
    179475.995293456 [16786-16786] ../src/vabackend.c: 138                     init CUDA ERROR 'unknown error' (999)

libva info: Found init function __vaDriverInit_1_0
    179475.995317039 [16786-16786] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
    179475.995321369 [16786-16786] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
    179475.995325236 [16786-16786] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
    179476.001936096 [16786-16786] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
    179476.001943214 [16786-16786] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
    179476.001947407 [16786-16786] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 100 1 0 fe
    179476.001967747 [16786-16786] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 535.113.01, major version: 535
    179476.002040125 [16786-16786] ../src/direct/direct-export-buf.c:  23       findGPUIndexFromFd CUDA ERROR 'initialization error' (3)

    179476.002046171 [16786-16786] ../src/vabackend.c:2236       __vaDriverInit_1_0 CUDA ERROR 'initialization error' (3)

libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed with error code 1 (operation failed),exit
```

elFarto, 2023-10-08:
I've just tested on that driver version and everything seems to work. Are you sure it's installed correctly, that `ERR!` in the `nvidia-smi` output doesn't look correct.

Dungeonseeker, 2023-10-08:
Same issue here...

```
NVD_BACKEND=direct NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
     33434.127928338 [57519-57519] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
     33434.127938032 [57519-57519] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
     33434.127939144 [57519-57519] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
     33434.132407950 [57519-57519] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
     33434.132416132 [57519-57519] ../src/vabackend.c:2231       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit
```
```
Sun Oct  8 17:12:51 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.113.01             Driver Version: 535.113.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3050 ...    Off | 00000000:01:00.0 Off |                  N/A |
| N/A   45C    P0              N/A /  60W |      6MiB /  4096MiB |      0%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A       847      G   Hyprland                                      1MiB |
+---------------------------------------------------------------------------------------+
```

Axisar, 2023-10-08:
> I've just tested on that driver version and everything seems to work. Are you sure it's installed correctly, that `ERR!` in the `nvidia-smi` output doesn't look correct.

Yeap, noticed ERR it after your mention, but its seems just some kind of laptop setup
already tried to force reinstall libva and  nvidia-vaapi-driver. what else i can try? driver itself working, i can play 3d games for example

```
mpv sample.mp4 
 (+) Video --vid=1 (*) (h264 1920x1080 30.000fps)
 (+) Audio --aid=1 (*) (aac 2ch 44100Hz)
[vaapi] libva: /usr/lib64/dri/nvidia_drv_video.so init failed
[ffmpeg] AVHWDeviceContext: cu->cuInit(0) failed -> CUDA_ERROR_UNKNOWN: unknown error                                   
[vaapi] libva: /usr/lib64/dri/nvidia_drv_video.so init failed                                                           
Using hardware decoding (vdpau-copy)
```
can this explain something^^

elFarto, 2023-10-10:
@Dungeonseeker Actually that's a different issue. Your issue is similar to #247.

@Axisar A 999 error from CUDA means something is very wrong. Possibly some of the driver modules aren't loaded, or the permissions for some of the device nodes in /dev are incorrect.

Axisar, 2023-10-10:
> @Axisar A 999 error from CUDA means something is very wrong. Possibly some of the driver modules aren't loaded, or the permissions for some of the device nodes in /dev are incorrect.

After removing old kernels and old nvidia modules it start working now.
thanks for a tip about driver

`[]$ NVD_BACKEND=direct NVD_LOG=1 IBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       173.338519209 [1550-1550] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
       173.338543269 [1550-1550] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
       173.338550025 [1550-1550] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
       173.372728862 [1550-1550] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
       173.372742873 [1550-1550] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
       173.372751902 [1550-1550] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 100 1 0 fe
       173.372800754 [1550-1550] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 535.113.01, major version: 535
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.19 (libva 2.19.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
       173.456839998 [1550-1550] ../src/vabackend.c:2081              nvTerminate Terminating 0x5617cf83d0c0
       173.456886989 [1550-1550] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances`

closed 2023-10-10 by elFarto