elFarto/nvidia-vaapi-driver #252: Firefox Stops Playing Videos

jasonwert, 2023-10-09:
I have everything working correctly, but after using Firefox for usually about a day or less, videos will stop playing. Once I restart Firefox acceleration works again. 

When I look in about:support I see this error at the bottom of the Graphics section:

```
Failure Log
(#0) 	GP+[GFX1-]: VideoBridgeParent receives IPC close with reason=AbnormalShutdown
```

Here are my vainfo & nvidia-smi

```
w3rt@legion:~$ vainfo
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
```

```
w3rt@legion:~$ nvidia-smi
Mon Oct  9 18:47:38 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.113.01             Driver Version: 535.113.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GTX 1660 Ti     Off | 00000000:01:00.0  On |                  N/A |
| N/A   41C    P0              21W /  80W |   1101MiB /  6144MiB |      2%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1105      G   /usr/lib/xorg/Xorg                          318MiB |
|    0   N/A  N/A      1667      G   /usr/bin/kwalletd5                            2MiB |
|    0   N/A  N/A      1928      G   /usr/bin/ksmserver                            2MiB |
|    0   N/A  N/A      1930      G   /usr/bin/kded5                                2MiB |
|    0   N/A  N/A      1931      G   /usr/bin/kwin_x11                            38MiB |
|    0   N/A  N/A      1978      G   /usr/bin/plasmashell                         32MiB |
|    0   N/A  N/A      1988      G   ...c/polkit-kde-authentication-agent-1        2MiB |
|    0   N/A  N/A      1990      G   ...-gnu/libexec/xdg-desktop-portal-kde        2MiB |
|    0   N/A  N/A      2085      G   ...86_64-linux-gnu/libexec/kdeconnectd        2MiB |
|    0   N/A  N/A      2086      G   /usr/bin/konsole                              2MiB |
|    0   N/A  N/A      2094      G   /usr/bin/synergy                              2MiB |
|    0   N/A  N/A      2107      G   /usr/bin/kaccess                              2MiB |
|    0   N/A  N/A      2116      G   ...-linux-gnu/libexec/DiscoverNotifier        2MiB |
|    0   N/A  N/A      2125      G   /usr/bin/kup-daemon                           2MiB |
|    0   N/A  N/A      2301      G   ...81a0d005e6f41108046422acb0.AppImage        8MiB |
|    0   N/A  N/A     29540      G   ...,WinRetrieveSuggestionsOnlyOnDemand       31MiB |
|    0   N/A  N/A    220066      G   /usr/lib/firefox/firefox                    289MiB |
|    0   N/A  N/A    220470      G   ...bin/plasma-browser-integration-host        2MiB |
|    0   N/A  N/A    220579    C+G   /usr/lib/firefox/firefox                    254MiB |
|    0   N/A  N/A    271278      G   /usr/bin/dolphin                              2MiB |
+---------------------------------------------------------------------------------------+

```

KDE Neon (Ubuntu based) 
Firefox 118.0.1 (64-bit)

Just looking for some idea to try. Seems to be a Firefox issue as I can play accelerated videos with mpv during this time with no issue and restarting Firefox fixes the issue for a bit. It's been doing this for a few weeks.

elFarto, 2023-10-10:
I also have this issue, but I've been unable to capture it happening so I'm not certain what's causing it. I'll see if I can find some time to investigate it further.

drhashes, 2023-10-28:
Same problem here. It was stable like a month ago. Since then I
have updated the video driver to 535.113.01 and also to new firefox versions.
Currently firefox 119.0 but the problem existed also with the 118 series.
I have been a bit low on RAM so I am currently testing my system now with no big
programs running so we'll see if that helps (I guess not).

vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD

+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.113.01             Driver Version: 535.113.01   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce GTX 1060 6GB    Off | 00000000:03:00.0  On |                  N/A |
| 49%   41C    P0              29W / 120W |   1044MiB /  6144MiB |     14%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1061      G   /usr/lib/xorg/Xorg                          619MiB |
|    0   N/A  N/A      1744      G   compton                                       9MiB |
|    0   N/A  N/A     30447      G   telegram-desktop                              1MiB |
|    0   N/A  N/A    112176      G   xreader                                       1MiB |
|    0   N/A  N/A    964736      G   /usr/lib/firefox/firefox                      1MiB |
|    0   N/A  N/A    964781      G   /usr/lib/firefox/firefox-bin                363MiB |
+---------------------------------------------------------------------------------------+

elFarto, 2023-10-29:
I've not had much luck reproducing this intentionally, but it does still happen to me occasionally. The best I've managed to do is get this error from Firefox, but I'm not sure if it's the cause of the issue, or the result of it.

```
[RDD 18592, MediaSupervisor #1] WARNING: pipe error: Bad file descriptor: file /builddir/build/BUILD/firefox-119.0/ipc/chromium/src/chrome/common/ipc_channel_posix.cc:702
[RDD 18592, MediaSupervisor #1] WARNING: pipe error: Bad file descriptor: file /builddir/build/BUILD/firefox-119.0/ipc/chromium/src/chrome/common/ipc_channel_posix.cc:702
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: VideoBridgeParent receives IPC close with reason=AbnormalShutdown (t=608.042) [GFX1-]: VideoBridgeParent receives IPC close with reason=AbnormalShutdown
```

drhashes, 2023-11-18:
Have not had any problems lately.

Changes that might have fixed it:

1. Firefox upgraded to 119.0.1

2. Disabled the add-on YouTube High Definition

3. Changed Adblock Plus to uBlock Origin.

trevnels, 2023-11-25:
Scrolling through Instagram Reels in Firefox 120 seems to reliably reproduce this for me after a few minutes. It looks like it may be some sort of memory leak - memory use slowly climbs by about 2GB as new videos are loaded until the problem hits. Firefox will become unresponsive for a few seconds, then start working again, but no videos across any site will play until the browser is restarted.

I will also note that Instagram video playback in general seems to be buggy, where videos will often stutter and/or not play to completion, as if a certain frame is corrupted and the decoder gives up. I'm not sure if this is a problem with Instagram, Firefox, or nvidia-vaapi-driver.

drhashes, 2023-12-28:
OS: Ubuntu 22.04
video card: 1060 6 gb
nvidia video driver: 535.113.01

nvidia-vaapi-driver from: 
https://github.com/elFarto/nvidia-vaapi-driver/releases

v0.0.11: problem occurs after 1-2h
v0.0.10: I think stable (no problem after 4-5days)

Something for others to test and report back here also. This finding could indicate that some update in v0.0.11 is responsible.

However, since noone is complaining having nvidia video driver 545 I guess v0.0.11 is stable for these users. This could then indicate an incompatibility between nvidia video driver 535 and v0.0.11.

argvx, 2024-03-28:
OS: LinuxMint 21.3
Tested GPUs: GTX 970 (4G), GTX 1080 (8G)
CPU: i9-9920X
RAM: 64G
nvidia drivers tested: 535, 545, 550.67
vaapi drivers tested: v0.0.10, v0.0.11

Same problem, random crashes, not working until firefox restart.
Especially noticeable when there are multiple videos, for example surveillance system (8 cams), it may work for 1 or even 5 hours one time, then crash.. or even just for 2-5 mins.. rarely it may work all day without any issues..

EiPiFun, 2024-06-17:
> I've not had much luck reproducing this intentionally, but it does still happen to me occasionally. The best I've managed to do is get this error from Firefox, but I'm not sure if it's the cause of [...]

I have a device which could reproduce this error log stably, it is an old optimus laptop (i5 6300HQ+GTX 960M) running Debian 12 and NVIDIA driver version 470 or 520 when I try to use the egl backend.

Debian, Ubuntu or Fedora is tested, seems not to be distro-related.
nvidia-vaapi-driver 0.0.7 or 0.0.12 (with egl backend) are both tested.

The only wired thing I notice is that this old laptop is purely optimus cause its GPU is marked as **3D controller** (without any hardware display port like HDMI or DP)

If I upgrade nvidia driver to version 535 or higher, it just works with direct backend.

I guess it is a problem related to video memory share.

Here is my output:

eipifun@debian:~$ lspci|grep NVIDIA
01:00.0 3D controller: NVIDIA Corporation GM107M [GeForce GTX 960M] (rev a2)

eipifun@debian:~$ pkill firefox;MOZ_LOG="FFmpegVideo:5" firefox-esr
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia_egl'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_egl_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
[GFX1-]: VideoBridgeParent receives IPC close with reason=AbnormalShutdown

eipifun@debian:~$ vainfo
libva info: VA-API version 1.17.0
libva info: User environment variable requested driver 'nvidia_egl'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_egl_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver
vainfo: Supported profile and entrypoints
VAProfileMPEG2Simple : VAEntrypointVLD
VAProfileMPEG2Main : VAEntrypointVLD
VAProfileVC1Simple : VAEntrypointVLD
VAProfileVC1Main : VAEntrypointVLD
VAProfileVC1Advanced : VAEntrypointVLD
&lt;unknown profile&gt; : VAEntrypointVLD
VAProfileH264Main : VAEntrypointVLD
VAProfileH264High : VAEntrypointVLD
VAProfileH264ConstrainedBaseline: VAEntrypointVLD

eipifun@debian:~$ nvidia-smi
Mon Jun 17 14:03:32 2024
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 520.56.06 Driver Version: 520.56.06 CUDA Version: 11.8 |

[image]

CtrlC-Root, 2025-02-04:
This happens to me pretty consistently. How can I capture more useful debugging information to help track this down?

CtrlC-Root, 2025-08-07:
I have not experienced this for months for what it's worth. I wish I could remember at what point it stopped and correlate it with a package update or something.