elFarto/nvidia-vaapi-driver #254: 32-bit NVD Causes GST Segment fault in wine while using wayland

CosmicFusion, 2023-10-28:
when having nvidia-vaapi-driver:i386 installed wine hardware accelerates video content of 32 bit games on the nvidia gpu beautifully, such as the GTA SA intro video but that's only when on X11

doing the same on wayland results in 
```ERROR: Caught a segmentation fault while loading plugin file:
/lib/i386-linux-gnu/gstreamer-1.0/libgstva.so

Please either:
- remove it and restart.
- run with --gst-disable-segtrap --gst-disable-registry-fork and debug.
Monitored process exited.
Initial process has exited (return code: 0)
Exit with return code 0```

elFarto, 2023-10-29:
Could you get a log with `NVD_LOG=1` set?

CosmicFusion, 2023-10-29:
[gta_sa_3.log](https://github.com/elFarto/nvidia-vaapi-driver/files/13198259/gta_sa_3.log)

elFarto, 2023-10-29:
Can you try with `NVD_BACKEND=direct`, the EGL backend doesn't work at the moment due to an NVIDIA issue.?

CosmicFusion, 2023-10-29:
> Can you try with `NVD_BACKEND=direct`, the EGL backend doesn't work at the moment due to an NVIDIA issue.?

It's direct

CosmicFusion, 2023-10-29:
Never mind actually 

The problem is clear now

Despite NVD_GPU and NVD_BACKEND being set system wide

Lutris isn't forwarding them to wine

Add these vars to lutris manually 

Makes the game launch, and it's intro plays and is hardware accelerated by nvidia and on wayland

So lutris issue not NVD

closed 2023-10-29 by CosmicFusion