elFarto/nvidia-vaapi-driver #256: Doesn't work with 545.29.02 (used to work with Beta 545.23.06)

solopasha, 2023-11-01:
vainfo:
```
Trying display: wayland
libva info: VA-API version 1.18.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /home/solopasha/.var/app/org.mozilla.firefox/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       189.015528287 [4-4] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 40
       189.015533076 [4-4] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
       189.015536132 [4-4] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
       189.020488494 [4-4] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
       189.020495267 [4-4] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 600 0 fe 1
       189.020513380 [4-4] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 545.29.02, major version: 545
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.18 (libva 2.20.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
       189.094463257 [4-4] ../src/vabackend.c:2081              nvTerminate Terminating 0x559fa2680c70
       189.094515485 [4-4] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances
```

firefox logs:
```
libva info: VA-API version 1.18.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /home/solopasha/.var/app/org.mozilla.firefox/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
       240.687291959 [316-327] ../src/vabackend.c:2171       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       240.687298100 [316-327] ../src/vabackend.c:2180       __vaDriverInit_1_0 Now have 0 (0 max) instances
       240.687301146 [316-327] ../src/vabackend.c:2206       __vaDriverInit_1_0 Selecting Direct backend
       240.692208989 [316-327] ../src/direct/nv-driver.c: 223            init_nvdriver Initing nvdriver...
       240.692218136 [316-327] ../src/direct/nv-driver.c: 228            init_nvdriver Got dev info: 600 0 fe 1
       240.692240137 [316-327] ../src/direct/nv-driver.c: 246            init_nvdriver NVIDIA kernel driver version: 545.29.02, major version: 545
libva info: va_openDriver() returns 0
       240.737530342 [316-327] ../src/vabackend.c:1409      nvQueryImageFormats In nvQueryImageFormats
       240.763341843 [316-327] ../src/vabackend.c: 648           nvCreateConfig got profile: 6 with 0 attributes
       240.763355068 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       240.763358043 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) 0x7fbaa01cbd00 8
       240.764832839 [316-327] ../src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       240.764841566 [316-327] ../src/vabackend.c: 648           nvCreateConfig got profile: 7 with 0 attributes
       240.764845643 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       240.764848409 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) 0x7fbaa01cbe80 8
       240.766258274 [316-327] ../src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       240.766265317 [316-327] ../src/vabackend.c: 648           nvCreateConfig got profile: 13 with 0 attributes
       240.766268893 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       240.766271398 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 4 (8) 0x7fbaa01cb400 8
       240.767817428 [316-327] ../src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       240.767824681 [316-327] ../src/vabackend.c: 648           nvCreateConfig got profile: 14 with 0 attributes
       240.767828208 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 9 (8) (nil) 0
       240.767831995 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 9 (8) 0x7fbaa01cb7c0 8
       240.769234676 [316-327] ../src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       240.769242170 [316-327] ../src/vabackend.c: 648           nvCreateConfig got profile: 19 with 0 attributes
       240.769245487 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 10 (8) (nil) 0
       240.769248112 [316-327] ../src/vabackend.c:1765 nvQuerySurfaceAttributes with 10 (8) 0x7fbaa01cb400 8
       240.770640905 [316-327] ../src/vabackend.c:1838 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
       240.770865384 [316-327] ../src/vabackend.c:2081              nvTerminate Terminating 0x7fbab8825f20
       240.770928502 [316-327] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances
[... 107 lines cut ...]
       241.932332918 [316-387] ../src/vabackend.c: 428          resolveSurfaces [RT] Resolve thread for 0x7fba88d23000 exiting
       241.932358546 [316-386] ../src/vabackend.c: 303           destroyContext pthread_timedjoin_np finished with 0
       241.944042824 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface 0 (0x7fba8b7dc340)
       241.944050067 [316-386] ../src/direct/direct-export-buf.c: 243      destroyBackingImage Destroying BackingImage: 0x7fba8b7ece60
       241.944313900 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dc700)
       241.944318910 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dc0c0)
       241.944322456 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dca20)
       241.944327846 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dc020)
       241.944331814 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dc980)
       241.944335781 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dcde0)
       241.944339588 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dcac0)
       241.944343225 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dd2e0)
       241.944347072 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dd240)
       241.944350709 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dd100)
       241.944354296 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dc8e0)
       241.944357822 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dd380)
       241.944361459 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dcb60)
       241.944364906 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dd1a0)
       241.944368472 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dcf20)
       241.944372009 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dcc00)
       241.944375676 [316-386] ../src/vabackend.c: 965        nvDestroySurfaces Destroying surface -1 (0x7fba8b7dc840)
       241.944380715 [316-386] ../src/vabackend.c:2081              nvTerminate Terminating 0x7fbab8825b60
       241.944519565 [316-386] ../src/vabackend.c:2095              nvTerminate Now have 0 (0 max) instances

```

jasonwert, 2023-11-01:
Not sure if it is related, but I am on 545.29.02 on Kde Neon 

w3rt@legion:~$ sudo journalctl -p 3 -b

Nov 01 13:30:13 legion kernel: 
Nov 01 13:30:13 legion kernel: NVRM: loading NVIDIA UNIX x86_64 Kernel Module  545.29.02  Thu Oct 26 21:21:38 UTC 2023
Nov 01 13:30:13 legion systemd[1]: snapd.aa-prompt-listener.service: Ignoring invalid environment assignment 'export MOZ_DISABLE_RDD_SANDBOX=1': /etc/environment
Nov 01 13:30:13 legion systemd[1]: snapd.service: Ignoring invalid environment assignment 'export MOZ_DISABLE_RDD_SANDBOX=1': /etc/environment

elFarto, 2023-11-01:
That looks like an EGL issue, it's using the Mesa EGL library, not the NVIDIA one.

pallaswept, 2023-11-02:
Same problem here.  Is this something we should be trying to fix or is it just in the driver and that's how it is?

elFarto, 2023-11-02:
Interesting, I've just upgraded to 545.29.02 and it's effecting my machine aswell. I'll take a look into it when I get a moment.

pallaswept, 2023-11-02:
Thanks mate. I did notice that the nvidia compute libs I built against were libnvidia-egl-wayland 1.1.12 and they dropped 1.1.13 a couple weeks ago, so rebuilding against that lib now to see if it makes any difference.

elFarto, 2023-11-02:
Hmm, this might not be good. I found an obvious difference between the versions that was tripping us up, but after fixing that it just errors with a different message: 
```
eglCreateImageKHR: EGL_BAD_MATCH error: In eglCreateImageKHR: could not bind to DMA buffer
```
I don't see any other changes that should effect us. I'll take a closer look when I get more time.

pallaswept, 2023-11-02:
Nu rush, we appreciate your work.

kj-sh604, 2023-11-04:
Just wanted to report that I am also facing the same issue (X11 with the Direct Rendering Backend). I am very much willing to provide logs and additional information/testing to help out. 

As mentioned by @pallaswept above, absolutely no rush! we appreciate everything that you do @elFarto and we just want to do our best to support.

elFarto, 2023-11-04:
Ok, it should be fixed in master now. Haven't had a chance to try it on an older driver version yet, but it should still be good.

pallaswept, 2023-11-04:
I've already rolled back to the old drivers and built this from master, so I'll test it out with the old drivers first and then upgrade and confirm the fix, and get back to you shortly with confirmation.

pallaswept, 2023-11-04:
Thanks @elFarto love your work mate.

thulle, 2023-11-04:
Can also confirm it works w latest git.

kj-sh604, 2023-11-04:
Confirming that it's working for me on the latest driver with the following related `env` variables:
```bash
# LIBVA_DRIVER_NAME=vdpau
LIBVA_DRIVER_NAME=nvidia
# VDPAU_DRIVER=nvidia
MOZ_DISABLE_RDD_SANDBOX=1
MOZ_X11_EGL=1
NVD_BACKEND=direct
ADW_DISABLE_PORTAL=1

```
**(X11 with Direct Rendering Backend)**

[image]

Thanks for all your work @elFarto . We sincerely appreciate what you do.

shelterx, 2023-11-04:
> Ok, it should be fixed in master now. Haven't had a chance to try it on an older driver version yet, but it should still be good.

Works fine with RTX 3060Ti / Driver 545.29.02
Thank you!

m00r3ik, 2023-11-04:
Works fine for me too 
Thank you !

PS
_There is a slightly unrelated question.
Telegram-desktop seems to have vaapi support, but it is probably not compatible with this driver yet.
If you have time and desire, I'm attached a log below

Thanks again !_
`      1147.376045927 [7230-7356] ../src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      1147.376056557 [7230-7356] ../src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1147.376058927 [7230-7356] ../src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      1147.378451082 [7230-7356] ../src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      1147.378476332 [7230-7356] ../src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 545.29.02, major version: 545, minor version: 29
      1147.378482412 [7230-7356] ../src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 900 1 2 6
      1147.451021656 [7230-7356] ../src/vabackend.c:1398      nvQueryImageFormats In nvQueryImageFormats
      1147.478541785 [7230-7356] ../src/vabackend.c: 637           nvCreateConfig got profile: 7 with 0 attributes
      1147.478555615 [7230-7356] ../src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      1147.478560355 [7230-7356] ../src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x7f7e2a2b0640 8
      1147.479683677 [7230-7356] ../src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      1147.479692357 [7230-7356] ../src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      1147.479695877 [7230-7356] ../src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x7f7e2a2b0640 8
      1147.480862660 [7230-7356] ../src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      1147.480879460 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465abe0)
      1147.480885350 [7230-7356] ../src/vabackend.c:1492            nvDeriveImage In nvDeriveImage
      1147.480890740 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465ac80)
      1147.480898470 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465ad20)
      1147.480901490 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465adc0)
      1147.480903830 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465ae60)
      1147.480906470 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465af00)
      1147.480909240 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465afa0)
      1147.480911650 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b040)
      1147.480916280 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b0e0)
      1147.480920630 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b180)
      1147.480924980 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b220)
      1147.480928240 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b2c0)
      1147.480930930 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b360)
      1147.480933660 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b400)
      1147.480936940 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b4a0)
      1147.480939600 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b540)
      1147.480942620 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b5e0)
      1147.480947511 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b680)
      1147.480950321 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b720)
      1147.480953201 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b7c0)
      1147.480955851 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b860)
      1147.480960561 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b900)
      1147.480964411 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465b9a0)
      1147.480967281 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465ba40)
      1147.480970321 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465bae0)
      1147.480973121 [7230-7356] ../src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 352x640, format 1 (0x7f7e1465bb80)
[... cut ...]
      1147.508811273 [7230-7356] ../src/vabackend.c: 954        nvDestroySurfaces Destroying surface -1 (0x7f7e1465ad20)
      1147.508814613 [7230-7356] ../src/vabackend.c: 954        nvDestroySurfaces Destroying surface -1 (0x7f7e1465ac80)
      1147.508817943 [7230-7356] ../src/vabackend.c: 954        nvDestroySurfaces Destroying surface -1 (0x7f7e1465abe0)
      1147.521148383 [7230-7356] ../src/vabackend.c:2050              nvTerminate Terminating 0x7f7e2a24a000
      1147.521350575 [7230-7356] ../src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
`

pallaswept, 2023-11-04:
> PS
_There is a slightly unrelated question.

Totally unrelated. Please log a new issue and don't hijack.

marcin-rzeznicki, 2023-11-05:
Thanks! Perhaps it's a high time for release, so that AUR picks it up. Thank you once again

elFarto, 2023-11-06:
v0.0.11 has been released with this fix included. Thanks for all your testing!

closed 2023-11-06 by elFarto