elFarto/nvidia-vaapi-driver #257: No hardware acceleration on Youtube in Firefox 119

Lastique, 2023-11-13:
Playing a video on Youtube in Firefox 119 does not use hardware-accelerated decoding. This is confirmed by `nvidia-smi`, as it shows Type "G" for the `firefox` process.

Example video: https://www.youtube.com/watch?v=mvkXCvZ2ivI
Video codec: VP9

Environment:

```
NVD_BACKEND=direct
```

Firefox command line:

```
NVD_LOG="$HOME/nvd.log" LIBVA_DRIVER_NAME=nvidia MOZ_DISABLE_RDD_SANDBOX=1 firefox
```

about:config:

```
media.ffmpeg.vaapi.enabled=true
media.rdd-ffmpeg.enabled=true
media.av1.enabled=true
gfx.x11-egl.force-enabled=true
widget.dmabuf.force-enabled=false
```

Note: Although AV1 is enabled, I can confirm that it is not used for the video I'm playing (see the link above). This can be seen in "Stats for nerds" on Youtube.

`nvd.log`:

```
      3474.120141637 [17537-17537] ../src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      3474.120168041 [17537-17537] ../src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
      3474.120171396 [17537-17537] ../src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      3474.138380800 [17537-17537] ../src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      3474.138429023 [17537-17537] ../src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 535.43.16, major version: 535, minor version: 43
      3474.138436229 [17537-17537] ../src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 100 1 2 6
      3474.196171525 [17537-17537] ../src/vabackend.c: 637           nvCreateConfig got profile: 6 with 0 attributes
      3474.196184852 [17537-17537] ../src/vabackend.c: 637           nvCreateConfig got profile: 7 with 0 attributes
      3474.196186699 [17537-17537] ../src/vabackend.c: 637           nvCreateConfig got profile: 13 with 0 attributes
      3474.196187905 [17537-17537] ../src/vabackend.c: 637           nvCreateConfig got profile: 14 with 0 attributes
      3474.196189160 [17537-17537] ../src/vabackend.c: 637           nvCreateConfig got profile: 19 with 0 attributes
      3474.196190351 [17537-17537] ../src/vabackend.c: 637           nvCreateConfig got profile: 21 with 0 attributes
      3474.196194386 [17537-17537] ../src/vabackend.c:2050              nvTerminate Terminating 0x55ca531effb0
      3474.196221614 [17537-17537] ../src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
```

Operating System: Kubuntu 22.04
KDE Plasma Version: 5.24.7
KDE Frameworks Version: 5.92.0
Qt Version: 5.15.3
Kernel Version: 6.2.0-1016-lowlatency (64-bit)
Graphics Platform: X11
Processors: 16 × 12th Gen Intel® Core™ i7-12700K
Memory: 31.1 GiB of RAM
Graphics Processor: NVIDIA GeForce RTX 2080 Ti/PCIe/SSE2
Nvidia driver: 535.43.16 (from [here](https://developer.nvidia.com/vulkan-driver))
Nvidia VAAPI driver: 0.0.11

Lastique, 2023-11-13:
I can add that hardware video decoding in Firefox worked on this system a while ago, then it stopped working. Unfortunately, I don't know the exact moment when it broke, so I can't tell what changed.

rkoot, 2023-11-18:
set widget.dmabuf.force-enabled=true and try again?

Lastique, 2023-11-18:
Thanks, that did the trick.

closed 2023-11-18 by Lastique