elFarto/nvidia-vaapi-driver #259: nvCreateContext CUDA ERROR 'invalid argument' when trying decode using ffmpeg

jackdaw213, 2023-11-25:
I'm using the driver for Firefox hardware acceleration but it's not working so I tried it with ffmpeg first but it does not work either

- Operating System: Arch Linux 
- KDE Plasma Version: 5.27.9
- KDE Frameworks Version: 5.112.0
- Qt Version: 5.15.11
- Kernel Version: 6.6.2-arch1-1 (64-bit)
- Graphics Platform: Wayland
- Processors: 12 × Intel® Core™ i5-10400 CPU @ 2.90GHz
- Memory: 23.3 GiB of RAM
- Graphics Processor: NVIDIA GeForce RTX 3080/PCIe/SSE2
- The driver was installed via AUR
Firefox infos:
[image]

NVD_LOG=1 vainfo:
```
Trying display: wayland
Trying display: x11
      3049.509766559 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      3049.509774859 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
      3049.509779000 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      3049.516303307 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
      3049.516322142 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
      3049.516324674 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      3049.516344727 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 545.29.06, major version: 545, minor version: 29
      3049.516347852 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 100 1 2 6
vainfo: VA-API version: 1.20 (libva 2.20.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      3049.597582041 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x5646cd8f1980
      3049.597655751 [22880-22880] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
```
ffmpeg -hwaccels: 
```
ffmpeg version n6.1 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13.2.1 (GCC) 20230801
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libjxl --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpl --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-opencl --enable-opengl --enable-shared --enable-version3 --enable-vulkan
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Hardware acceleration methods:
vdpau
cuda
vaapi
qsv
drm
[... 149 lines cut ...]
[out#0/mp4 @ 0x55c80bd74240] video:248kB audio:2kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 4.175495%
frame=  347 fps= 71 q=-1.0 Lsize=     260kB time=00:00:05.73 bitrate= 371.8kbits/s speed=1.17x    
[libx264 @ 0x55c80bdb7e00] frame I:2     Avg QP:16.06  size: 74614
[libx264 @ 0x55c80bdb7e00] frame P:87    Avg QP:14.25  size:   541
[libx264 @ 0x55c80bdb7e00] frame B:258   Avg QP:12.67  size:   222
[libx264 @ 0x55c80bdb7e00] consecutive B-frames:  0.9%  0.0%  0.0% 99.1%
[libx264 @ 0x55c80bdb7e00] mb I  I16..4: 27.3% 65.6%  7.1%
[libx264 @ 0x55c80bdb7e00] mb P  I16..4:  0.0%  0.1%  0.0%  P16..4:  0.1%  0.0%  0.0%  0.0%  0.0%    skip:99.7%
[libx264 @ 0x55c80bdb7e00] mb B  I16..4:  0.0%  0.0%  0.0%  B16..8:  0.7%  0.0%  0.0%  direct: 0.0%  skip:99.3%  L0:68.6% L1:31.4% BI: 0.0%
[libx264 @ 0x55c80bdb7e00] 8x8 transform intra:64.6% inter:55.7%
[libx264 @ 0x55c80bdb7e00] coded y,uvDC,uvAC intra: 8.4% 5.4% 5.0% inter: 0.0% 0.0% 0.0%
[libx264 @ 0x55c80bdb7e00] i16 v,h,dc,p: 71% 25%  4%  0%
[libx264 @ 0x55c80bdb7e00] i8 v,h,dc,ddl,ddr,vr,hd,vl,hu: 71%  4% 22%  0%  0%  1%  0%  1%  1%
[libx264 @ 0x55c80bdb7e00] i4 v,h,dc,ddl,ddr,vr,hd,vl,hu: 46% 22%  8%  2%  5%  5%  4%  4%  4%
[libx264 @ 0x55c80bdb7e00] i8c dc,h,v,p: 92%  4%  4%  0%
[libx264 @ 0x55c80bdb7e00] Weighted P-Frames: Y:0.0% UV:0.0%
[libx264 @ 0x55c80bdb7e00] ref P L0: 76.4%  4.9%  9.0%  9.8%
[libx264 @ 0x55c80bdb7e00] ref B L0: 50.9% 49.0%  0.2%
[libx264 @ 0x55c80bdb7e00] ref B L1: 99.8%  0.2%
[libx264 @ 0x55c80bdb7e00] kb/s:350.60
[aac @ 0x55c80d8e8540] Qavg: 65536.000
      2983.122702151 [22615-22615] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x55c80be6b0e0
      2983.122908390 [22615-22615] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances

```

rkoot, 2023-11-25:
This driver *only* provides hw accel in firefox. So it won't work in chome, ffmpeg etc etc etc.

elFarto, 2023-11-25:
It should work in ffmpeg, as that's how Firefox access VA-API.

The error you're getting is odd, a looks like a fundamental issue and I'm not sure what's causing it. Could you try playing back a smaller video (resolution wise)?

jackdaw213, 2023-11-25:
So I downscaled the original video to 1280x720, and the same issue still happens
```
ffmpeg version n6.1 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13.2.1 (GCC) 20230801
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libiec61883 --enable-libjack --enable-libjxl --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librav1e --enable-librsvg --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpl --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-opencl --enable-opengl --enable-shared --enable-version3 --enable-vulkan
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'input.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    encoder         : Lavf60.16.100
  Duration: 00:00:05.78, start: 0.000000, bitrate: 140 kb/s
  Stream #0:0[0x1](und): Video: h264 (High) (avc1 / 0x31637661), yuv420p(tv, bt709, progressive), 1280x720 [SAR 43:32 DAR 43:18], 123 kb/s, 60 fps, 60 tbr, 15360 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
      encoder         : Lavc60.31.102 libx264
  Stream #0:1[0x2](und): Audio: aac (LC) (mp4a / 0x6134706D), 48000 Hz, stereo, fltp, 2 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
      vendor_id       : [0][0][0][0]
       530.997050546 [6621-6621] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
       530.997059710 [6621-6621] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
       530.997061568 [6621-6621] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
       531.004613680 [6621-6621] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
       531.004658300 [6621-6621] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 545.29.06, major version: 545, minor version: 29
       531.004662255 [6621-6621] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 100 1 2 6
       531.063696566 [6621-6621] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1398      nvQueryImageFormats In nvQueryImageFormats
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> h264 (libx264))
  Stream #0:1 -> #0:1 (aac (native) -> aac (native))
Press [q] to stop, [?] for help
       531.095578704 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 637           nvCreateConfig got profile: 7 with 0 attributes
       531.095591056 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) 0
       531.095593487 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x7f1b501ffbc0 8
       531.096575316 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       531.096583321 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) (nil) 32539
       531.096585873 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1754 nvQuerySurfaceAttributes with 4 (8) 0x7f1b501ffbc0 8
       531.097760195 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1821 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
       531.097780430 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x7f1b502066d0)
       531.097784501 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:1492            nvDeriveImage In nvDeriveImage
       531.097788556 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x7f1b502002a0)
       531.097791552 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x7f1b502009b0)
       531.097794638 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x7f1b50200de0)
       531.097797839 [6621-6628] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c: 920        nvCreateSurfaces2 Creating surface 1280x720, format 1 (0x7f1b502011e0)
[... cut ...]
[libx264 @ 0x55f12d3bdcc0] ref P L0: 89.5%  3.9%  5.9%  0.7%
[libx264 @ 0x55f12d3bdcc0] ref B L0: 33.6% 66.4%  0.0%
[libx264 @ 0x55f12d3bdcc0] ref B L1: 99.8%  0.2%
[libx264 @ 0x55f12d3bdcc0] kb/s:133.73
[aac @ 0x55f12eb4ab00] Qavg: 65536.000
       532.161915473 [6621-6621] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x55f12d3be6b0
       532.162055940 [6621-6621] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances

```

elFarto, 2023-12-17:
It looks like 33 surfaces is exactly 1 too many for the driver to handle. I've added some code to master to limit the surface count to 32 which should let it run, although it may not run correctly if ffmpeg actually tries to use all those surfaces (there's no way it actually needs that many to decode the video).

jackdaw213, 2024-07-01:
FFmpeg is working fine now, thank you for your work. However, Firefox (Nightly 129) somehow needs the MOZ_DISABLE_RDD_SANDBOX flag set to either 0 or 1 for HW decoding to work. This is odd since RDD should not be the problem for VAAPI anymore and RDD is on by default so MOZ_DISABLE_RDD_SANDBOX=0 should not change anything right ? But without it, no HW decoding

> NVD_LOG=1 firefox-nightly
```
4851.971784688 [62440-62453] ../src/vabackend.c: 141                     init ERROR: Potential Firefox sandbox detected, failing to init!
      4851.971797487 [62440-62453] ../src/vabackend.c: 142                     init If running in Firefox, set env var MOZ_DISABLE_RDD_SANDBOX=1 to disable sandbox.
      4851.971822937 [62440-62453] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4851.971825982 [62440-62453] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
      4869.402930588 [62440-62742] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4869.402939467 [62440-62742] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 1 (0 max) instances
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
      4869.576256505 [62440-62748] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4869.576264999 [62440-62748] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 2 (0 max) instances
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
      4869.752926316 [62440-62756] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4869.752934401 [62440-62756] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 3 (0 max) instances
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
      4869.796281288 [62440-62760] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4869.796291997 [62440-62760] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 4 (0 max) instances
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
      4869.852938573 [62440-62764] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4869.852947055 [62440-62764] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 5 (0 max) instances
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
      4869.952920904 [62440-62768] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4869.952929050 [62440-62768] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 6 (0 max) instances
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
      4875.159599303 [62440-62842] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      4875.159608136 [62440-62842] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 7 (0 max) instances
libva error: /usr/lib/dri/nvidia_drv_video.so init failed
```

> MOZ_DISABLE_RDD_SANDBOX=0 NVD_LOG=1 firefox-nightly

This will work, with MOZ_DISABLE_RDD_SANDBOX be 0 or 1

jackdaw213, 2024-07-01:
I forgot to mention but if I switch the display output to my Intel iGPU, Firefox HW decoding on the iGPU works just fine without RDD flag

closed 2025-08-07 by elFarto