elFarto/nvidia-vaapi-driver #260: Nvidia-vaapi-driver is not initialized (CUDA ERROR 802) in very old PC on LinuxMint21.2 for GTX1050TI card

Wyacheslaw8, 2023-12-14:
Hello
I compiled and installed nvidia-vaapi-driver successfully.

My results:

`NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD128`
```
      1236.335911535 [4264-4264] ../src/vabackend.c: 130                     init CUDA ERROR 'system not yet initialized' (802)

      1236.336062669 [4264-4264] ../src/vabackend.c:2145       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      1236.336076079 [4264-4264] ../src/vabackend.c:2154       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1236.336086974 [4264-4264] ../src/vabackend.c:2180       __vaDriverInit_1_0 Selecting Direct backend
      1236.357610389 [4264-4264] ../src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      1236.357717943 [4264-4264] ../src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 535.129.03, major version: 535, minor version: 129
      1236.357730514 [4264-4264] ../src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 100 1 0 fe
      1236.357886398 [4264-4264] ../src/direct/direct-export-buf.c:  23       findGPUIndexFromFd CUDA ERROR 'initialization error' (3)

      1236.357899528 [4264-4264] ../src/vabackend.c:2210       __vaDriverInit_1_0 CUDA ERROR 'initialization error' (3)

libva error: /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so init failed
vaInitialize failed with error code 1 (operation failed),exit

```

`inxi -Fxz`

```
 System:
  Kernel: 5.15.0-76-generic x86_64 bits: 64 compiler: gcc v: 11.3.0
    Desktop: Cinnamon 5.8.4 Distro: Linux Mint 21.2 Victoria
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop Mobo: ASUSTeK model: A8N-E v: 2.XX
    serial: <superuser required> BIOS: Phoenix v: ASUS A8N-E Revision 1013
    date: 04/07/2006
CPU:
  Info: dual core model: AMD Opteron 185 bits: 64 type: MCP arch: K8 rev.E
    rev: 2 cache: L1: 256 KiB L2: 2 MiB
  Speed (MHz): avg: 1000 min/max: 1000/2600 cores: 1: 1000 2: 1000
    bogomips: 10452
  Flags: ht lm nx pae sse sse2 sse3
Graphics:
  Device-1: NVIDIA GP107 [GeForce GTX 1050 Ti] vendor: Gigabyte
    driver: nvidia v: 535.129.03 bus-ID: 01:00.0
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia
    resolution: 1920x1200~60Hz
  OpenGL: renderer: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2
    v: 4.6.0 NVIDIA 535.129.03 direct render: Yes
Audio:
  Device-1: NVIDIA GP107GL High Definition Audio vendor: Gigabyte
    driver: snd_hda_intel v: kernel bus-ID: 01:00.1
  Sound Server-1: ALSA v: k5.15.0-76-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: NVIDIA CK804 Ethernet vendor: ASUSTeK K8N4/A8N Series Mainboard
    type: network bridge driver: forcedeth v: kernel port: e000 bus-ID: 00:0a.0
  IF: enp0s10 state: up speed: 1000 Mbps duplex: full mac: <filter>
Drives:
  Local Storage: total: 1.05 TiB used: 96.48 GiB (9.0%)
  ID-1: /dev/sda vendor: Samsung model: SSD 850 PRO 1TB size: 953.87 GiB
  ID-2: /dev/sdb type: USB vendor: Samsung model: Flash Drive FIT
    size: 119.51 GiB
Partition:
  ID-1: / size: 91.11 GiB used: 11.77 GiB (12.9%) fs: ext4 dev: /dev/sda5
  ID-2: /boot/efi size: 512.1 MiB used: 6.1 MiB (1.2%) fs: vfat
    dev: /dev/sda3
Swap:
  ID-1: swap-1 type: partition size: 5.59 GiB used: 0 KiB (0.0%)
[... 148 lines cut ...]
```
filename:       /lib/modules/5.15.0-76-generic/updates/dkms/nvidia-drm.ko
version:        535.129.03
supported:      external
license:        MIT
srcversion:     8B4888C7A7BA5B8AF77707F
alias:          pci:v000010DEd*sv*sd*bc06sc80i00*
alias:          pci:v000010DEd*sv*sd*bc03sc02i00*
alias:          pci:v000010DEd*sv*sd*bc03sc00i00*
depends:        drm,drm_kms_helper,nvidia-modeset
retpoline:      Y
name:           nvidia_drm
vermagic:       5.15.0-76-generic SMP mod_unload modversions 
sig_id:         PKCS#7
signer:         localhost.localdomain Secure Boot Module Signature key
sig_hashalgo:   sha512
parm:           modeset:Enable atomic kernel modesetting (1 = enable, 0 = disable (default)) (bool)
```

`cat /proc/cmdline`
```
BOOT_IMAGE=/boot/vmlinuz-5.15.0-76-generic root=UUID=f617cccc-971c-437e-84c1-feb51a64b92f ro nvidia-drm.modeset=1 quiet splash
```

PS: My player Celluloid is working fine - hardware acceleration is working for VP9 videos.

elFarto, 2023-12-17:
That is a very old PC, but I don't see any reason why it shouldn't work. An 802 error usually means there's something wrong with the driver install, but everything you've posted looks ok.

Honestly I've not found any useful way to debug these 802 issues. You might be able to get more help by posting in the NVIDIA forums saying that cuInit(0) returns an 802 error.

Wyacheslaw8, 2023-12-30:
I have found the decision.
The problem is in Nvidia video card driver and in Nvidia Network Adapter in nForce3 chipset. I disabled Nvidia Network Adapter in BIOS.

`inxi -Fxz`
```
System:
  Kernel: 5.15.0-91-generic x86_64 bits: 64 compiler: gcc v: 11.4.0
    Desktop: Cinnamon 5.8.4 Distro: Linux Mint 21.2 Victoria
    base: Ubuntu 22.04 jammy
Machine:
  Type: Desktop Mobo: ASUSTeK model: A8N-E v: 2.XX
    serial: <superuser required> BIOS: Phoenix v: ASUS A8N-E Revision 1013
    date: 04/07/2006
CPU:
  Info: dual core model: AMD Opteron 185 bits: 64 type: MCP arch: K8 rev.E
    rev: 2 cache: L1: 256 KiB L2: 2 MiB
  Speed (MHz): avg: 1000 min/max: 1000/2600 cores: 1: 1000 2: 1000
    bogomips: 10453
  Flags: ht lm nx pae sse sse2 sse3
Graphics:
  Device-1: NVIDIA GP107 [GeForce GTX 1050 Ti] vendor: Gigabyte
    driver: nvidia v: 535.129.03 bus-ID: 01:00.0
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: nvidia
    unloaded: fbdev,modesetting,nouveau,vesa gpu: nvidia
    resolution: 1920x1200~60Hz
  OpenGL: renderer: NVIDIA GeForce GTX 1050 Ti/PCIe/SSE2
    v: 4.6.0 NVIDIA 535.129.03 direct render: Yes
Audio:
  Device-1: NVIDIA GP107GL High Definition Audio vendor: Gigabyte
    driver: snd_hda_intel v: kernel bus-ID: 01:00.1
  Sound Server-1: ALSA v: k5.15.0-91-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Message: No device data found.
  IF-ID-1: enxfcde56ff0106 state: unknown speed: -1 duplex: half
    mac: <filter>
Bluetooth:
  Device-1: Qualcomm Mobile Router type: USB driver: rndis_host v: kernel
    bus-ID: 1-4.1:3
Drives:
  Local Storage: total: 953.87 GiB used: 100.73 GiB (10.6%)
  ID-1: /dev/sda vendor: Samsung model: SSD 850 PRO 1TB size: 953.87 GiB
Partition:
  ID-1: / size: 91.11 GiB used: 36.02 GiB (39.5%) fs: ext4 dev: /dev/sda5
  ID-2: /boot/efi size: 512.1 MiB used: 6.1 MiB (1.2%) fs: vfat
    dev: /dev/sda3
Swap:
  ID-1: swap-1 type: partition size: 5.59 GiB used: 257 MiB (4.5%)
    dev: /dev/sda6
Sensors:
  System Temperatures: cpu: 35.0 C mobo: 34.0 C gpu: nvidia temp: 42 C
  Fan Speeds (RPM): cpu: 1721 case-1: 1939 gpu: nvidia fan: 0%
  Power: 12v: N/A 5v: N/A 3.3v: 3.25 vbat: N/A
Info:
  Processes: 221 Uptime: 1h 28m Memory: 3.82 GiB used: 2.31 GiB (60.4%)
  Init: systemd runlevel: 5 Compilers: gcc: 11.4.0 Packages: 2459 Shell: Bash
  v: 5.1.16 inxi: 3.3.13
```

And now the  Nvidia-vaapi-driver is working fine in my old PC.

`NVD_LOG=1 vainfo --display drm --device /dev/dri/renderD128`
```
      5590.954726691 [10063-10063] ../src/vabackend.c:2145       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      5590.954803793 [10063-10063] ../src/vabackend.c:2154       __vaDriverInit_1_0 Now have 0 (0 max) instances
      5590.954907154 [10063-10063] ../src/vabackend.c:2180       __vaDriverInit_1_0 Selecting Direct backend
      5590.979429497 [10063-10063] ../src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      5590.979576717 [10063-10063] ../src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 535.129.03, major version: 535, minor version: 129
      5590.979607446 [10063-10063] ../src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 100 1 0 fe
vainfo: VA-API version: 1.14 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      5591.277208264 [10063-10063] ../src/vabackend.c:2055              nvTerminate Terminating 0x5608206ea440
      5591.277302406 [10063-10063] ../src/vabackend.c:2069              nvTerminate Now have 0 (0 max) instances

```
Happy New Year!

closed 2024-01-27 by elFarto