elFarto/nvidia-vaapi-driver #263: Errors with firefox

valen38, 2023-12-17:
I've done all the steps in the readme but it still won't work with firefox

os: voidlinux
kernel: 6.5.13
driver version: 535.129.03
gpu: RTX 3050, intel integrated graphics

output when starting firefox
```
$ NVD_LOG=1 firefox
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: glxtest: libEGL no displayVENDOR (t=0.626221) [GFX1-]: glxtest: libEGL no displayVENDOR
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: glxtest: libEGL no displayVENDOR (t=0.626221) |[1][GFX1-]: vaapitest: ERROR (t=0.682221) [GFX1-]: vaapitest: ERROR
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: glxtest: libEGL no displayVENDOR (t=0.626221) |[1][GFX1-]: vaapitest: ERROR (t=0.682221) |[2][GFX1-]: vaapitest: VA-API test failed: failed to open renderDeviceFD.
 (t=0.682221) [GFX1-]: vaapitest: VA-API test failed: failed to open renderDeviceFD.
```

```
$ NVD_LOG=1 vainfo
Trying display: wayland
Trying display: x11
libva info: VA-API version 1.20.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      1044.365954749 [4434-4434] ../src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      1044.365957533 [4434-4434] ../src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1044.365958764 [4434-4434] ../src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      1044.369999793 [4434-4434] ../src/backend-common.c:  31            isNvidiaDrmFd Invalid driver for DRM device: i915
      1044.370015665 [4434-4434] ../src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD129
      1044.370016979 [4434-4434] ../src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      1044.370031710 [4434-4434] ../src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 535.129.03, major version: 535, minor version: 129
      1044.370033746 [4434-4434] ../src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 100 1 2 6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.20 (libva 2.20.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            : VAEntrypointVLD
      VAProfileMPEG2Main              : VAEntrypointVLD
      VAProfileVC1Simple              : VAEntrypointVLD
      VAProfileVC1Main                : VAEntrypointVLD
      VAProfileVC1Advanced            : VAEntrypointVLD
      VAProfileH264Main               : VAEntrypointVLD
      VAProfileH264High               : VAEntrypointVLD
      VAProfileH264ConstrainedBaseline: VAEntrypointVLD
      VAProfileHEVCMain               : VAEntrypointVLD
      VAProfileVP8Version0_3          : VAEntrypointVLD
      VAProfileVP9Profile0            : VAEntrypointVLD
      VAProfileAV1Profile0            : VAEntrypointVLD
      VAProfileHEVCMain10             : VAEntrypointVLD
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      1044.413799034 [4434-4434] ../src/vabackend.c:2050              nvTerminate Terminating 0x5570edcc7700
      1044.413852087 [4434-4434] ../src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
```

elFarto, 2023-12-21:
Can you try starting Firefox with this command: `MOZ_GFX_DEBUG=1 firefox`. This will enable vaapitest logging which should shed some light onto what's happening. Make sure not to have any instances of Firefox running before you run that command, or it won't work correctly.

valen38, 2023-12-21:
here
```
$ MOZ_GFX_DEBUG=1 firefox
GLX_TEST: childgltest start
GLX_TEST: get_pci_status start
GLX_TEST: get_pci_status finished
GLX_TEST: x11_egltest start
GLX_TEST: get_egl_status start
GLX_TEST: get_egl_gl_status start
GLX_TEST: get_egl_gl_status finished
GLX_TEST: get_egl_status finished with return: 1
GLX_TEST: get_xrandr_info start
GLX_TEST: get_xrandr_info finished
GLX_TEST: x11_egltest finished
PCI_VENDOR_ID
0x8086
PCI_DEVICE_ID
0x46a3
PCI_VENDOR_ID
0x10de
PCI_DEVICE_ID
0x25a0
DRI_DRIVER
iris
VENDOR
Intel
RENDERER
Mesa Intel(R) Graphics (ADL GT2)
VERSION
4.6 (Compatibility Profile) Mesa 23.3.1
TFP
TRUE
DRM_RENDERDEVICE
/dev/dri/renderD128
MESA_VENDOR_ID
0x8086
MESA_DEVICE_ID
0x46a3
DDX_DRIVER
modesetting;NVIDIA-G0;
TEST_TYPE
EGL
GLX_TEST: childgltest finished
[Parent 2849, Main Thread] WARNING: Theme parsing error: gtk.css:6691:68: Invalid name of pseudo-class: 'glib warning', file /topsrcdir/toolkit/xre/nsSigHandlers.cpp:187

(Firefox:2849): Gtk-WARNING **: 15:50:11.861: Theme parsing error: gtk.css:6691:68: Invalid name of pseudo-class
[Parent 2849, Main Thread] WARNING: Theme parsing error: gtk.css:6691:68: Invalid name of pseudo-class: 'glib warning', file /topsrcdir/toolkit/xre/nsSigHandlers.cpp:187

(Firefox:2849): Gtk-WARNING **: 15:50:11.996: Theme parsing error: gtk.css:6691:68: Invalid name of pseudo-class
[Parent 2849, Main Thread] WARNING: Theme parsing error: gtk.css:6691:68: Invalid name of pseudo-class: 'glib warning', file /topsrcdir/toolkit/xre/nsSigHandlers.cpp:187

(Firefox:2849): Gtk-WARNING **: 15:50:12.020: Theme parsing error: gtk.css:6691:68: Invalid name of pseudo-class
vaapitest start, device /dev/dri/renderD128
libva info: VA-API version 1.20.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva error: /usr/lib64/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
vaInitialize failed 1
ERROR
VA-API test failed: failed to initialise VAAPI connection.
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: vaapitest: ERROR (t=2.0091) [GFX1-]: vaapitest: ERROR
Crash Annotation GraphicsCriticalError: |[0][GFX1-]: vaapitest: ERROR (t=2.0091) |[1][GFX1-]: vaapitest: VA-API test failed: failed to initialise VAAPI connection.
 (t=2.0091) [GFX1-]: vaapitest: VA-API test failed: failed to initialise VAAPI connection.

console.error: ({})
console.error: (new TypeError("a is null", "resource://gre/modules/components-utils/mozjexl.js", 1))
```

elFarto, 2023-12-22:
The error you're getting is caused because you're attempting to run Firefox on the Intel card, but use the NVIDIA card for video decoding. This is not currently a supported setup. You'll need to run Firefox on the NVIDIA card. However trying to convince Firefox to run on the NVIDIA card without switching everything to use it doesn't appear to work (Firefox seems to have some hardcoded places where it uses the default EGL implementation which causes failures).

valen38, 2023-12-22:
Running firefox with the nvidia gpu worked

closed 2023-12-22 by valen38