elFarto/nvidia-vaapi-driver #264: Ubuntu 22.04 Direct Backend no Video Engine Utilization

KTong821, 2023-12-29:
Thanks for the driver and support, I've been trying to get it working on the following specs:

Intel i7 13700KF
GeForce RTX 4070
Using NVIDIA driver metapackage from nvidia-driver-545 (proprietary)
Ubuntu 22.04

Built from master latest on the direct backend, here's `vainfo`:

```
libva info: VA-API version 1.19.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      1614.901599643 [25418-25418] ../src/vabackend.c:2154       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      1614.901603530 [25418-25418] ../src/vabackend.c:2163       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1614.901604599 [25418-25418] ../src/vabackend.c:2189       __vaDriverInit_1_0 Selecting Direct backend
      1614.905328572 [25418-25418] ../src/direct/direct-export-buf.c:  66      direct_initExporter Searching for GPU: 0 0 128
      1614.905339441 [25418-25418] ../src/direct/direct-export-buf.c:  88      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      1614.905340761 [25418-25418] ../src/direct/nv-driver.c: 254            init_nvdriver Initing nvdriver...
      1614.905350018 [25418-25418] ../src/direct/nv-driver.c: 272            init_nvdriver NVIDIA kernel driver version: 545.23.08, major version: 545, minor version: 23
      1614.905365145 [25418-25418] ../src/direct/nv-driver.c: 279            init_nvdriver Got dev info: 100 1 2 6
libva info: va_openDriver() returns 0
vainfo: VA-API version: 1.19 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      1615.079411733 [25418-25418] ../src/vabackend.c:2064              nvTerminate Terminating 0x55e7d00d1e50
      1615.079886296 [25418-25418] ../src/vabackend.c:2078              nvTerminate Now have 0 (0 max) instances
```

I noticed I hadn't run the `extract_headers` script after building, so I go ahead and do so:

```
Dropping privileges to 'kevin' before running ninja...
ninja: Entering directory `/home/kevin/nvidia-vaapi-driver/build'
[12/15] Compiling C object nvidia_drv_video.so.p/src_direct_nv-driver.c.o
FAILED: nvidia_drv_video.so.p/src_direct_nv-driver.c.o 
cc -Invidia_drv_video.so.p -I. -I.. -I../nvidia-include -I/usr/local/include -I/usr/include/libdrm -I/usr/include/gstreamer-1.0 -I/usr/include/x86_64-linux-gnu -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -fvisibility=hidden -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c11 -O0 -g -Wno-missing-field-initializers -Wno-unused-parameter -Werror=format -Werror=format-security -Werror=incompatible-pointer-types -Werror=init-self -Werror=int-conversion -Werror=missing-declarations -Werror=missing-prototypes -Werror=pointer-arith -Werror=undef -Werror=vla -Wsuggest-attribute=format -Wwrite-strings -fPIC -pthread -MD -MQ nvidia_drv_video.so.p/src_direct_nv-driver.c.o -MF nvidia_drv_video.so.p/src_direct_nv-driver.c.o.d -o nvidia_drv_video.so.p/src_direct_nv-driver.c.o -c ../src/direct/nv-driver.c
../src/direct/nv-driver.c: In function ‘get_device_info’:
../src/direct/nv-driver.c:205:51: error: storage size of ‘devInfo545’ isn’t known
  205 |         struct drm_nvidia_get_dev_info_params_545 devInfo545;
      |                                                   ^~~~~~~~~~
../src/direct/nv-driver.c:206:35: error: ‘DRM_IOCTL_NVIDIA_GET_DEV_INFO_545’ undeclared (first use in this function); did you mean ‘DRM_IOCTL_NVIDIA_GET_DEV_INFO’?
  206 |         const int ret = ioctl(fd, DRM_IOCTL_NVIDIA_GET_DEV_INFO_545, &devInfo545);
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                   DRM_IOCTL_NVIDIA_GET_DEV_INFO
../src/direct/nv-driver.c:206:35: note: each undeclared identifier is reported only once for each function it appears in
../src/direct/nv-driver.c:205:51: warning: unused variable ‘devInfo545’ [-Wunused-variable]
  205 |         struct drm_nvidia_get_dev_info_params_545 devInfo545;
      |                                                   ^~~~~~~~~~
[14/15] Compiling C object nvidia_drv_video.so.p/src_vabackend.c.o
ninja: build stopped: subcommand failed.
Could not rebuild /home/kevin/nvidia-vaapi-driver/build
```

During initial meson setup all dependencies are YES, and `/etc/environment` looks like

```
LIBVA_DRIVER_NAME=nvidia
NVD_BACKEND=direct
MOZ_DISABLE_RDD_SANDBOX=1
```

Trawled through exiting threads looking for advice, can't seem to find anything that works. Firefox doesn't show up in `nvidia-smi` nor under Video Engine Utilization in `nvidia-settings`. `LIBVA_MESSAGING_LEVEL=1` isn't giving causing `firefox` command to produce anything. Any advice appreciated!

drhashes, 2023-12-30:
> I noticed I hadn't run the extract_headers script after building, so I go ahead and do so

This is not needed and should not be done. 

Don't build from master except for testing. 
The latest release is here: 
https://github.com/elFarto/nvidia-vaapi-driver/archive/refs/tags/v0.0.11.tar.gz

gunzip v0.0.11.tar.gz
tar xf v0.0.11.tar
cd nvidia-vaapi-driver
meson setup build
sudo meson install -C build

Restart firefox.
This will work if your system is setup right.
Read the instructions at:
https://github.com/elFarto/nvidia-vaapi-driver

Finally, make sure the environment are setup right.
I have:
export NVD_BACKEND=direct
export MOZ_DISABLE_RDD_SANDBOX=1
export LIBVA_DRIVER_NAME=nvidia

in /etc/environment

reboot.

elFarto, 2023-12-31:
Yes, the extract_headers.sh script is there for reference. Unfortunately the NVIDIA driver has made changes that has led me to make manual changes to the headers to make it compatible with multiple versions of the driver, so the script doesn't really work anymore.

However, building from master should work.

KTong821, 2024-01-05:
Works beautifully now. Many thanks.

closed 2024-01-05 by KTong821