elFarto/nvidia-vaapi-driver #266: libva error: vaGetDriverNames() failed with unknown libva error

s1nceri7y, 2024-01-14:
Arch Linux 6.6.10-zen1-1-zen, x11.
VA-API version: 1.20 (libva 2.20.1), nvidia-vaapi-driver-0.0.11, NVIDIA kernel driver version: 545.29.06

The hardware decoding is not working neither in firefox nor in chromium or vlc.
The vainfo is crashing on x11 but works on DRM.

  
I mentioned my problem in [other issue](https://github.com/elFarto/nvidia-vaapi-driver/issues/261#issuecomment-1868527462) some time ago.

<details><summary>uname -a</summary>
<p>
Linux a66 6.6.10-zen1-1-zen #1 ZEN SMP PREEMPT_DYNAMIC Fri, 05 Jan 2024 16:20:18 +0000 x86_64 GNU/Linux

</p>
</details> 

<details><summary>NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo</summary>
<p>

```
% NVD_LOG=1 LIBVA_DRIVER_NAME=nvidia vainfo
Trying display: wayland
Trying display: x11
libva error: vaGetDriverNames() failed with unknown libva error
      1726.304124993 [8365-8365] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
      1726.304134250 [8365-8365] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1726.304136815 [8365-8365] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      1726.310611388 [8365-8365] ../nvidia-vaapi-driver-0.0.11/src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      1726.310619653 [8365-8365] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
      1726.310640583 [8365-8365] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 545.29.06, major version: 545, minor version: 29
      1726.310645362 [8365-8365] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 900 1 2 6
vainfo: VA-API version: 1.20 (libva 2.20.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
      1726.400928736 [8365-8365] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x55d0751d5410
      1726.401010310 [8365-8365] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances
``` 

</p>
</details> 

<details><summary> NVD_LOG=1 vainfo --display drm</summary>
<p>

```
x@a66 ~ % 
Trying display: drm
      2421.729795953 [8525-8525] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      2421.729806553 [8525-8525] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2421.729810651 [8525-8525] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
      2421.736128922 [8525-8525] ../nvidia-vaapi-driver-0.0.11/src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
      2421.736137638 [8525-8525] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
[... 84 lines cut ...]

<details><summary> vlc </summary>
<p>

```
% vlc
VLC media player 3.0.20 Vetinari (revision 3.0.20-0-g6f0d0ab126b)
[0000558362d00550] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
[0000558362d8e6b0] main playlist: playlist is empty
libva error: vaGetDriverNames() failed with unknown libva error
[00007f50c4001f70] glconv_vaapi_x11 gl error: vaDeriveImage: operation failed
[00007f50cc000c70] main video output error: video output creation failed
[00007f50e000de10] main decoder error: failed to create video output
```

</p>
</details> 

I've attached the vlc --verbose=2, hope it helps. 
[video.log](https://github.com/elFarto/nvidia-vaapi-driver/files/13930009/video.log)

Video Engine utilization: 0% (in nvidia-settings)

Let me know if I can provide more details. 
Thanks!

drhashes, 2024-01-14:
Check if you have the kernel option:
nvidia-drm.modeset=1

Use the shell command:
cat /proc/cmdline

s1nceri7y, 2024-01-14:
> Check if you have the kernel option: nvidia-drm.modeset=1

I do.

```shell
 x@a66 ~ % cat /proc/cmdline                                                 
BOOT_IMAGE=/boot/vmlinuz-linux-zen root=UUID=... rw nvidia-drm.modeset=1 loglevel=3 quiet
x@a66 ~ % sudo cat /sys/module/nvidia_drm/parameters/modeset
Y
```

drhashes, 2024-01-14:
Just some info regarding the false error:

 "libva error: vaGetDriverNames() failed with unknown libva error"
i.e. as you are saying:
"The vainfo is crashing on x11 but works on DRM."

This is just a pecularity with vainfo in the package libva-utils.
It's testing:
Trying display: wayland
Trying display: x11

and gives up and prints the error. A proper behaved vainfo would just continou with testing drm:
Trying display: drm

and it would have found it. This is reported here:
https://github.com/intel/libva-utils/issues/356

drhashes, 2024-01-14:
(By the way, the above "error" is unrelated to your problems.)

Now in firefox about:config set:
media.av1.enabled false

This way we disable av1 and it should fallback to vp9 8bit on youtube.
Then play:

https://www.youtube.com/watch?v=rFVpSwgCkCo

Make sure you run at 4k 60fps and also right click mouse and choose "stats for nerds". Here you can see Codecs: it should be vp09.

Check in nvidia-settings if "video engine utilization" now works.

s1nceri7y, 2024-01-14:
Thank you for the answer
The Video engine utilization still 0%. 
The GPU utilization however jumps from 10% up to 40-50% and CPU from 1% up to 20%.
Codecs: vp09.00.51.08.01.01.01.01.00 (315) / opus (251)

s1nceri7y, 2024-01-14:
Let me know if it's makes sense to rename the issue to something like `glconv_vaapi_x11 gl error: vaDeriveImage: operation failed`.
It's the error I saw in VLC console.

drhashes, 2024-01-14:
Check that your local ffmpeg supports vaapi:
ffmpeg -hwaccels

Should say something like:
Hardware acceleration methods: vaapi

drhashes, 2024-01-14:
Also, perhaps there is a problem with your local firefox installation?

You could
sudo apt-get install mpv

and assuming you got some local movie.mkv

mpv --vo=gpu --hwdec=vaapi-copy movie.mkv

Then again check Video Engine Utilization..
That works here.

s1nceri7y, 2024-01-14:
<details><summary>%  ffmpeg -hwaccels</summary>
<p>

```
ffmpeg version n6.1.1 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 13.2.1 (GCC) 20230801
  configuration: --prefix=/usr --disable-debug --disable-static --disable-stripping --enable-amf --enable-avisynth --enable-cuda-llvm --enable-lto --enable-fontconfig --enable-frei0r --enable-gmp --enable-gnutls --enable-gpl --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libdav1d --enable-libdrm --enable-libfreetype --enable-libfribidi --enable-libgsm --enable-libharfbuzz --enable-libiec61883 --enable-libjack --enable-libjxl --enable-libmodplug --enable-libmp3lame --enable-libopencore_amrnb --enable-libopencore_amrwb --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libplacebo --enable-libpulse --enable-librav1e --enable-librsvg --enable-librubberband --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libv4l2 --enable-libvidstab --enable-libvmaf --enable-libvorbis --enable-libvpl --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxcb --enable-libxml2 --enable-libxvid --enable-libzimg --enable-nvdec --enable-nvenc --enable-opencl --enable-opengl --enable-shared --enable-version3 --enable-vulkan
  libavutil      58. 29.100 / 58. 29.100
  libavcodec     60. 31.102 / 60. 31.102
  libavformat    60. 16.100 / 60. 16.100
  libavdevice    60.  3.100 / 60.  3.100
  libavfilter     9. 12.100 /  9. 12.100
  libswscale      7.  5.100 /  7.  5.100
  libswresample   4. 12.100 /  4. 12.100
  libpostproc    57.  3.100 / 57.  3.100
Hardware acceleration methods:
vdpau
cuda
vaapi
qsv
drm
opencl
vulkan
```
</p>
</details> 

**The mpv command worked for me!**
Video Engine utilization: 16-19% (vp9 3840x2160 59.940fps 10 bit)
The overall GPU utilization still high(46% vs 7% on pause), but the cpu usage is ~0%. I think its a good sign.

The question is: what's wrong with my fresh firefox installation, chromium and vlc? I've created a new profile in about:profiles, but it still doesn't using the HW acceleration.  

With media.ffmpeg.vaapi.enabled=true, Firefox reports the HW acceleration codecs but doesn't use them. 
[image]

drhashes, 2024-01-15:
chromium and vlc you can forget. This library does not support those apps.

If your mpv is 0.35 or newer
mpv --version

then 
--hwdec=vaapi
instead of
--hwdec=vaapi-copy

should work which is more efficient.

Regarding firefox, I should go through
https://github.com/elFarto/nvidia-vaapi-driver

very carefully so nothing important was missed setting things up. Otherwise you need to figure out what has happened in arch's firefox. The version and so on asking others if they experienced something similar.

drhashes, 2024-01-15:
1. start firefox like this:
NVD_BACKEND=direct firefox

2. start firefox like this:
MOZ_DRM_DEVICE=/dev/dri/renderD128 firefox

3.
Make sure firefox is not installed using flatpak or snap.

Some more Arch info here:
https://github.com/elFarto/nvidia-vaapi-driver/issues/23

mirh, 2024-01-15:
I regret taking shortcuts for fixes, rather than actually getting down to the precise causes.. 
But just for the records, the specific error mentioned in the title (which isn't the problem here, but at the same time I don't think it's entirely unrelated) can be fixed by compiling libva with the `-Dwith_legacy=nvctrl` arch-meson build option.

s1nceri7y, 2024-01-15:
Following this advice https://github.com/elFarto/nvidia-vaapi-driver/issues/258#issuecomment-1820748049 I've removed previously hardcoded MOZ_DRM_DEVICE from /etc/environment and  installed libva-git with `-Dwith_legacy=nvctrl`.

<details><summary>/etc/environment      </summary>
<p>

```
% cat /etc/environment
#
# This file is parsed by pam_env module
#
# Syntax: simple "KEY=VAL" pairs on separate lines
#
LIBVA_DRIVER_NAME=nvidia
VDPAU_DRIVER=nvidia
NVD_BACKEND=direct
```

</p>
</details> 

Now I don't have any errors in vainfo:

<details><summary>% NVD_LOG=1 vainfo      </summary>
<p>

```
% NVD_LOG=1 vainfo                                      
Trying display: wayland
Trying display: x11
       520.236233291 [3256-3256] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2140       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 10
       520.236242739 [3256-3256] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2149       __vaDriverInit_1_0 Now have 0 (0 max) instances
       520.236245504 [3256-3256] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2175       __vaDriverInit_1_0 Selecting Direct backend
       520.242476106 [3256-3256] ../nvidia-vaapi-driver-0.0.11/src/direct/direct-export-buf.c:  85      direct_initExporter Found NVIDIA GPU 0 at /dev/dri/renderD128
       520.242484070 [3256-3256] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 246            init_nvdriver Initing nvdriver...
       520.242506542 [3256-3256] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 264            init_nvdriver NVIDIA kernel driver version: 545.29.06, major version: 545, minor version: 29
       520.242511621 [3256-3256] ../nvidia-vaapi-driver-0.0.11/src/direct/nv-driver.c: 271            init_nvdriver Got dev info: 900 1 2 6
vainfo: VA-API version: 1.21 (libva 2.20.1)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD
      VAProfileHEVCMain               :	VAEntrypointVLD
      VAProfileVP8Version0_3          :	VAEntrypointVLD
      VAProfileVP9Profile0            :	VAEntrypointVLD
      VAProfileAV1Profile0            :	VAEntrypointVLD
      VAProfileHEVCMain10             :	VAEntrypointVLD
      VAProfileHEVCMain12             :	VAEntrypointVLD
      VAProfileVP9Profile2            :	VAEntrypointVLD
      VAProfileHEVCMain444            :	VAEntrypointVLD
      VAProfileHEVCMain444_10         :	VAEntrypointVLD
      VAProfileHEVCMain444_12         :	VAEntrypointVLD
       520.344225598 [3256-3256] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2050              nvTerminate Terminating 0x56dd3802a410
       520.344301118 [3256-3256] ../nvidia-vaapi-driver-0.0.11/src/vabackend.c:2064              nvTerminate Now have 0 (0 max) instances

```
</p>
</details> 

The firefox still doesn't used the HW acceleration.

I've tried `MOZ_DISABLE_RDD_SANDBOX=1 firefox` and now it's worked.
I'm not sure how safe is `MOZ_DISABLE_RDD_SANDBOX`. I also remember that it's worked before without this flag. It seemed to me like it's something redundant, but it's not. 
https://www.reddit.com/r/firefox/comments/v0esaq/linux_users_good_news_seems_like_the_bug_that/

Thank you everyone for the help.
The issue is fixed for me. Let me know if I should close it.

mirh, 2024-01-15:
That's coming and explained in https://github.com/elFarto/nvidia-vaapi-driver/issues/6#issuecomment-1007673789

drhashes, 2024-01-15:
Congrats!

Looks to me that the only real problem was that
MOZ_DISABLE_RDD_SANDBOX=1
was needed.

Sometimes one might think it's something really complicated only to find out..no.

I think you can close this now.

closed 2024-01-15 by s1nceri7y